SpikeInterface/spikeinterface #4: Missing dependency

mhhennig, 2019-08-10:
When installing SI from this location, the `requests` package is not installed.
`spikesorters/ironclust/mdaio.py` wants it.

samuelgarcia, 2019-08-26:
I think we planed to move mdio from sorters to extractirs and make a weak dependency.
@colehurwitz @alejoe91 @magland is it still the case ?

magland, 2019-08-26:
my opinion is to include mdaio in spikesorters to reduce the dependencies. I don't think that spikeextractors should necessarily expose mdaio.

alejoe91, 2019-09-02:
then we should add requests as requirement in the setup.py!

alejoe91, 2019-09-04:
https://github.com/SpikeInterface/spikesorters/pull/50

samuelgarcia, 2019-09-04:
So we can remove from requirement.txt no ?

alejoe91, 2019-09-04:
yes, I'll merge, release, and update

alejoe91, 2019-09-08:
https://github.com/SpikeInterface/spikeinterface/pull/23

closed 2019-09-08 by alejoe91