SpikeInterface/spikeinterface #14: Spikinterface has no attributes

McDr, 2019-09-02:
I first tried to run spiketoolkit without the spikeinterface meta package, just installed separately. I tried to run the command 
`import spiketoolkit as st`
`st.sorters.available_sorters()`

and got the error 

> AttributeError: module 'spiketoolkit' has no attribute 'sorters'

I found an open issue in spiketoolkit that suggested I try running spikeinterface instead as it is now packaged this way.  [](https://github.com/SpikeInterface/spiketoolkit/issues/209) 

I tried this and got the same thing: 
```
import spikeinterface as si 
si.sorters.available_sorters()
```

> AttributeError: module 'spikeinterface' has no attribute 'sorters'

It also seems that spikeinterface has no attributes that work (sorters, extractors, etc.). However, when I try it in Spyder IDE, it suggests all of them. 
[image]

samuelgarcia, 2019-09-03:
For the first part.
For cycle import reason sorters is now totally out of spiketoolkit.
So you need to do ```import spikesorters as ss``` and ```ss.available_sorters()```

For ```AttributeError: module 'spikeinterface' has no attribute 'sorters' ```, this is strange.
Because, I have the same bug when I do it in ipython/jupyter but not in a script.

@alejoe91 : do you experience the same ?

See here https://github.com/SpikeInterface/spikeinterface/pull/13, there is a test_import that is OK with travis and my machine. But the same in ipython do not work.
We could fix this by importing everything in the init but the import would be slower.
with:
```python
from . import extractors
from . import sorters
...
```
what do you htink ?

alejoe91, 2019-09-03:
@McDr can you try:

```
import spikeinterface.sorters as ss
ss.available_sorters()
```

This works for me.

@samuelgarcia this is strange. If I do:

```
import spikeiterface.sorters as ss
ss.available_sorters()
```

it works. If I only import `spikeinterface` all I can access is `version`...

samuelgarcia, 2019-09-03:
SAme for me in ipython but not in a script this the test I made on the PR.

alejoe91, 2019-09-03:
@samuelgarcia what you suggest works. Let's do that

samuelgarcia, 2019-09-03:
Yes but it make the metapackage enormous!!!!
Like if scipy would import everything at the beginning but not on demand.
I am not very happy with this solution.

alejoe91, 2019-09-03:
what about lazy imports?

samuelgarcia, 2019-09-03:
It was a lazy import but this bug sometimes.

The ```from . import sorters``` is a full import.

alejoe91, 2019-09-03:
I'll make a PR to show you what I mean :)

alejoe91, 2019-09-03:
Ok I was kidding. What I had in mind doesn't work. Let's just use this:

```
from . import extractors
from . import toolkit
from . import sorters
from . import comparison
from . import widgets
```

It's not so huge. Alternatively, we can 'force' the user to import separate modules when needed:

```
import spikeinterface.extractors as se
```

McD-NMI, 2019-09-03:
@alejoe91 What you suggested: 

`importspikeinter.spikesorters as ss` 
`ss.available_sorters() `

works for me as script or in IPython.

alejoe91, 2019-09-03:
Hi @McD-NMI and @McDr 

With the latest commit, also:

```
import spikeinterface as si
si.sorters.available_sorters()
```

should work. Can you confirm that?

McD-NMI, 2019-09-03:
@alejoe91 

I can confirm this works now!

closed 2019-09-03 by McDr

yilmazduygu, 2021-09-18:
Hi! 
I've been trying to get my recordings sorted using one of the tutorial notebooks. Unfortunately I can't retrieve the waveforms, because I get the error `AttributeError: module 'spikeinterface' has no attribute 'WaveformExtractor'`
I'm also getting this output when I try `si.version`:
`<module 'spikeinterface.version' from '/opt/anaconda3/envs/spiketutorial/lib/python3.6/site-packages/spikeinterface/version.py'>`, which I thought was strange.

Attaching this, in case it's helpful:
[pip_list.txt](https://github.com/SpikeInterface/spikeinterface/files/7190460/pip_list.txt)

-duygu

alejoe91, 2021-09-19:
Hi @yilmazduygu 

The new version (>=0.90) doesn't use the packages `spikeextractors`, `spiketoolkit`, `spikesorters`, `spikecomparison`, and `spikewidgets`. So you can uninstall them. To access the version you can do `si.__version__`.

Can you share your entire script/notebook? Which tutorial are you running? Indeed it'strange that the `WaveformExtractor` is not found...

Cheers
Alessio