SpikeInterface/spikeinterface #40: Getting started tutorial use of mountainsort throws error

rly, 2019-10-08:
When I run this line:
https://github.com/SpikeInterface/spikeinterface/blob/75dc13fdb947f031f7448167e6daa83f3116dc81/examples/getting_started/plot_getting_started.py#L135

I get the following error:
```

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-13-6da5239fbc7f> in <module>
----> 1 sorting_MS4 = ss.run_mountainsort4(recording=recording_cmr, detect_threshold=6)

~/.conda/envs/mybase/envs/ms/lib/python3.6/site-packages/spikesorters/sorterlist.py in run_mountainsort4(*args, **kargs)
    108 
    109 def run_mountainsort4(*args, **kargs):
--> 110     return run_sorter('mountainsort4', *args, **kargs)
    111 
    112 

~/.conda/envs/mybase/envs/ms/lib/python3.6/site-packages/spikesorters/sorterlist.py in run_sorter(sorter_name_or_class, recording, output_folder, delete_output_folder, grouping_property, parallel, verbose, **params)
     51                          parallel=parallel, verbose=verbose, delete_output_folder=delete_output_folder)
     52     sorter.set_params(**params)
---> 53     sorter.run()
     54     sortingextractor = sorter.get_result()
     55 

~/.conda/envs/mybase/envs/ms/lib/python3.6/site-packages/spikesorters/basesorter.py in run(self)
    100         if not self.parallel:
    101             for i, recording in enumerate(self.recording_list):
--> 102                 self._run(recording, self.output_folders[i])
    103         else:
    104             # run in threads

~/.conda/envs/mybase/envs/ms/lib/python3.6/site-packages/spikesorters/mountainsort4/mountainsort4.py in _run(self, recording, output_folder)
    108             detect_interval=p['detect_interval'],
    109             num_workers=p['num_workers'],
--> 110             verbose=self.verbose
    111         )
    112 

TypeError: mountainsort4() got an unexpected keyword argument 'verbose'
```

Packages:
ml-ms4alg                     0.2.3
spikecomparison               0.2.0
spikeextractors               0.7.0
spikeforestwidgets            0.1.4
spikeinterface                0.9.0
spikemetrics                  0.1.2
spikesorters                  0.2.0
spiketoolkit                  0.5.0
spikewidgets                  0.3.1

rly, 2019-10-08:
Ah, never mind! I had installed ml-ms4alg from conda, which has an old version that is missing the `verbose` keyword.

closed 2019-10-08 by rly

kwrigh42, 2020-07-29:
I am getting the same error message: TypeError: mountainsort4() got an unexpected keyword argument 'verbose'. How did you fix this?

samuelgarcia, 2020-07-29:
You have to update ml-ms4alg and all SI package.