SpikeInterface/spikeinterface #45: Saving/Loading combined recording/sorting data

McD-NMI, 2019-10-25:
So I am having a having a bit of a conceptual problem with trying to design a pipeline (I am quite new to electrophysiology). Essentially I want to load my file, sort the data, export it to a NEO compatible format, and import it to Elephant for further analysis. I have got up to the point of sorting the data, and now I am trying to firgure the best way to export it for further analysis. 

However, it seems that for all file types, SpikeInterface writes Recording and Sorting data to different files... and I am wondering how this works for downstream analysis. I haven't worked with NEO/Elephant at all yet, but perhaps I can just load both files and merge them within Neo/Elephant? 

For example, I want to use the NixIO or Kwik extractors to write Recording and Sorting data... but would it be possible to combine these data into a single data source? 

Hope this makes sense.

colehurwitz, 2019-10-25:
So how our exporting works is that if write_recording or write_sorting is implemented by a format class (it is a static function), then ALL extractors can be turned into that format. 

So if you have a sorting output, you can choose any of the export functions and convert that output into the export format.

From spikeextractors import MDASortingExtractor

MDASortingExtractor.write_sorting(sorting=sorting, save_path=firings.mda)

So all formats can then be exported to MDA in this example. As you can see, you don’t need a MDA object to export, just the class reference. You can choose any format that is compatible with NEO and has a write function and do the same thing I just showed. Let me know if that makes sense!

colehurwitz, 2019-10-25:
Ooh, I think I understand your question. It seems like you want to export recordings and sorting to the same file. Currently, some exporters allow you to save to a preexisting file I believe so that may allow you to save both to the same file.

Also, the toolkit.export_to_phy saves both a recording and sorting into the phy format although it is a bit heavy to run.

McD-NMI, 2019-10-29:
Thanks! I can work with just the sorted data for now, but it would be a nice addition to NWB/NIX, and the Neo extractor as well when that come out.

colehurwitz, 2019-11-11:
As it is now, export_to_phy tries to extract _all the spikes_ from the recording so you can do manual curation later. This will unfortunately not scale to large datasets. I am not sure exactly how to deal with that. @alejoe91 any ideas or plans for this? I am not sure where we left it.

McD-NMI, 2019-11-12:
Currently I am trying to write my own script that will map my recording and sorting to a Neo object, and then export it to a Neo-Nix object using the NeoIO. I havent had much time to work on this though. When I am done, I could try to add this to SpikeInterface if you want. I am trying to generalize it as much as a I can so it is not only specific to my MEA design. 

One thing that might be nice with the SpikeInterface documentation is for the Compatible Technology to have a table that shows the current capabilities of each extractor (whether it can read and/or write, and what data it can extract/write). It might also be nice to have an infographic overview of the SpikeInterface data structure, to make it easier for mapping the data?

alejoe91, 2019-11-12:
> As it is now, export_to_phy tries to extract _all the spikes_ from the recording so you can do manual curation later. This will unfortunately not scale to large datasets. I am not sure exactly how to deal with that. @alejoe91 any ideas or plans for this? I am not sure where we left it.

@colehurwitz unfortunately phy needs to extract all spikes to compute features from all of them. Otherwise the manual curation would be biased as some spikes will not show...I don't think there is a way around that

alejoe91, 2019-11-12:
> Currently I am trying to write my own script that will map my recording and sorting to a Neo object, and then export it to a Neo-Nix object using the NeoIO. I havent had much time to work on this [...]

Hi @McD-NMI 
I think having a function in spikeextractors that exports to NEO will be a good addition. The user should be able to pass any combination of recording and sorting extractor:

```
neo_block = se.export_to_neo(recording=rec_extractor, sorting=sort_extractor)
neo_block = se.export_to_neo(recording=rec_extractor, sorting=None)
neo_block = se.export_to_neo(recording=None, sorting=sort_extractor)
```

@samuelgarcia can probably help! I think it could go in the extraction_tools.

samuelgarcia, 2019-11-12:
Hi All.
Using NixIO format can be easy.
I can do it.

colehurwitz, 2019-11-12:
@alejoe91 thanks for replying! You are much more knowledgeable about this part than I am. 

Do you think the memory limitations of the phy exporter could potentially be overcome by extracting and writing out in chunks?

alejoe91, 2019-11-12:
mmm the waveforms are extracted to compute PCA features, so it's a bit more complicated than that. One could load a subset of features to compute the PCA model, then waveforms could be loaded, transformed, and saved out in chunks. But this would require an extensive rewrite.. I think that for large-scale recordings one needs an appropriate machine :)

colehurwitz, 2019-11-12:
Right, we would still need to extract waveforms for PCA, although I think we can pick an appropriate amount of spikes for that. It would be quite nice if this worked on regular sized machines as well. Also, I think some recordings will be too big for even the largest machines! 

I think I will make an issue for this that we can address at any point (low priority).

samuelgarcia, 2021-07-22:
Done in 0.90

closed 2021-07-22 by samuelgarcia