SpikeInterface/spikeinterface #46: [new interface] MatClust

yarikoptic, 2019-11-01:
I saw https://www.mathworks.com/matlabcentral/fileexchange/39663-matclust used in a recent paper https://www.sciencedirect.com/science/article/pii/S0896627319307858?via%3Dihub , seems to be not interfaced here.  But also I am not sure if matclust is primarily for "assisted" and not automated sorting.

samuelgarcia, 2019-11-01:
It look like that mclust is more a GUI for manual cluster cut more that a spike sorter engine.
Not sure if it is relevant to port this in spikeinterface. Maybe it could be usefull to write an spikeextractor at least to format the data format in the good to export a result to mclust like we do for phy.

yarikoptic, 2019-11-01:
> It look like that mclust is more a GUI for manual cluster cut more that a spike sorter engine.

that is what I feared

> Maybe it could be usefull to write an spikeextractor at least to format the data format in the good to export a result to mclust like we do for phy.

do you want me to refile issue there or would you?