SpikeInterface/spikeinterface #48: Automatic citation maker

colehurwitz, 2019-11-13:
I was discussing with another open source developer who is implementing an automatic citation creator for people that use the framework.

I think we can do something similar with SpikeInterface where the citations can be grabbed for different file formats or spike sorters from some internal property that stores citation formats (bibtex, etc.). I can try to do something like this for spikeextractors first maybe..

colehurwitz, 2020-01-15:
I will get to this soon! I have some ideas. :)

yarikoptic, 2020-05-11:
not sure if that "open source developer" might have been me, but I would like to pitch about http://duecredit.org/ project which might be of help here! ;-)

closed 2021-07-22 by samuelgarcia

yarikoptic, 2021-07-22:
so what you have converged on?  duecredit sounds like a good fit FWIW ;)

samuelgarcia, 2021-07-22:
Hi Yarik,
We decide to do nothing at the moment.
We will re open this later if someone have time.