SpikeInterface/spikeinterface #49: installation of sorters

WeissShahaf, 2019-11-18:
after following all installation instructions, only herding spikes is available in spikeinterface when running print available sorters.

tried in windows 7,10, with python 3.7 3.7.4, anaconda3, miniconda.

WeissShahaf, 2019-12-15:
any thoughts?

mhhennig, 2019-12-15:
Could you describe in a bit more detail what you did? Note that for the matlab sorters you have to set an environment variable, pointing to the relevant directory. However, spykingcircus for example should work if you used pip to install. Can you run it standalone?

alexmcclanahan, 2020-01-17:
@mhhennig, super excited to experiment with SpikeInterface; I think it will yield very useful information for the field, so, thanks!

I've experienced similar issues to those outlined by @WeissShahaf. I have successfully installed SpikeInterface. The individual sorters that are python-based do install with pip or conda in terminal. However, they aren't able to be called, as an example (assuming a recording is properly loaded):

import spikesorters as ss (or even import spikeinterface.sorters as ss)
sorting_TDC = ss.run_tridesclous(recording, output_folder='tmp_TDC')

It also appears that the installed_sorters() function returns an empty array, while the available_sorters() returns that list (as it should, when I look in the sorterlist.py file). My intuition is that the sorters are being installed somewhere that the sorter files cannot find when they're called. I will keep tinkering to try to fix this locally on my machine and post if I make headway. 

Not sure if this is the same issue as @WeissShahaf, but hopefully this can facilitate a solution.

samuelgarcia, 2020-02-25:
Hi @alexmcclanahan 
Could you update all sorters python based and all package from spikeinterface:
```
pip install -- upgrade spikeextractors  spiketoolkit spikeorters spikecomparisons spikewidgets
pip install --upgrade spyking-circus tridesclous herdingspikes
```

Note that from tridesclous you should install openl (or numba) for high channel counts otherwise it will be slow.

alexmcclanahan, 2020-03-03:
@samuelgarcia: thanks -- I've tried that, to no avail. It seems as if the issue on my machine is that the various BaseSorter functions are erroring out as they don't see the sorters they point to during the BaseSorter.__init__ call within the individual sorter. I have tried installing each sorter individually on two systems, after installing SpikeInterface with the same errors both times.

I will continue trying to unravel if this is an issue on my end during off hours.

samuelgarcia, 2020-03-03:
Can you do give your ```pip list``` ?

alexmcclanahan, 2020-03-04:
Sure, here it is:
[pip_list.txt](https://github.com/SpikeInterface/spikeinterface/files/4289663/pip_list.txt)

samuelgarcia, 2020-03-09:
For tridesclous evrything looks fine.
What happen when you do tridesclous in a console ?

MingChingChiang, 2020-04-14:
Hi, I also have problems when executing ss.run_tridesclous, 
even I have followed the installation steps from the tridesclous website. 
When I type tdc in my terminal, I could launch the tridesclous GUI. 
So...I'm not sure what happened to this problem.
I'm using macOS 10.14.6, python 3.7.7
Here's my PIP list. 
[PIP_List.txt](https://github.com/SpikeInterface/spikeinterface/files/4473286/PIP_List.txt)

samuelgarcia, 2020-04-14:
Hi @MingChingChiang 
Could you do this 
```sorting_TDC = ss.run_tridesclous(recording, output_folder='tmp_TDC', raise_error=True)```
and copy/paste the error message ?

MingChingChiang, 2020-04-14:
Sure! I got these error messages:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-6-875d53373fb4>", line 2, in <module>
    sorting_TDC = ss.run_tridesclous(recording, output_folder=tmp_TDC, raise_error=True)
  File "/Users/Ming-Ching/Documents/Python_Project/SI/venv/lib/python3.7/site-packages/spikesorters-0.3.0-py3.7.egg/spikesorters/sorterlist.py", line 230, in run_tridesclous
    return run_sorter('tridesclous', *args, **kwargs)
  File "/Users/Ming-Ching/Documents/Python_Project/SI/venv/lib/python3.7/site-packages/spikesorters-0.3.0-py3.7.egg/spikesorters/sorterlist.py", line 79, in run_sorter
    parallel=parallel, verbose=verbose, delete_output_folder=delete_output_folder)
  File "/Users/Ming-Ching/Documents/Python_Project/SI/venv/lib/python3.7/site-packages/spikesorters-0.3.0-py3.7.egg/spikesorters/tridesclous/tridesclous.py", line 53, in __init__
    BaseSorter.__init__(self, **kargs)
  File "/Users/Ming-Ching/Documents/Python_Project/SI/venv/lib/python3.7/site-packages/spikesorters-0.3.0-py3.7.egg/spikesorters/basesorter.py", line 48, in __init__
    Please install it with:  \n{} """.format(self.sorter_name, self.installation_mesg)
AssertionError: This sorter tridesclous is not installed.
        Please install it with:  
       >>> pip install https://github.com/tridesclous/tridesclous/archive/master.zip
    More information on tridesclous at:
      * https://github.com/tridesclous/tridesclous
      * https://tridesclous.readthedocs.io

samuelgarcia, 2020-04-14:
what happen if you do
```import tridesclous```
in ipython ?

MingChingChiang, 2020-04-15:
It returns: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-24-6bc6153f136e>", line 1, in <module>
    import tridesclous
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'tridesclous'
I'll try reinstalling my iPython into my virtualenv again to see whether I can fix the problem...

samuelgarcia, 2020-04-15:
Very strange that you have it in the pip list but not being able to import.
Which method did you used for installing tridesclous ```pip install tridesclous```?

MingChingChiang, 2020-04-15:
Yes, I used pip install tridesclous for installation.
The other strange thing is that after I created a new environment with python==3.7.6 and reinstall everything, including spikeinterface and tridesclous, everything just worked fine.
Not sure whether the python version has such strong impacts or not...
But at least I learned that avoid using python 3.7.7 or higher version for spikeinterface might be a better option.
 
> Very strange that you have it in the pip list but not being able to import.
> Which method did you used for installing tridesclous `pip install tridesclous`?

samuelgarcia, 2020-04-15:
Thank you.
This is indeed very important feedback.
I will check.

mr33678, 2021-05-13:
So....this issue was never solved nor closed, right? Any developments? I have a similar problem...Thx for any help.

ShubhankarSaha, 2021-05-15:
> @mhhennig, super excited to experiment with SpikeInterface; I think it will yield very useful information for the field, so, thanks!
>
> I've experienced similar issues to those outlined by [...]

I have the same issue. available_sorters() returns the list of sorters I see in my spiketutorial environment. However, installed_sorters() returns null. Please respond how to solve this. Thanks for your help.

alejoe91, 2021-05-16:
> So....this issue was never solved nor closed, right? Any developments? I have a similar problem...Thx for any help.

The specific issue for tridesclous should be solved. You can `pip install --upgrade tridesclous` to get version 1.6.1

closed 2021-07-22 by samuelgarcia