SpikeInterface/spikeinterface #51: Temporal structure of recordings

subhacom, 2019-11-29:
Does the `NumpyRecordingExtractor` allow different trials from the same recording session. Should one just concatenate all the trials and keep track of the boundaries separately?

It appears that different formats allow different levels of information - is it possible to add some documentation on the limitations/information loss when converting from one format to another?

mhhennig, 2020-01-15:
Apologies for the delay!
Yes, as it stands different trials would be concatenated and then sorted together, and have to be separated afterwards. None of the extractors does automate this at the moment. Combining separate trials/conditions is something we need to implement.
And yes, we should document what each format provides, thanks for the pointer.

colehurwitz, 2020-01-15:
Apologies for missing this issue! I think there are two ways to handle this currently in SpikeInterface:

1. **Concatenate the trials and then put them into a single `NumpyRecordingExtractor`.** You can then track the separate trials by manually using the `add_epoch(self, epoch_name, start_frame, end_frame)` function of the extractor. You can retrieve a `SubRecordingExtractor` for each specified epoch with `get_epoch` or get the info with `get_epoch_info`.

2. **Instantiate a `NumpyRecordingExtractor` for each trial separately and make a `MultiRecordingTimeExtractor` with them (using `concatenate_recordings_by_time(recordings, epoch_names=None)`).** By specifying the epoch_names for each `NumpyRecordingExtractor`,  you can again retrieve a `SubRecordingExtractor` for each specified epoch with `get_epoch` or get the info with `get_epoch_info',

samuelgarcia, 2020-02-25:
Can we close this ?

mhhennig, 2020-04-19:
We should make some documentation for this case. Let's leave it open, I have assigned myself as I have some example code for this.

closed 2021-05-18 by alejoe91