SpikeInterface/spikeinterface #55: Getting stuck on the toy example of sorting

chengchichu, 2020-02-13:
I was running the mountainsort4 with python version3.6 with anaconda environment. When I went through the toy example provided on 

https://spikeinterface.readthedocs.io/en/latest/getting_started/plot_getting_started.html

I got stuck on the sorting step. It return the message
```

  File "/home/chang/anaconda3/envs/myenv1/lib/python3.6/site-packages/spikesorters/mountainsort4/mountainsort4.py", line 110, in _run
    verbose=self.verbose

TypeError: mountainsort4() got an unexpected keyword argument 'verbose'
```

and can't proceed 

Do you know how to solve this? I am a user of previous version ml3 and would like to use some new feature from spikeinterface to extract the spike waveform.

samuelgarcia, 2020-02-25:
Could you check with the actual master ?

alejoe91, 2020-12-04:
this is fixed eith later ms_m4alg releases

closed 2020-12-04 by alejoe91