SpikeInterface/spikeinterface #56: Many examples do not build because of many changes in all packages.

samuelgarcia, 2020-02-20:
@colehurwitz @alejoe91

alexgonzl, 2020-02-21:
I can second this. I was running into QT issues when loading spike interface in Jupyter after upgrading to spyking-circus 0.9.2.

alejoe91, 2020-02-21:
Yes everything is broken here! :) @samuelgarcia I suggest next week we merge what we can, make a release of all packages and update the docs and requirements of spikeinterface

samuelgarcia, 2020-02-24:
Not so much are borken but there is a subcopy of all pacakge with ```git clone``` that hack the version installed (but do not update locally) This make everything broken!!!!! I will fix this mechanism.

alejoe91, 2020-02-24:
The git clone should not install the packages though, right? That is done only to pulled the source code to autogen the API docs

samuelgarcia, 2020-02-24:
No because
```
sys.path.insert(0, os.path.abspath('sources/spikeextractors/'))
sys.path.insert(0, os.path.abspath('sources/spiketoolkit/'))
sys.path.insert(0, os.path.abspath('sources/spikesorters/'))
sys.path.insert(0, os.path.abspath('sources/spikecomparison/'))
sys.path.insert(0, os.path.abspath('sources/spikewidgets/'))
```
replace the installed pacakge.
I will do this only for readthedoc, there must be an environement variable that allow to check it.

alejoe91, 2020-02-24:
Ah that's why! Tricky!

samuelgarcia, 2020-02-24:
Not so.
I thin this is good to build doc local with another copy of package because we can make mistake with branch.
I add git fecth for local mahcine and this should be OK have a look in conf.py.

closed 2020-02-25 by samuelgarcia