SpikeInterface/spikeinterface #59: Getting OSError: [Errno 8] Exec format error when using ss.run_klusta

MingChingChiang, 2020-04-10:
Hi, 
I'm not sure whether here is the right place to report my question.
When I try to execute the ss.run_klusta on my data. 
I got OSError: [Errno 8] Exec format error. 
I'm using Pycharm 2020.1 Pro. Ed. on macOS Mojave for my data analysis. 
My current python version is 3.7.7
I attached the full error messages below.
Hope someone knows how to fix the problem.
And I have no problem running ss.run_mountainsort4 or ss.run_herdingspikes.

Best,
Ming-Ching

[image]

MingChingChiang, 2020-04-13:
I found that after I follow the alternative way for spikeinterface development version installation, the OSError is disappeared. Now I can run the Klusta sorting. 
So I suggest for the mac user, maybe not using pip for spikeinterface installation, but follow the alternative way install from source.

MingChingChiang, 2020-04-16:
I got the same problem again when I setting up the environment on different mac.
Not sure what's wrong with my packages...
I attached my PIP LIST...
If someone has ant idea how to solve the problem please let me know
[PIP_List.txt](https://github.com/SpikeInterface/spikeinterface/files/4486420/PIP_List.txt)

mhhennig, 2020-04-19:
Just a guess, maybe script.bat needs to have the executable flag set? @colehurwitz, you could maybe check on your mac...

MingChingChiang, 2020-04-20:
> Just a guess, maybe script.bat needs to have the executable flag set? @colehurwitz, you could maybe check on your mac...

Hi, during the weekend, I have tried several parameters for this problem. 
I found that after I carefully set the channel location and the channel map, 
I can avoid the OSError during the ss.klusta execution. 
However, since I tried to sort the data recorded from electrode bundles instead of silicon probes.
I have hard time setting dummy locations and channel maps.
For now, I just followed the tutorial steps, modified the codes, and it worked fine now...

alejoe91, 2020-09-11:
This should be fixed with latest release

closed 2020-12-04 by alejoe91