SpikeInterface/spikeinterface #60: Getting 'numpy/arrayobject.h' file not found when installing klustakwik2

fvcoen, 2020-04-27:
Hello,

I'm trying to set up an environment for spikeinterface, however I'm running into an error when trying to install klustakwik2. Upon running `pip install klustakwik2` I get `klustakwik2/numerics/cylib/compute_cluster_masks_cy.c:613:10: fatal error: 'numpy/arrayobject.h' file not found`.

Obviously, this is a bit of an issue since most of the examples are using klusta. I've looked into it and it seems that `include_dirs=[numpy.get_include()]` in `setup.py` gets ignored somehow. I've tried a few things but nothing seems to work... Any suggestions?

By the way, I'm running python 3.7.7 on MacOS and I'm installing the package in a virtual environment (venv).

Here is the full error message:

```
(SpikeInterface) macbook-pro-4:Fresh fvcoen$ pip install klustakwik2
    [...]
    running build_ext
    building 'klustakwik2.numerics.cylib.compute_cluster_masks_cy' extension
    creating build/temp.macosx-10.14-x86_64-3.7
    creating build/temp.macosx-10.14-x86_64-3.7/klustakwik2
    creating build/temp.macosx-10.14-x86_64-3.7/klustakwik2/numerics
    creating build/temp.macosx-10.14-x86_64-3.7/klustakwik2/numerics/cylib
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/Users/fvcoen/Code/Fresh/include -I/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c klustakwik2/numerics/cylib/compute_cluster_masks_cy.c -o build/temp.macosx-10.14-x86_64-3.7/klustakwik2/numerics/cylib/compute_cluster_masks_cy.o
    klustakwik2/numerics/cylib/compute_cluster_masks_cy.c:613:10: fatal error: 'numpy/arrayobject.h' file not found
    #include "numpy/arrayobject.h"
             ^~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    running install
    running build
    running build_py
    running egg_info
    writing klustakwik2.egg-info/PKG-INFO
    writing dependency_links to klustakwik2.egg-info/dependency_links.txt
    writing entry points to klustakwik2.egg-info/entry_points.txt
    writing top-level names to klustakwik2.egg-info/top_level.txt
    reading manifest file 'klustakwik2.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.pyxbld' under directory 'klustakwik2/numerics/cylib'
    warning: no previously-included files matching '__pycache__' found under directory '*'
    warning: no previously-included files matching '*.py[co]' found under directory '*'
    writing manifest file 'klustakwik2.egg-info/SOURCES.txt'
    running build_ext
    building 'klustakwik2.numerics.cylib.compute_cluster_masks_cy' extension
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/Users/fvcoen/Code/Fresh/include -I/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c klustakwik2/numerics/cylib/compute_cluster_masks_cy.c -o build/temp.macosx-10.14-x86_64-3.7/klustakwik2/numerics/cylib/compute_cluster_masks_cy.o
    klustakwik2/numerics/cylib/compute_cluster_masks_cy.c:613:10: fatal error: 'numpy/arrayobject.h' file not found
    #include "numpy/arrayobject.h"
             ^~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    error: command 'clang' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /Users/fvcoen/Code/Fresh/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/36/95fz5r4d181c_hfh6014lvnr0000gn/T/pip-install-jnxjfbjk/klustakwik2/setup.py'"'"'; __file__='"'"'/private/var/folders/36/95fz5r4d181c_hfh6014lvnr0000gn/T/pip-install-jnxjfbjk/klustakwik2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/36/95fz5r4d181c_hfh6014lvnr0000gn/T/pip-record-srjiswrq/install-record.txt --single-version-externally-managed --compile --install-headers /Users/fvcoen/Code/Fresh/include/site/python3.7/klustakwik2 Check the logs for full command output.
```

Thanks in advance for your help!

samuelgarcia, 2020-04-27:
Hi.
I think you should report this to klusta repo.

fvcoen, 2020-04-28:
Will do, thank you for your answer :)

closed 2020-04-28 by fvcoen