SpikeInterface/spikeinterface #66: Working with .mcd files

abhijithkrishnan93, 2020-07-18:
Is it possible to open .mcd (Multi Channel Systems)or its raw binary variant (converted using MC Data Tool) in spikeinterface.
I saw the mcsh5 extractor. What are the steps to use the .mcd or converted raw file using this extractor?

samuelgarcia, 2020-07-18:
Hi. 
I started a branch on neo for the msrd format : the new native format of multi channel system.
MCD is the old one I think.
The branch is here
https://github.com/samuelgarcia/python-neo/tree/format_msrd

Could you try it ? If it works I could then merge it in master and create a wrapper in spikeextractor.

abhijithkrishnan93, 2020-07-18:
> Hi.
> I started a branch on neo for the msrd format : the new native format of multi channel system.
> MCD is the old one I think.
> The branch is here [...]

Hi,
My lab still uses the old .mcd format. Is there some way to convert. Also I am not familiar with neo. Is the documentation starightforward?

samuelgarcia, 2020-07-20:
I see.
the mcd file don't have public specification.

The mcs raw binary is in fact a simple binary format with a custom small header.
If it helps I can write a wrapper for the neo.RawMCSIO in spikeextractors.

Leave this open.

abhijithkrishnan93, 2020-07-20:
I was able to use it with a wrapper based on other neo extractors.

class mcsrawRecordingExtractor(NeoBaseRecordingExtractor):
     extractor_name='mcsrawRecoding'
     mode='folder'
     NeoRawIOClass='RawMCSRawIO'

and then,

recording=mcsrawRecordingExtractor(filename='Data0349/Data0349.raw')

samuelgarcia, 2020-07-20:
Exactly.
Note that mode is 'file' in that case.

Could you make a PR ?

abhijithkrishnan93, 2020-07-20:
I have made a PR in spikeextractors. I will close the issue now I think.

abhijithkrishnan93, 2020-11-28:
import spikeextractors as se
recording=se.MCSRawRecordingExtractor(filename='Data0552/Data0552.raw')

Gives following error...

AttributeError                            Traceback (most recent call last)
<ipython-input-2-908b8c60a822> in <module>
----> 1 recording=se.MCSRawRecordingExtractor(filename='Data0552/Data0552.raw')

~/miniconda3/envs/spikeinterface/lib/python3.6/site-packages/spikeextractors/extractors/neoextractors/neobaseextractor.py in __init__(self, **kargs)
     55         # TODO propose a meachanisim to select the appropriate channel groups
     56         # in neo one channel group have the same dtype/sampling_rate/group_id
---> 57         num_chan_group = len(self.neo_reader.get_group_channel_indexes())
     58         assert num_chan_group == 1, 'This file have several channel groups spikeextractors support only one groups'
     59 

AttributeError: 'RawMCSRawIO' object has no attribute 'get_group_channel_indexes'

it appears mcsrawio has no get_group_channel_indexes attribute. Was something changed in the neobaseextractor.py as it was working 1-2 months back

alejoe91, 2020-11-28:
@samuelgarcia  has something changed in `neo.rawio`?

samuelgarcia, 2020-11-30:
Oups yes.
`get_group_signal_channel_indexes` have replace `get_group_channel_indexes` in 0.9 release one week ago.
We need to patch.

closed 2021-07-22 by samuelgarcia