SpikeInterface/spikeinterface #72: Slow curation and export

guitchounts, 2020-08-24:
I'm having trouble with curation and export to Phy steps taking a really long time. I'm sorting a 45-minute recording with 32 channels using Mountainsort4. The sorting takes ~11 minutes. Curating based on firing rates took several seconds, but curating based on SNR took 6800 seconds and the export to Phy step is taking several hours. I'm running this on a local machine (16-Gb RAM i7 MacBook Pro). Any idea how to improve this performance? (I'm also running the same code on a cluster, and those jobs don't finish after running for 12 hours...) 

[image]

mhhennig, 2020-08-25:
My guess is that the bottleneck is likely the function collecting spike cut-outs (both used in curation and phy export). What format is your raw data file? Also, do you apply any filters on the recording extractor?

guitchounts, 2020-08-25:
The raw data was acquired with Open Ephys GUI, stored in their binary format. I'm applying bandpass filtering and common mode rejection on the recording extractor, and using the cmr'd extractor for sorting, curation, phy, etc. 

Is it possible to speed up the cut-out step somehow? or at least to save the cutouts in one step so they don't have to be re-computed several times?

samuelgarcia, 2020-08-25:
in spike interface there is a cache system to avoid to recompute the preprocessing chain and extractor several times waveforms snippet (very long) @alejoe91 do we have a notebook example for the cache workflow ?

alejoe91, 2020-08-25:
Hi @guitchounts 

there are a few things you can do: 

1. Cache the `recording_cmr` (VERY IMPORTANT): this dumps a binary version of your recording (in your case bandpass filtered + cmr) to a dat binary file. You can simply do: `recording_cmr = se.CacheRecordingExtractor(recording_cmr, 'recording_cmr.dat')`. The process will take a few minutes, but it'll speed up everything else (otherwise each single waveforms is filtered separately later!)

2. Use multiple jobs: when computing the SNR (and therefore the waveforms in the background) you can use the `n_jobs=8` (or 4-16...). This will compute waveforms and templates in parallel. Same holds for the export_to_phy. 

3. Note that the export to phy is slower because ALL waveforms need to be extracted to compute amplitudes and PC scores. If you are not interested in those (you'll still be able to look at templates, similarity, and cross-correlograms), you can try the to set the `compute_amplitudes=False` and `compute_pc_features=False` when calling the `export_to_phy`

Hope this helps! :)

alejoe91, 2020-12-04:
Potprocessing efficiency issues are solved with latest releases of spiketoolkit

closed 2020-12-04 by alejoe91