SpikeInterface/spikeinterface #75: Error with using neo based spike extractors

audrocks17, 2020-09-06:
Hello. I have been working to get spikeinterface up and running, but am running into an issue with running the NeuralyxRecordingExtractor. To install the packages, I cloned everything from each repository, ran the setup.py in every folder. Initially when I attempted to run NeuralyxRecordingExtractor I got an error that I needed to install neo, which was easily solved by doing so. But I can't seem to get past the following error: 

FileNotFoundError                         Traceback (most recent call last)
<ipython-input-9-4ba579a1a640> in <module>
----> 1 recording = se.NeuralynxRecordingExtractor(file_name='C:/Users/abranch6/Desktop/spikewhatever/data/883-11_RawData_300s.nrd', error_checking=False)

c:\programdata\anaconda3\lib\site-packages\spikeextractors-0.9.1-py3.7.egg\spikeextractors\extractors\neoextractors\neobaseextractor.py in __init__(self, **kargs)
     51     def __init__(self, **kargs):
     52         RecordingExtractor.__init__(self)
---> 53         _NeoBaseExtractor.__init__(self, **kargs)
     54 
     55         # TODO propose a meachanisim to select the appropriate channel groups

c:\programdata\anaconda3\lib\site-packages\spikeextractors-0.9.1-py3.7.egg\spikeextractors\extractors\neoextractors\neobaseextractor.py in __init__(self, block_index, seg_index, **kargs)
     27         neoIOclass = eval('neo.rawio.' + self.NeoRawIOClass)
     28         self.neo_reader = neoIOclass(**kargs)
---> 29         self.neo_reader.parse_header()
     30 
     31         if block_index is None:

c:\programdata\anaconda3\lib\site-packages\neo\rawio\baserawio.py in parse_header(self)
    149 
    150         """
--> 151         self._parse_header()
    152         self._group_signal_channel_characteristics()
    153 

c:\programdata\anaconda3\lib\site-packages\neo\rawio\neuralynxrawio.py in _parse_header(self)
     83         event_annotations = []
     84 
---> 85         for filename in sorted(os.listdir(self.dirname)):
     86             filename = os.path.join(self.dirname, filename)
     87 

FileNotFoundError: [WinError 3] The system cannot find the path specified: ''

I have made sure that everything is on my python path, and checked all the permissions on all the folders, and can't think of what else to do. Do you have any suggestions of what the problem might be? Thank you in advance!!

alejoe91, 2020-09-07:
Hi @audrocks17 

Can you try passing the folder containing the Neuralynx files? also you might need to add an `r` before your path as windows likes raw strings... `file_name=r'C:/Users/abranch6/Desktop/spikewhatever/data/883-11_RawData_300s.nrd'`

@samuelgarcia other ideas?

audrocks17, 2020-09-07:
Thanks so much for the suggestion! If I understand correctly, I tried to define the file name by defining the current working directory and the file name in the first part of the script:

import sys
import os
sys.path.append(r'C:/ProgramData/Anaconda3/Lib/site-packages')
path = os.getcwd()
print(path)
import spikeextractors as se
import spikewidgets as sw
import spiketoolkit as st
import spikesorters as ss
import spikecomparison as sc
import matplotlib.pylab as plt
import numpy as np
%matplotlib notebook
import spiketoolkit as st
basefilename = '883-11_RawData_300s.nrd'
C:\Users\abranch6\spikewhatever\data

And then ran: recording = se.NeuralynxRecordingExtractor(file_name= basefilename, error_checking=False)

But I still get the exact same error as before.

alejoe91, 2020-09-07:
> Thanks so much for the suggestion! If I understand correctly, I tried to define the file name by defining the current working directory and the file name in the first part of the script:
>
> import sys [...]

Hi,

you don't need the `sys.path.append` stuff if you're using a proper environment :)

Can you try (after the imports):

```
basefilename = r'C:\Users\abranch6\spikewhatever\data' # assuming your nrd file is in this folder

recording = se.NeuralynxRecordingExtractor(file_name=basefilename, error_checking=False)
```
?

audrocks17, 2020-09-07:
Ha, yeah, that was left over from troubleshooting. Okay, I tried what you suggested just now and still get the same error.

alejoe91, 2020-09-07:
Then we have to ask @samuelgarcia :)

audrocks17, 2020-09-07:
Okay, thank you for the suggestions!

samuelgarcia, 2020-09-07:
Hi.

You have to use this:

```python
rec = NeuralynxRecordingExtractor(dirname='r'C:\Users\abranch6\spikewhatever\data')
```

this is because neuralynx is directory based and not file based.

audrocks17, 2020-09-07:
Thanks! I am trying to implement your suggestion, and I do get past the original error, but to a new error, so maybe I am misunderstanding. If, after importing everything, I then run: 

recording = se.NeuralynxRecordingExtractor(dirname=r'C:\Users\abranch6\spikewhatever\data')

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-24-37250bf7e342> in <module>
----> 1 recording = se.NeuralynxRecordingExtractor(dirname=r'C:\Users\abranch6\spikewhatever\data')

c:\programdata\anaconda3\lib\site-packages\spikeextractors-0.9.1-py3.7.egg\spikeextractors\extractors\neoextractors\neobaseextractor.py in __init__(self, **kargs)
     51     def __init__(self, **kargs):
     52         RecordingExtractor.__init__(self)
---> 53         _NeoBaseExtractor.__init__(self, **kargs)
     54 
     55         # TODO propose a meachanisim to select the appropriate channel groups

c:\programdata\anaconda3\lib\site-packages\spikeextractors-0.9.1-py3.7.egg\spikeextractors\extractors\neoextractors\neobaseextractor.py in __init__(self, block_index, seg_index, **kargs)
     27         neoIOclass = eval('neo.rawio.' + self.NeoRawIOClass)
     28         self.neo_reader = neoIOclass(**kargs)
---> 29         self.neo_reader.parse_header()
     30 
     31         if block_index is None:

c:\programdata\anaconda3\lib\site-packages\neo\rawio\baserawio.py in parse_header(self)
    149 
    150         """
--> 151         self._parse_header()
    152         self._group_signal_channel_characteristics()
    153 

c:\programdata\anaconda3\lib\site-packages\neo\rawio\neuralynxrawio.py in _parse_header(self)
    223             # case  NO ncs but HAVE nev or nse
    224             self._timestamp_limits = [(ts0, ts1)]
--> 225             self._seg_t_starts = [ts0 / 1e6]
    226             self._seg_t_stops = [ts1 / 1e6]
    227             self.global_t_start = ts0 / 1e6

TypeError: unsupported operand type(s) for /: 'NoneType' and 'float'

samuelgarcia, 2020-09-07:
Can you list the folder of r'C:\Users\abranch6\spikewhatever\data' ?

audrocks17, 2020-09-07:
Sorry, I'm not sure I understand, but the only file that is in the folder is the data file 883-11_RawData_300s.nrd

samuelgarcia, 2020-09-07:
Neuralynx is supposed to have a folder with thoses file extentions  : 'nse', 'ncs', 'nev', 'ntt'

NCS contains signals for one channel
NEV contains events
NSE contains spikes and waveforms for mono electrodes
NTT contains spikes and waveforms for tetrodes

audrocks17, 2020-09-07:
Ah, I see. I am trying to run this from an .nrd file, is that what's causing the problem?

samuelgarcia, 2020-09-07:
I knwon what is nrd files ?
Is it a new file format from neuralynx ?

audrocks17, 2020-09-07:
I just realized that I forgot to clone the repository that contains the new script that my labmate wrote work with nrd files, sorry, stupid mistake. Thank you so much for your help! I really appreciate it!

audrocks17, 2020-09-07:
the nrd file is the unprocessed output from Cheetah, which then gets processed into the file types you mentioned. We are hoping to run spikeinterface from the raw file directly.

alejoe91, 2020-12-04:
@samuelgarcia maybe nrd files should be supported by the NEO wrapper?

samuelgarcia, 2020-12-04:
I would strongly encourage to write a neo issur about this nrd files.

Maybe @PeterNSteinmetz and @JuliaSprenger will be able to help you.

PeterNSteinmetz, 2020-12-04:
.nrd is a rather different format (see https://support.neuralynx.com/hc/en-us/articles/360040444811-TechTip-Neuralynx-Data-File-Formats) . It is not presently supported by the NeuralynxRawIO and would have to be added (likely a reasonable amount of work). Agreed that should be an issue in the Neo project to request that.

closed 2021-05-18 by alejoe91

valejandracch, 2022-06-28:
> I just realized that I forgot to clone the repository that contains the new script that my labmate wrote work with nrd files, sorry, stupid mistake. Thank you so much for your help! I really appreciate it!

@audrocks17 do you think that you can share that script with me? im trying to work with nrd files also with spikeinterfase and i have problems as well.
Thanks!

audrocks17, 2022-06-28:
Sure, the script is located at:
https://github.com/KnierimLab/spikeextractors/blob/master/spikeextractors/extractors/neoextractors/neuralynxextractor.py