SpikeInterface/spikeinterface #78: AttributeError: module 'spikeinterface.extractors' has no attribute

Idavr, 2020-10-15:
Hi there!

Let me first say that I really appreciate you creating a python based framework to work with electrophysiology as I have had an easier time getting into python than MATLAB in the early stages of my phd. 

I am trying to get spikeinterface to work for the very first time, and created a new environment and did the pip install spikeinterface like described. Wanting to try an example code snippet from your paper on my own data, I get an AttributeError both in iPython and Jupyter Notebook. The code snippet I am testing is

[image]

Resulting in an AttributeError: module 'spikeinterface.extractors' has no attribute 'MyFormatRecordingExtractor'.

Also, when trying to run one of the examples here on the GitHub site on a different computer, I also get an AttributeError from the same module, but this time finding that spikeinterface.extractors has no attribute "example_dataset".

What am I doing wrong? Appreciate whatever input you can provide.

samuelgarcia, 2020-10-15:
Welcome.
MyFormatRecordingExtractor is a generic name that do not exists.
You should the one for your data format like `OpenEphysRecordingExtractor`.

Idavr, 2020-10-23:
Well... That was an embarrassing introduction. Thank you for setting me straight, I found the video tutorial and will work more on this before asking again :) Appreciate the quick feedback, and sorry for responding so late!

closed 2020-10-23 by alejoe91