SpikeInterface/spikeinterface #79: Issue With running

ohnopleasehelpme, 2020-10-16:
Hello

I had this working over the summer - now it stopped working again: here is the code followed by the error:

import spikeinterface.extractors as se
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw
import matplotlib.pylab as plt
import numpy as np
import neo

**CODE**
reader = neo.NeuralynxIO(dirname='/Users/.../PycharmProjects/SpikeGUI/data')
print(reader)
print(reader.segment_count(0))

rec_list = [se.NeuralynxRecordingExtractor(dirname='/.../PycharmProjects/SpikeGUI/data/', seg_index=i)
            for i in range(reader.segment_count(0))]
rec = se.MultiRecordingTimeExtractor(rec_list)

channel_ids = rec.get_channel_ids()
fs = rec.get_sampling_frequency()
num_chan = rec.get_num_channels()

print('Channel ids:', channel_ids)
print('Sampling frequency:', fs)
print('Number of channels:', num_chan)

recording_prb = rec.load_probe_file('custom_probe.prb')
print('Channels after loading the probe file:', recording_prb.get_channel_ids())
print('Channel groups after loading the probe file:', recording_prb.get_channel_groups())

# w_elec = sw.plot_electrode_geometry(recording_prb, markersize=50)

print(ss.installed_sorters())
print(ss.get_default_params('mountainsort4'))

rec_f = st.preprocessing.bandpass_filter(recording_prb, freq_min=300, freq_max=6000)
# w = sw.plot_timeseries(rec_f, trange=[0, 2])

sorting_KL = ss.run_klusta(recording=rec_f)
st.postprocessing.export_to_phy(rec_f, sorting_KL, output_folder='phy_K')

**ERROR**
runfile('/Users/hemalsemwal/PycharmProjects/SpikeGUI/nueralynx_extract.py', wdir='/Users/hemalsemwal/PycharmProjects/SpikeGUI')
/Users/hemalsemwal/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/elephant/pandas_bridge.py:22: DeprecationWarning: pandas_bridge module will be removed in Elephant v0.8.x
  DeprecationWarning)
11:27:52 [I] klustakwik KlustaKwik2 version 0.2.6
Backend MacOSX is interactive backend. Turning interactive mode on.
Traceback (most recent call last):
  File "/Users/hemalsemwal/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-6a042b03c901>", line 1, in <module>
    runfile('/Users/hemalsemwal/PycharmProjects/SpikeGUI/nueralynx_extract.py', wdir='/Users/hemalsemwal/PycharmProjects/SpikeGUI')
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/Users/hemalsemwal/PycharmProjects/SpikeGUI/nueralynx_extract.py", line 11, in <module>
    reader = neo.NeuralynxIO(dirname='/Users/hemalsemwal/PycharmProjects/SpikeGUI/data')
  File "/Users/hemalsemwal/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/neo/io/neuralynxio.py", line 36, in __init__
    BaseFromRaw.__init__(self, dirname)
  File "/Users/hemalsemwal/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/neo/io/basefromrawio.py", line 81, in __init__
    self.parse_header()
  File "/Users/hemalsemwal/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/neo/rawio/baserawio.py", line 151, in parse_header
    self._parse_header()
  File "/Users/hemalsemwal/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/neo/rawio/neuralynxrawio.py", line 98, in _parse_header
    info = read_txt_header(filename)
  File "/Users/hemalsemwal/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/neo/rawio/neuralynxrawio.py", line 682, in read_txt_header
    dt1 = re.search(datetime1_regex, txt_header).groupdict()
AttributeError: 'NoneType' object has no attribute 'groupdict'

alejoe91, 2020-12-04:
Hey @ohnopleasehelpme 

There have been some updates to the neo packages. @samuelgarcia do you know what's wrong here?

samuelgarcia, 2020-12-04:
I also need to update the neo layer here.
I will do it very soon.

closed 2021-05-18 by alejoe91