SpikeInterface/spikeinterface #84: Installing 0.9.9 spikeinterface installs the wrong version of spikesorters (and other sub packages)

davidparks21, 2020-11-17:
Installing 0.9.9 of `spikeinterface` causes the installation of the latest version of `spikesorters`. `spikesorters` is not fully backwards compatible. In particular we get the error: `AttributeError: module 'spikeinterface.sorters' has no attribute 'installed_sorter_list'` with the newer version of spikesorters. 

We install `spikeinterface` as an umbrella package, so when we encountered this issue we downgraded spikeinterface to install `spikeinterface=0.9.9`. However this had the unfortunate consequence of installing `spikesorters=0.4.2` rather than `spikesorters=0.3.3` as it was when we had originally installed `spikeinterface=0.9.9`. 

If backwards compatibility isn't guaranteed then `requirements.txt` should contain an upper limit to the install version of the sub packages. Instead of just `spikesorters>=0.3.2`, it would make more sense to have `spikesorters>=0.3.2,<=0.3.3` to ensure installs of 0.9.9 are compatible with code written to 0.9.9.

alejoe91, 2020-11-17:
Thanks for bringing up this issue, David. 

You are right, we should have used also a < in the requirements. However, we made major updates to the API with version 0.10.0 and I strongly recommend to upgrade to that version.
From that version on, we are not planning to make major changes to the API. 

I'll try to fix the github version to include the version check in the requirements, but I'm afraid that the pip installable version could not be changed.

Cheers
Alessio

davidparks21, 2020-11-17:
Thanks for the quick response Alessio, we're good now, we've got the individual version limits on the packages, and hopefully this will document it for anyone else that comes looking. We'll make the update to 0.10.0 as soon as we get a free moment to clean up the few minor changes.

samuelgarcia, 2020-11-18:
Putting a "<" in requirements in the master branch is a pain for installation.
I would not recommand this.

I think that release of spikeinterface must have "==" for all,package.
In short:
 * spikeinterface with pip is a snapshop of version for the familly package.
 * spikeinterface with github ensure some compatibilty with ">" but would be more liberal with dev version.

alejoe91, 2020-12-10:
This is fixed now: https://github.com/SpikeInterface/spikeinterface/blob/0.11.0/requirements.txt

The release versions will have `==` and the master version has `>=` to allow for updates

closed 2020-12-10 by alejoe91