SpikeInterface/spikeinterface #89: Spike validation

Kasia-Bz, 2021-02-12:
Hello, I am trying to use the validation toolkit from Spike interface. I use Kilosort for the spike sorting and then manual  curation with phy. I was wondering if I use the validation toolkit, does it check the clusters which were marked as ‘good’ by Kilosort, or does it use the ‘good’ clusters which were marked manually in phy, or both?

alejoe91, 2021-02-12:
Hi @Kasia-Bz 

Do you run Kilosort through SpikeInterface? In that case, you could already exclude the bad units from the returned `SortingExtractor` by using the `keep_good_only` argument to `True`.

If you run Phy directly on the Kilosort output you can mark as noise all the units you want to exclude for further analysis. 
Then, you can load the Phy data as follows:

```
sorting_phy = se.PhySortingExtractor('path-to-phy-folder', exclude_cluster_groups=['noise'])
```

If you want to further remove bad KSLabels, you can do the following:
```
good_ks_units = []
for u in sorting_phy.get_unit_ids():
    if 'KSLabel' in self.get_unit_property_names(u):
        if sorting_phy.get_unit_property(u, 'KSLabel') == 'good':
            good_ks_units.append(u)
sorting_phy_ks_good = se.SubSortingExtractor(sorting_phy, unit_ids=good_ks_units)
```

Does this answer your question?

Kasia-Bz, 2021-02-12:
Yes, that's exactly what I wanted to know :) Thank you.

Kasia-Bz, 2021-02-17:
Hi @alejoe91 
I want to follow up again on the question I initially asked. I am using Kilosort outside of the spike interface. As an output I get file: "cluster_info" in which the KS_Label contains the values of good units sorted by Kilosort3. The 'good' clusters, which I assigned are in the "group" section of that file. I also have another file: cluster_KSLabel, but there I only have the units marked as good by Kilosort. Does it mean that if I would like to run validation on the manually sorted clusters, than I would need to set: 
```
good_ks_units = []
for u in sorting_phy.get_unit_ids():
    if 'KSLabel' in self.get_unit_property_names(u):
        if sorting_phy.get_unit_property(u, 'group') == 'good':
            good_ks_units.append(u)
sorting_phy_ks_good = se.SubSortingExtractor(sorting_phy, unit_ids=good_ks_units)
```

alejoe91, 2021-02-17:
Hi!

Yes I believe that is correct as both properties should be available. You can test it. Let me know if it works!

Kasia-Bz, 2021-02-18:
Hi @alejoe91 ,
I tried to run: 

```
import spikeextractors as se
import spiketoolkit as st
recording = se.PhyRecordingExtractor('/media/kasia/K/Neuropixeldata/LS_k_9/110221/kilosort3/')
sorting_phy = se.PhySortingExtractor('/media/kasia/K/Neuropixeldata/LS_k_9/110221/kilosort3/')
good_ks_units = []
for u in sorting_phy.get_unit_ids():
    if 'KSLabel' in self.get_unit_property_names(u):
        if sorting_phy.get_unit_property(u, 'group') == 'good':
            good_ks_units.append(u)
sorting_phy_ks_good = se.SubSortingExtractor(sorting_phy, unit_ids=good_ks_units)
firing_rates = st.validation.compute_firing_rates(sorting_phy_ks_good, duration_in_frames=recording.get_num_frames())
isi_violations = st.validation.compute_isi_violations(sorting_phy_ks_good, duration_in_frames=recording.get_num_frames(), isi_threshold=0.0015)
l_ratio = st.validation.compute_l_ratios(sorting_phy_ks_good, recording, num_channels_to_compare=13)
isolation_distance = st.validation.compute_isolation_distances(sorting_phy_ks_good, recording, num_channels_to_compare=13)
print(isi_violations)
print(l_ratio)
print(isolation_distance)
```

Unfortunately this code does not work and I get an error: 

```
Traceback (most recent call last):
  File "/media/kasia/K/validation2.py", line 17, in <module>
    if 'KSLabel' in self.get_unit_property_names(u):
NameError: name 'self' is not defined
```
What could be causing it? did I forget to import something?

alejoe91, 2021-02-18:
Hi,

As the error says you have a `self`, but it should be `sorting_phy` (line 7 in your script)

Kasia-Bz, 2021-02-18:
Indeed, but even after removing it I still get an error: 
```
Traceback (most recent call last):
  File "/media/kasia/K/validation2.py", line 18, in <module>
    if 'KSLabel' in get_unit_property_names(u):
NameError: name 'get_unit_property_names' is not defined
```

alejoe91, 2021-02-18:
You need `sorting_phy.get_unit_property_names(u)`

Kasia-Bz, 2021-02-19:
I was able to run the code, but when trying to calculate l_ratio I get an error: 
```
TypeError                                 Traceback (most recent call last)
<ipython-input-23-e6dddba96351> in <module>
----> 1 l_ratio = st.validation.compute_l_ratios(sorting_phy_ks_good, recording, num_channels_to_compare=13)
/opt/conda/envs/fklab/lib/python3.7/site-packages/spiketoolkit/validation/quality_metrics.py in compute_l_ratios(sorting, recording, num_channels_to_compare, max_spikes_per_cluster, unit_ids, **kwargs)
    756 
    757     l_ratio = LRatio(metric_data=md)
--> 758     l_ratios = l_ratio.compute_metric(num_channels_to_compare, max_spikes_per_cluster, **kwargs)
    759     return l_ratios
    760 
/opt/conda/envs/fklab/lib/python3.7/site-packages/spiketoolkit/validation/quality_metric_classes/l_ratio.py in compute_metric(self, num_channels_to_compare, max_spikes_per_cluster, **kwargs)
     35             seed=seed,
     36             spike_cluster_subset=self._metric_data._unit_indices,
---> 37             verbose=self._metric_data.verbose,
     38         )[1]
     39         l_ratios_list = []
TypeError: calculate_pc_metrics() missing 1 required positional argument: 'channel_locations'
```

Also, when I tried to use the good cluster from manual sorting, by setting: 
```
good_ks_units = []
for u in sorting_phy.get_unit_ids():
    if 'KSLabel' in self.get_unit_property_names(u):
        if sorting_phy.get_unit_property(u, 'group') == 'good':
            good_ks_units.append(u)
sorting_phy_ks_good = se.SubSortingExtractor(sorting_phy, unit_ids=good_ks_units)
``
I get an error: 
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-24-0fb92a14c4a5> in <module>
      2 for u in sorting_phy.get_unit_ids():
      3     if 'KSLabel' in sorting_phy.get_unit_property_names(u):
----> 4         if sorting_phy.get_unit_property(u, 'group') == 'good':
      5             good_ks_units.append(u)
      6 sorting_phy_ks_good = se.SubSortingExtractor(sorting_phy, unit_ids=good_ks_units)
/opt/conda/envs/fklab/lib/python3.7/site-packages/spikeextractors/sortingextractor.py in get_unit_property(self, unit_id, property_name)
    377                         return self._properties[unit_id][property_name]
    378                     else:
--> 379                         raise ValueError(str(property_name) + " has not been added to unit " + str(unit_id))
    380                 else:
    381                     raise ValueError(str(property_name) + " must be a string")
ValueError: group has not been added to unit 0
```

Could you help me with it please?

closed 2021-07-22 by samuelgarcia