SpikeInterface/spikeinterface #91: Error when running Ironclust on Mac

crewl2, 2021-03-10:
After watching the video and following the instructions using spiketutorials, I've run into an error running ironclust.

_RemoteTraceback                          Traceback (most recent call last)
_RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/Users/laurencrew/opt/miniconda3/lib/python3.8/site-packages/joblib/externals/loky/process_executor.py", line 431, in _process_worker
    r = call_item()
  File "/Users/laurencrew/opt/miniconda3/lib/python3.8/site-packages/joblib/externals/loky/process_executor.py", line 285, in __call__
    return self.fn(*self.args, **self.kwargs)
  File "/Users/laurencrew/opt/miniconda3/lib/python3.8/site-packages/joblib/_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "/Users/laurencrew/opt/miniconda3/lib/python3.8/site-packages/joblib/parallel.py", line 262, in __call__
    return [func(*args, **kwargs)
  File "/Users/laurencrew/opt/miniconda3/lib/python3.8/site-packages/joblib/parallel.py", line 262, in <listcomp>
    return [func(*args, **kwargs)
  File "/Users/laurencrew/opt/miniconda3/lib/python3.8/site-packages/spikesorters/ironclust/ironclust.py", line 225, in _run
    raise Exception('ironclust returned a non-zero exit code')
Exception: ironclust returned a non-zero exit code
"""

The above exception was the direct cause of the following exception:

Exception                                 Traceback (most recent call last)
~/opt/miniconda3/lib/python3.8/site-packages/spikesorters/basesorter.py in run(self, raise_error, parallel, n_jobs, joblib_backend)
    159             else:
--> 160                 Parallel(n_jobs=n_jobs, backend=joblib_backend)(
    161                     delayed(self._run)(rec.dump_to_dict(), output_folder)

~/opt/miniconda3/lib/python3.8/site-packages/joblib/parallel.py in __call__(self, iterable)
   1053             with self._backend.retrieval_context():
-> 1054                 self.retrieve()
   1055             # Make sure that we get a last message telling us we are done

~/opt/miniconda3/lib/python3.8/site-packages/joblib/parallel.py in retrieve(self)
    932                 if getattr(self._backend, 'supports_timeout', False):
--> 933                     self._output.extend(job.get(timeout=self.timeout))
    934                 else:

~/opt/miniconda3/lib/python3.8/site-packages/joblib/_parallel_backends.py in wrap_future_result(future, timeout)
    541         try:
--> 542             return future.result(timeout=timeout)
    543         except CfTimeoutError as e:

~/opt/miniconda3/lib/python3.8/concurrent/futures/_base.py in result(self, timeout)
    438             elif self._state == FINISHED:
--> 439                 return self.__get_result()
    440             else:

~/opt/miniconda3/lib/python3.8/concurrent/futures/_base.py in __get_result(self)
    387         if self._exception:
--> 388             raise self._exception
    389         else:

Exception: ironclust returned a non-zero exit code

During handling of the above exception, another exception occurred:

SpikeSortingError                         Traceback (most recent call last)
<ipython-input-13-7f7cb2cdc43d> in <module>
      1 # run spike sorting by group
----> 2 sorting_IC = ss.run_ironclust(recording_cache, 
      3                               output_folder='results_split_ic',
      4                               grouping_property='group', parallel=True, verbose=True)
      5 print(f'IronClust found {len(sorting_IC.get_unit_ids())} units')

~/opt/miniconda3/lib/python3.8/site-packages/spikesorters/sorterlist.py in run_ironclust(*args, **kwargs)
    374         The spike sorted data
    375     """
--> 376     return run_sorter('ironclust', *args, **kwargs)
    377 
    378 

~/opt/miniconda3/lib/python3.8/site-packages/spikesorters/sorterlist.py in run_sorter(sorter_name_or_class, recording, output_folder, delete_output_folder, grouping_property, parallel, verbose, raise_error, n_jobs, joblib_backend, **params)
     88                          verbose=verbose, delete_output_folder=delete_output_folder)
     89     sorter.set_params(**params)
---> 90     sorter.run(raise_error=raise_error, parallel=parallel, n_jobs=n_jobs, joblib_backend=joblib_backend)
     91     sortingextractor = sorter.get_result()
     92 

~/opt/miniconda3/lib/python3.8/site-packages/spikesorters/basesorter.py in run(self, raise_error, parallel, n_jobs, joblib_backend)
    167         except Exception as err:
    168             if raise_error:
--> 169                 raise SpikeSortingError(f"Spike sorting failed: {err}. You can inspect the runtime trace in "
    170                                         f"the {self.sorter_name}.log of the output folder.'")
    171             else:

SpikeSortingError: Spike sorting failed: ironclust returned a non-zero exit code. You can inspect the runtime trace in the ironclust.log of the output folder.'

mhhennig, 2021-03-22:
Could this be the same error as #92? n_jobs passed to `write_recording()`?

alejoe91, 2021-03-22:
No I don't think so. Ironclust failed here but it's started correctly. The other issue is fixed

closed 2021-07-22 by samuelgarcia