SpikeInterface/spikeinterface #93: Temp resources clooging computer

benefron, 2021-03-16:
Hi,
I am having issues with the computer being clogged with temp waveform files.
I have ~25 minutes recordings that I sorted. I am caching the recording and running a multi sorting comparison. 
When reaching the export to phy of the agreement sorting (after curation based on snr, firing rate and is violation) I have around 300gb of temp files. Is there any way to overcome this problem? 
Thanks,
Ben

alejoe91, 2021-03-16:
Hi Ben, can you try to upgrade spiketoolkit? We improved the memory management of the export to Phy function.

benefron, 2021-03-16:
I just upgraded to 0.74. I will see if it is better.
Thanks

benefron, 2021-03-16:
Still very high demand for memory.
Is there something I can do to lower it in the function settings?

alejoe91, 2021-03-16:
As high as before? how many spikes do you have in total? Unfortunately to compute pc scores for all spikes you need to extract all waveforms...

benefron, 2021-03-16:
I didn't compare.
I am running it now on a cluster so I need to allocate a lot of resources and have to wait in queue.
If I serialize it will it require fewer temp files? will it delete what it has finished?

alejoe91, 2021-03-16:
Yes it should clean everything up! What do you mean with serialize? since you want a single phy output, you need to run it on the entire dataset. If you are only interested in templates, waveforms, ISI, crosscorrelograms etc, you could avoid saving pc and amplitudes using the `compute_pc_features=False` and  `compute_amplitudes=True`. This will avoid computing pc and amplitudes and templates by default are computed only with 300 spikes

benefron, 2021-03-16:
I see. But then I will have no PCA right?
OK. I will try to run it on the cluster and hope that resources will be freed later today.
I have 32 channels and 25 minutes recordings so it can be a lot of spikes.
I am running it on n_jobs=10. I thought maybe if I run it on less it can delete what it has finished using.
It only deletes the things in the end right, not while running the analysis?

alejoe91, 2021-03-16:
300 gb sounds too much for that dataset. Unless you have many very noisy clusters that spike all the time..maybe it's safe to add a threshold curator removing spike trains with firing rate higher than 200 or so?

mhhennig, 2021-03-22:
I know this problem :)
Do you know which step precisely causes it? Is it the metrics calculation?

samuelgarcia, 2021-05-17:
Hi.
Could you try with the new API (installation from source) ?

benefron, 2021-05-24:
I am trying but I encountered some problems working with the new API. I opened an issue about it.

closed 2021-07-22 by samuelgarcia