SpikeInterface/spikeinterface #94: blackrock import error

paul-aparicio, 2021-03-16:
Hello, I am having a problem when trying to import a blackrock .ns6 file. I have the latest version of neo installed. The error I get is pasted below. Any help would be greatly appreciated, thank you!

In [19]: filename =r'F:\Data\SignalCheck\TY20210308\TY20210308_signalCheck_morning.ns6'

In [20]: recording = se.BlackrockRecordingExtractor(filename)
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
~\anaconda3\lib\site-packages\spikeextractors-0.9.5-py3.8.egg\spikeextractors\extractors\neoextractors\neobaseextractor.py in __init__(self, block_index, seg_index, **kargs)
     62             # Neo >= 0.9.0
---> 63             channel_indexes_list = self.neo_reader.get_group_signal_channel_indexes()
     64         except AttributeError:

AttributeError: 'BlackrockRawIO' object has no attribute 'get_group_signal_channel_indexes'

During handling of the above exception, another exception occurred:

AttributeError                            Traceback (most recent call last)
<ipython-input-20-ddd3ee20f015> in <module>
----> 1 recording = se.BlackrockRecordingExtractor(filename)

~\anaconda3\lib\site-packages\spikeextractors-0.9.5-py3.8.egg\spikeextractors\extractors\neoextractors\blackrockextractor.py in __init__(self, filename, nsx_to_load, block_index, seg_index, **kwargs)
     34     def __init__(self, filename: PathType, nsx_to_load: Optional[int] = None, block_index: Optional[int] = None,
     35                  seg_index: Optional[int] = None, **kwargs):
---> 36         super().__init__(filename=filename, nsx_to_load=nsx_to_load,
     37                          block_index=block_index, seg_index=seg_index, **kwargs)
     38

~\anaconda3\lib\site-packages\spikeextractors-0.9.5-py3.8.egg\spikeextractors\extractors\neoextractors\neobaseextractor.py in __init__(self, block_index, seg_index, **kargs)
     64         except AttributeError:
     65             # Neo < 0.9.0
---> 66             channel_indexes_list = self.neo_reader.get_group_channel_indexes()
     67         num_chan_group = len(channel_indexes_list)
     68         assert num_chan_group == 1, 'This file have several channel groups spikeextractors support only one groups'

AttributeError: 'BlackrockRawIO' object has no attribute 'get_group_channel_indexes'

alejoe91, 2021-03-17:
Hi! That should be fixed using the latest versions. Can you try to pip install --upgrade spikeextractors and neo? Also I would run an upgrade for all other packages as well (or to the main spikeinterface package, which should upgrade all)

samuelgarcia, 2021-03-17:
Normaly I fixe that already fix that here : https://github.com/SpikeInterface/spikeextractors/pull/592
But it is not merged yet.
Could try to take this banch ?

samuelgarcia, 2021-03-17:
And note that you certainly use an not yet released neo version.
With an released neo version you should not have the problem.

paul-aparicio, 2021-03-18:
Thanks for your responses, I appreciate your help, I am rather new to using python, but I thought I had installed the latest versions? I uninstalled them and reinstalled them using pip. It had a different error, but I did get the code to work once I used a full path to the data file (i.e. including drive name) instead of a relative path.

samuelgarcia, 2021-03-19:
@alejoe91 you need to merge this : https://github.com/SpikeInterface/spikeextractors/pull/592

closed 2021-03-19 by paul-aparicio