SpikeInterface/spikeinterface #102: Problem running spike sorters

paul-aparicio, 2021-04-02:
Hello, 
When I attempt to run spike sorting I receive the error pasted below. I get the same error when I try other matlab based sorters (I have tried kilosort2 and waveclus). Thank you for any help!

In[25]: sorting_IC = ss.run_ironclust(recording_cache, output_folder='results_split_ironclust', grouping_property='group', parallel=True)

exception calling callback for <Future at 0x27ac4bfe9d0 state=finished raised BrokenProcessPool>
joblib.externals.loky.process_executor._RemoteTraceback: 
"""
Traceback (most recent call last):
  File "C:\Users\paula\anaconda3\lib\site-packages\joblib\externals\loky\process_executor.py", line 404, in _process_worker
    call_item = call_queue.get(block=True, timeout=timeout)
  File "C:\Users\paula\anaconda3\lib\multiprocessing\queues.py", line 116, in get
    return _ForkingPickler.loads(res)
AttributeError: 'BlackrockRawIO' object has no attribute '__read_nsx_header_variant_a'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\paula\anaconda3\lib\site-packages\joblib\externals\loky\_base.py", line 625, in _invoke_callbacks
    callback(self)
  File "C:\Users\paula\anaconda3\lib\site-packages\joblib\parallel.py", line 347, in __call__
    self.parallel.dispatch_next()
  File "C:\Users\paula\anaconda3\lib\site-packages\joblib\parallel.py", line 780, in dispatch_next
    if not self.dispatch_one_batch(self._original_iterator):
  File "C:\Users\paula\anaconda3\lib\site-packages\joblib\parallel.py", line 847, in dispatch_one_batch
    self._dispatch(tasks)
  File "C:\Users\paula\anaconda3\lib\site-packages\joblib\parallel.py", line 765, in _dispatch
    job = self._backend.apply_async(batch, callback=cb)
  File "C:\Users\paula\anaconda3\lib\site-packages\joblib\_parallel_backends.py", line 531, in apply_async
    future = self._workers.submit(SafeFunction(func))
  File "C:\Users\paula\anaconda3\lib\site-packages\joblib\externals\loky\reusable_executor.py", line 177, in submit
    return super(_ReusablePoolExecutor, self).submit(
  File "C:\Users\paula\anaconda3\lib\site-packages\joblib\externals\loky\process_executor.py", line 1102, in submit
    raise self._flags.broken
joblib.externals.loky.process_executor.BrokenProcessPool: A task has failed to un-serialize. Please ensure that the arguments of the function are all picklable.

alejoe91, 2021-04-03:
Hi! Can you paste the code that you are running?

paul-aparicio, 2021-04-03:
Thanks for your response, sorry for not being clearer, this was the command that elicited the error:

In[25]: sorting_IC = ss.run_ironclust(recording_cache, output_folder='results_split_ironclust', grouping_property='group', parallel=True)

alejoe91, 2021-04-03:
Also the previous part please :)

What operating system are you using?

paul-aparicio, 2021-04-03:
Ahh.  I was basically running through this tutorial to learn (https://github.com/SpikeInterface/spiketutorials/blob/master/NWB_Developer_Breakout_Session_May2020/SpikeInterface_Tutorial.ipynb), but I changed it to my data. The non plotting commands are pasted below:

import spikeinterface 
import spikeinterface.extractors as se
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw
import matplotlib.pyplot as plt
import numpy as np

rawdatafile = 'TY20210308_signalCheck_morning.ns6'
recording = se.BlackrockRecordingExtractor(rawdatafile)

recording_prb = recording.load_probe_file('TY_array.prb')
recording_f = st.preprocessing.bandpass_filter(recording_prb, freq_min=250, freq_max=7500)
recording_rm_noise = st.preprocessing.remove_bad_channels(recording_f, bad_channel_ids=[34, 50, 52, 67, 81])
recording_cmr = st.preprocessing.common_reference(recording_rm_noise, reference='median')
recording_cache = se.CacheRecordingExtractor(recording_cmr)
recording_cache.move_to('filtered.dat')
recording_cache.dump_to_pickle('recording.pkl')

ss.IronClustSorter.set_ironclust_path('F:/MLToolBoxes/ironclust')
Setting IRONCLUST_PATH environment variable for subprocess calls to: F:\MLToolBoxes\ironclust

ss.Kilosort2Sorter.set_kilosort2_path('F:/MLToolBoxes/Kilosort-2.0/')
ss.Kilosort3Sorter.set_kilosort3_path('F:/MLToolBoxes/Kilosort/')
ss.WaveClusSorter.set_waveclus_path('F:/MLToolBoxes/wave_clus/')
Setting KILOSORT2_PATH environment variable for subprocess calls to: F:\MLToolBoxes\Kilosort-2.0
Setting KILOSORT3_PATH environment variable for subprocess calls to: F:\MLToolBoxes\Kilosort
Setting WAVECLUS_PATH environment variable for subprocess calls to: F:\MLToolBoxes\wave_clus

# run spike sorting by group 
sorting_IC = ss.run_ironclust(recording_cache, output_folder='results_split_ironclust', grouping_property='group', parallel=True)
print(f'Ironclust found {len(sorting_IC.get_unit_ids())} units')

(as a different issue, the only sorters that show up as installed after the above commands are:
In [24]: ss.installed_sorters()
Out[25]: ['ironclust', 'kilosort2', 'waveclus']

Im not sure why Kilosort3 doesn't install? There was no error output).

paul-aparicio, 2021-04-05:
If it matters at all, I tried duplicating this in Linux (Ubuntu 18.04) and I get the same error as above:

AttributeError: 'BlackrockRawIO' object has no attribute '__read_nsx_header_variant_a'

alejoe91, 2021-04-05:
Hi! What version of neo and SpikeInterface are you using? The problem seems to be related with the BlackRock input..@samuelgarcia any clue?

paul-aparicio, 2021-04-05:
I'm running neo 0.9.0 and spikeextractors 0.9.6

I can't find spikeinterface when I tried conda list? On Windows, I installed it with pip, on Linux I installed it by downloading from GitHub and running python setup.py install

samuelgarcia, 2021-04-06:
This error is strange. Is is seems related to neo and joblib.
Does the same code work with no parralel ?

paul-aparicio, 2021-04-06:
Hello! On the linux install, I tried restarting the jupyter kernel and rerunning the code without the parallel variable in the argument list. This indeed worked. I verified this by restarting the jupyter kernel and running the code again with the parallel variable set to True as an argument. The same error as previously described occurred.  Thanks for your help.

closed 2021-04-11 by paul-aparicio