SpikeInterface/spikeinterface #103: Problem using mountainsort on Ubuntu

frankdopamine, 2021-04-03:
Hi everyone,
previously in our lab, we used mountainsort to sort our recorded data, currently we are trying to switch to spike interface. I’m having issues running mountainsort with spikeinterface. To show you the problem I used your example code “plot_1_sorters_example.py”. 
If I run mountainsort like this: _sorting_MS4 = ss.run_mountainsort4(recording=recording, **default_ms4_params, output_folder='tmp_MS4', raise_error = False)_ I get the following error (**Here you can see the extended traceback: test_mountainsort with raise_error= False.txt**):

   **File "/home/fra/anaconda3/lib/python3.8/site-packages/spikesorters/basesorter.py", line 259, in get_result
    raise SpikeSortingError(f"None of the sorting outputs could be loaded")
spikesorters.sorter_tools.SpikeSortingError: None of the sorting outputs could be loaded.**

 Similarly, I run it with the option with the **raise_error = True** option, I get this (**Here you can see the extended traceback: test_mountainsort with raise_error= True**):

**RuntimeError: 
        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.
        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:
            if __name__ == '__main__':
                freeze_support()
                ...
        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.**

Mountainsort was previously installed via pip in the manner suggested in the manual: **pip install ml_ms4alg**
I have tried different versions of ubuntu (16 LTS and 20 LTS) and WSL2 and created different environments in conda, but I always get the same errors. I tried to use other sorting systems (only Herdingspikes2, SpykingCircus and Tridesclous for the moment) and they worked fine the problem so it seems specific to mountainsort.

[test_mountainsort with raise_error= False.txt](https://github.com/SpikeInterface/spikeinterface/files/6253051/test_mountainsort.with.raise_error.False.txt)
[test_mountainsort with raise_error= True.txt](https://github.com/SpikeInterface/spikeinterface/files/6253052/test_mountainsort.with.raise_error.True.txt)
[plot_1_sorters_example.txt](https://github.com/SpikeInterface/spikeinterface/files/6253054/plot_1_sorters_example.txt)

alejoe91, 2021-04-04:
Hi @frankdopamine 

It seems to be a problem with the multiprocessing in mountainsort4. @magland can you take a look at this?

The strange thing is that MS4 runs fine on our test routine, which runs the same code on an ubuntu machine..

closed 2021-07-22 by samuelgarcia

oaaij-gnahz, 2022-12-26:
Hi,

I am learning about SpikeInterface and I also encountered this issue. I checked the log output `spikeinterface.log` and the field "error_trace" shows:

```
Traceback (most recent call last):
  File "<home>/miniconda3/envs/si_env1/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 225, in run_from_folder
    SorterClass._run_from_folder(output_folder, sorter_params, verbose)
  File "<home>/miniconda3/envs/si_env1/lib/python3.9/site-packages/spikeinterface/sorters/mountainsort4/mountainsort4.py", line 133, in _run_from_folder
    old_api_sorting = mountainsort4.mountainsort4(**ms4_params)
  File "<home>/miniconda3/envs/si_env1/lib/python3.9/site-packages/mountainsort4/mountainsort4.py", line 44, in mountainsort4
    MS4.sort()
  File "<home>/miniconda3/envs/si_env1/lib/python3.9/site-packages/mountainsort4/ms4alg.py", line 858, in sort
    dask.compute(*dask_list, num_workers=self._num_workers)
  File "<home>/miniconda3/envs/si_env1/lib/python3.9/site-packages/dask/base.py", line 600, in compute
    results = schedule(dsk, keys, **kwargs)
  File "<home>/miniconda3/envs/si_env1/lib/python3.9/site-packages/dask/multiprocessing.py", line 233, in get
    result = get_async(
  File "<home>/miniconda3/envs/si_env1/lib/python3.9/site-packages/dask/local.py", line 500, in get_async
    for key, res_info, failed in queue_get(queue).result():
  File "<home>/miniconda3/envs/si_env1/lib/python3.9/concurrent/futures/_base.py", line 439, in result
    return self.__get_result()
  File "<home>/miniconda3/envs/si_env1/lib/python3.9/concurrent/futures/_base.py", line 391, in __get_result
    raise self._exception
concurrent.futures.process.BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending.
```

Previously when I was using the standalone [ml_ms4alg](https://github.com/magland/ml_ms4alg) I did not see this problem. Could you please advise on how to troubleshoot? 

Thank you so much!