SpikeInterface/spikeinterface #115: Importing EDF Files

farazmoradi, 2021-04-25:
Hi, Is there any way that we can import EDF files directly to the spike interface?

alejoe91, 2021-04-25:
Hi @farazmoradi 

EDF files are currently not supported. If you have Python code to load them, it's pretty straightforward to write an extractor for it. We have some instructions on how to do it here: https://spikeinterface.readthedocs.io/en/latest/recording_extractor.html

samuelgarcia, 2021-04-26:
I plan to do it in neo since many years.
So it would also avaliable here.
I would be happy to help if you start a PR for EDF in neo.
Note that python-mne handle it already.

farazmoradi, 2021-04-26:
Thanks, I get that!

alejoe91, 2022-08-01:
https://github.com/SpikeInterface/spikeinterface/pull/853

closed 2022-08-01 by alejoe91