SpikeInterface/spikeinterface #132: Working with common reference data grouped.

valejandracch, 2021-05-03:
Hello!

Im working with data obtained with openEphys. Cambridge probe with 2 shanks and 32 channels.
The recording has been extracted using the OpenEphysRecordingExtractor  and I added the probe file:

Original channels: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34]
Channels after loading the probe file: [29, 25, 31, 26, 30, 22, 16, 19, 23, 24, 17, 18, 28, 27, 20, 21, 2, 6, 0, 5, 1, 9, 15, 12, 8, 7, 14, 13, 3, 4, 11, 10]
Channel groups after loading the probe file: [0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1]

After I cheked for bad channels and passed some notch filters and then id ussed the common reference but then when i try to plot but i have the following:
(help please)

recording_cmr_groups = st.preprocessing.common_reference(recording_notch , reference='median', groups= [[0],[1]])

wbpSPKG = sw.plot_timeseries(recording_cmr_groups, trange=[10, 12])

Figure 8
Zoom to rectangle x/y fixes axis, CTRL fixes aspect

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-23-ebac8b88ddde> in <module>
----> 1 wbpSPKG = sw.plot_timeseries(recording_cmr_groups, trange=[10, 12])

C:\ProgramData\Anaconda3\envs\spike\lib\site-packages\spikewidgets\widgets\timeserieswidget\timeserieswidget.py in plot_timeseries(recording, channel_ids, trange, color_groups, color, figure, ax)
     39         color=color,
     40         figure=figure,
---> 41         ax=ax
     42     )
     43     W.plot()

C:\ProgramData\Anaconda3\envs\spike\lib\site-packages\spikewidgets\widgets\timeserieswidget\timeserieswidget.py in __init__(self, recording, channel_ids, trange, color_groups, color, figure, ax)
     60             assert len(trange) == 2, "'trange' should be a list with start and end time in seconds"
     61             self._visible_trange = [int(t * recording.get_sampling_frequency()) for t in trange]
---> 62         self._initialize_stats()
     63         # self._vspacing = self._mean_channel_std * 20
     64         self._vspacing = self._max_channel_amp * 1.5

C:\ProgramData\Anaconda3\envs\spike\lib\site-packages\spikewidgets\widgets\timeserieswidget\timeserieswidget.py in _initialize_stats(self)
    139         for ii in range(M0):
    140             self._channel_stats[self._visible_channels[ii]] = _compute_channel_stats_from_data(chunk0[ii, :])
--> 141         self._mean_channel_std = np.mean([self._channel_stats[m]['std'] for m in self._visible_channels])
    142         self._max_channel_amp = np.max([self._channel_stats[m]['max_amp'] for m in self._visible_channels])
    143 

C:\ProgramData\Anaconda3\envs\spike\lib\site-packages\spikewidgets\widgets\timeserieswidget\timeserieswidget.py in <listcomp>(.0)
    139         for ii in range(M0):
    140             self._channel_stats[self._visible_channels[ii]] = _compute_channel_stats_from_data(chunk0[ii, :])
--> 141         self._mean_channel_std = np.mean([self._channel_stats[m]['std'] for m in self._visible_channels])
    142         self._max_channel_amp = np.max([self._channel_stats[m]['max_amp'] for m in self._visible_channels])
    143 

KeyError: 31

maurotoro, 2021-05-06:
Dearest, I found a bug that might be related to this.

If one instantiates a recording, adds a probe and does some preprocessing, the resulting object is always nice and keeps all metadata, probe and segments and the like. But, once one passes the recording through the remove_bad_channels the result gets broken. 

I checked and it could be that in the remove_bad_channels.py, around lines 42 the BasePreprocessor is __init__ed before the ChannelSliceRecording, plus is receiving two *self* and that could be messing the result. I changed this and sending the pr. On my side it looks as if it worked.

Suerte!

alejoe91, 2021-05-09:
Hi @valejandracch 

I think there could be a bug in the timeseries widget when channels are missing, I'll look into it.

Note that the `groups` option in `st.preprocessing.common_reference` is not supposed to take group ids as input, but rather a list of channel ids belonging to the different groups.

In your case you would need to do something like this:
```
# get unique groups
channel_groups = recording_notch.get_channel_groups()
group_ids = np.unique(channel_groups)
# get channels by group
channels_groups_for_cmr = []
for group_id in group_ids:
      channels_in_group_id = []
      for ch_group, ch in zip(channel_groups, recording_notch.get_channel_ids()):
               if ch_group == group_id:
                        channels_in_group.append(ch)
      channels_groups_for_cmr.append(channels_in_group_id)

print(channels_groups_for_cmr)
```

This should print:
```
[[29, 25, 31, 26, 30, 22, 16, 19, 23, 24, 17, 18, 28, 27, 20, 21], [2, 6, 0, 5, 1, 9, 15, 12, 8, 7, 14, 13, 3, 4, 11, 10]]
```

I admit this solution is not optimal and passing an argument like `group_by` would be better. We are in the process of refractoring the code (https://github.com/SpikeInterface/spikeinterface) and we'll definitely add a solution like this

alejoe91, 2021-05-10:
Hi @valejandracch 

Can you paste here the full code you are running? I'll try to look into this in the next days

valejandracch, 2021-05-11:
import spikeinterface
import spikeinterface.extractors as se        # file IO and probe handling
import spikeinterface.toolkit as st           # processing toolkit for pre-, post-processing, validation, and automatic curation
import spikeinterface.sorters as ss           # Python wrappers of spike sorters
import spikeinterface.comparison as sc        # comparison of spike sorting output
import spikeinterface.widgets as sw           # visualization
import pyopenephys
import phy
import numpy as np
import pandas as pd
import scipy
import matplotlib.pylab as plt
%matplotlib notebook
#%matplotlib inline

15:09:54 [I] klustakwik KlustaKwik2 version 0.2.6

recording_folder = '//10.152.96.1/home/Openephys/ProbeFeb/M2_2021-03-18_15-01-41_L1L2SL5sD17'
loc='D:/Alejandra-WST2/ProbeSortings/M2_2021-03-18_15-01-41_L1L2SL5sD17/'

s=recording_folder.split('/')
recordName= s[-1]
recordName
s=recordName.split('_')
#recordDic={"experiment":s[-1], 'Mouse': s[0], 'date': s[1], 'depth' : '1st', 'trainingDay': 0 }
#recordDic
s

['M2', '2021-03-18', '15-01-41', 'L1L2SL5sD17']

recordName

'M2_2021-03-18_15-01-41_L1L2SL5sD17'

recording = se.OpenEphysRecordingExtractor(recording_folder, experiment_id=0, recording_id=0)
#se.installed_recording_extractor_list
type(recording)

Loading Open-Ephys: reading settings...
Decoding data from  binary  format
Reading oebin file

spikeextractors.extractors.openephysextractors.openephysextractors.OpenEphysRecordingExtractor

channel_ids = recording.get_channel_ids()
fs = recording.get_sampling_frequency()
num_chan = recording.get_num_channels()

print('Channel ids:', channel_ids)
print('Sampling frequency:', fs)
print('Number of channels:', num_chan)

Channel ids: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34]
Sampling frequency: 30000
Number of channels: 35

(recording.get_num_frames()/30000)/60

29.593457777777775

 

recording_remove_bad_auto = st.preprocessing.remove_bad_channels(recording, bad_channel_ids=None, bad_threshold=2,
                                                                 seconds=10)

print(len(recording_remove_bad_auto.get_channel_ids()),'good channels')

35 good channels

recording_1 = st.preprocessing.remove_bad_channels(recording_remove_bad_auto, bad_channel_ids=[35, 36, 37, 38, 39, 40, 41, 42])
[... 95 lines cut ...]
 6,
 0,
 5,
 1,
 9,
 15,
 12,
 8,
 7,
 14,
 13,
 3,
 4,
 11,
 10]

Strng = 'preprocessed_dataB'
exte ='.dat'

output_folderdat= loc+Strng+recordName+exte
output_folderdat

'D:/Alejandra-WST2/ProbeSortings/M2_2021-03-18_15-01-41_L1L2SL5sD17/preprocessed_dataBM2_2021-03-18_15-01-41_L1L2SL5sD17.dat'

recording_cache = se.CacheRecordingExtractor(recording_cmr_groups, save_path =output_folderdat, n_jobs=1, chunk_mb=500, verbose=True)

alejoe91, 2021-05-11:
Can you separate the code from the output? Also you can paste a `code section using the coding format :)

valejandracch, 2021-05-11:
import spikeinterface
import spikeinterface.extractors as se        # file IO and probe handling
import spikeinterface.toolkit as st           # processing toolkit for pre-, post-processing, validation, and automatic curation
import spikeinterface.sorters as ss           # Python wrappers of spike sorters
import spikeinterface.comparison as sc        # comparison of spike sorting output
import spikeinterface.widgets as sw           # visualization
import pyopenephys
import phy
import numpy as np
import pandas as pd
import scipy
import matplotlib.pylab as plt
get_ipython().run_line_magic('matplotlib', 'notebook')
#%matplotlib inline

recording_folder = '//10.152.96.1/home/Openephys/ProbeFeb/M2_2021-03-18_15-01-41_L1L2SL5sD17'
loc='D:/Alejandra-WST2/ProbeSortings/M2_2021-03-18_15-01-41_L1L2SL5sD17/'

s=recording_folder.split('/')
recordName= s[-1]
recordName
s=recordName.split('_')

recording = se.OpenEphysRecordingExtractor(recording_folder, experiment_id=0, recording_id=0)
#se.installed_recording_extractor_list
type(recording)

channel_ids = recording.get_channel_ids()
fs = recording.get_sampling_frequency()
num_chan = recording.get_num_channels()

print('Channel ids:', channel_ids)
print('Sampling frequency:', fs)
print('Number of channels:', num_chan)

(recording.get_num_frames()/30000)/60

recording_remove_bad_auto = st.preprocessing.remove_bad_channels(recording, bad_channel_ids=None, bad_threshold=2,
                                                                 seconds=10)

print(len(recording_remove_bad_auto.get_channel_ids()),'good channels')

recording_1 = st.preprocessing.remove_bad_channels(recording_remove_bad_auto, bad_channel_ids=[35, 36, 37, 38, 39, 40, 41, 42])
recording_prb = recording_1.load_probe_file('TwoShank32ProbeOK.prb')

print('Original channels:', recording.get_channel_ids())
print('Channels after loading the probe file:', recording_prb.get_channel_ids())
print('Channel groups after loading the probe file:', recording_prb.get_channel_groups())

recording_prb.dump_to_dict()

recording_notch = st.preprocessing.notch_filter(recording_prb, freq=50, q=10)
recording_notch = st.preprocessing.notch_filter(recording_notch, freq=1000, q=30)

recording_notch.get_channel_groups()

# get unique groups
channel_groups = recording_notch.get_channel_groups()
group_ids = np.unique(channel_groups)
# get channels by group
channels_groups_for_cmr = []
for group_id in group_ids:
      channels_in_group_id = []
      for ch_group, ch in zip(channel_groups, recording_notch.get_channel_ids()):
               if ch_group == group_id:
                        channels_in_group_id.append(ch)
      channels_groups_for_cmr.append(channels_in_group_id)

print(channels_groups_for_cmr)

recording_bpSPK = st.preprocessing.bandpass_filter(recording_notch, freq_min=300, freq_max=6000)

recording_bpSPK.get_channel_groups()

recording_cmr_groups = st.preprocessing.common_reference(recording_bpSPK, reference='median', groups= channels_groups_for_cmr)

recording_cmr_groups.get_channel_groups()

sw.plot_timeseries(recording_cmr_groups, trange=[10, 12])

recording_cmr_groups.get_channel_gains()

recording_cmr_groups.get_channel_ids()

Strng = 'preprocessed_dataB'
exte ='.dat'

output_folderdat= loc+Strng+recordName+exte
output_folderdat

recording_cache = se.CacheRecordingExtractor(recording_cmr, save_path =output_folderdat,  n_jobs=1, chunk_mb=500, verbose=True)

valejandracch, 2021-05-12:
> 
> 
> Can you separate the code from the output? Also you can paste a `code section` using the coding format :)

sorry, now is in code format.  Thanks for the help

closed 2021-07-22 by samuelgarcia