SpikeInterface/spikeinterface #133: passing relative file paths to sorter

atranvan, 2021-05-04:
Hi, 

I was wondering if there's an easy way to use relative paths instead of absolute paths in prm files generated through spikeinterface. 

If I provide relative file paths to the recording objects, the resulting extractors contain full paths.

I tried making a json file containing relative paths and then using se.load_extractor_from_json but the resulting loaded recording also contains full path to the data.

my issue with absolute paths is for visualization resulting from sorting with klusta (haven't tried with other sorters yet) on a different computer using phy (running the sorting on a cluster where we can't perform visualization,  then copying results and visualizing on local computer) - absolute paths corresponding to location of data files on cluster mean I can't see the waveforms on a local computer...

thanks in advance!

alejoe91, 2021-05-05:
> Hi,
> 

Hi @atranvan  :)

> I was wondering if there's an easy way to use relative paths instead of absolute paths in prm files generated through spikeinterface.
> 

Currently not, but I agree that this could be something useful! I can make a patch with something with two option:
1. an extra `relative_to` argument to the `dump` functions. The `relative_to` will be either None (by default, meaning absolute path) or a path with respect to which the relative path is computed
2. an extra function `change_paths_to_relative` that can take a dict, json, or pkl (the dumping options) and overwrites in place the object by changing paths from absolute to relative.

I think that would work for all cases. Does that sound ok?

Give me a couple of days and thanks for using SI!

atranvan, 2021-05-05:
Hi @alejoe91

Thanks a lot for the quick answer! Yes it sounds like either option would be really useful! Looking forward to trying it out!

samuelgarcia, 2021-05-05:
Hi Alex.
For the next release of spikeinterface this is something we have in mind but not implemented yet.
Keep this issue open please.
Samuel

alejoe91, 2021-05-05:
@atranvan first option is implemented here: https://github.com/SpikeInterface/spikeextractors/pull/630

atranvan, 2021-05-05:
Hi @alejoe91 ,

thanks so much for the quick update!
I am running into a few issues :
I updated to get the commit above through : 
`pip install git+https://github.com/alejoe91/spikeinterface.git@relative_paths`

I may have missed a version previous to that but some extractor function I was using before (BinDatRecordingExtractor) has also changed (hence I may be doing something else wrong) : 

I used to create my recording object as

```
recording = se.BinDatRecordingExtractor(recording_file, numchan=num_channels, sampling_frequency=sampling_frequency,
                                        dtype=dtype, gain=gain, channel_offset=channel_offset, time_axis=time_axis)
```

which, from extractor full list, I am guessing I need to substitute by : 
```
recording = se.BinaryRecordingExtractor(recording_file, num_chan=num_channels, sampling_frequency=sampling_frequency,
                                        dtype=dtype, gain=gain, offset=channel_offset, time_axis=time_axis) 
```
where recording_file is a string (path to the .dat file)

However, when I try the `dump_to_json()` function with relative_to argument, I get this error: 

`TypeError: expected str, bytes or os.PathLike object, not list`

The recording object appears as: 
```
BinaryRecordingExtractor: 32 channels - 1 segments - 24.4kHz
  files_path: ['/camp/lab/rancze/working/Alex/analysis/TDT_analysis/ephys_vest_vis/test_si/dat/car_data.dat']
```

Any clue what I'm doing wrong?
thanks so much in advance!

alejoe91, 2021-05-05:
@samuelgarcia any idea?

samuelgarcia, 2021-05-05:
BinaryRecordingExtractor take now a list of files because Recording is now multi segment.
In the init if it is not a list then it is converted to a list one one element.
Alessio did you take this in account ?

alejoe91, 2021-05-05:
How could I have known :P I'll make a fix. Is this the only case the file path arg is a list?

samuelgarcia, 2021-05-05:
note that it can be both srt/Path or list of str/Path.
It really depend on the context.
Have a look to implementation, the multi segment stuff have made some internal changes like this.

samuelgarcia, 2021-05-17:
Close with #136

closed 2021-05-17 by samuelgarcia