SpikeInterface/spikeinterface #142: SubRecordingExtractor / segmenting frames in new API

timsainb, 2021-05-12:
Hey All,

I've run into a wall with re-implementing my spikesorting pipeline in the new API. My pipeline relies on merging together a recording and the last few hours of a previous recording, so that I can call a unit a unit across recordings/days. Unfortunately in the new API this hasn't been implemented yet. 

In a previous thread (https://github.com/SpikeInterface/spikesorters/issues/220#issuecomment-834119000), @samuelgarcia mentioned that this is on his todo list. 

In the previous API it was:

```python
recording = se.SubRecordingExtractor(
                    parent_recording=recording,
                    start_frame=start_frame,
                    end_frame=end_frame,
                )
```

I can potentially try to implement this myself with a little help. Something along the lines of:

spikeinterface/spikeinterface/core/frameslicerecording.py 

```python

class FrameSliceRecording(BaseRecording):
    """
    Class to slice a Recording object based on channel_ids.
    """

    def __init__(self, parent_recording, start_frame, end_frame):

        self._parent_recording = parent_recording
        self._start_frame = start_frame
        self._end_frame = end_frame
        self._channel_ids = parent_recording.get_channel_ids()

        # some checks
        # TODO assert that all frames are within bounds of recording

        BaseRecording.__init__(self,
                               sampling_frequency=parent_recording.get_sampling_frequency(),
                               channel_ids=self._channel_ids,
                               dtype=parent_recording.get_dtype())

        # link recording segment
        for parent_segment in self._parent_recording._recording_segments:
            sub_segment = FrameSliceRecordingSegment(parent_segment, start_frame, end_frame)
            self.add_recording_segment(sub_segment)

        # update dump dict
        self._kwargs = {'parent_recording': parent_recording.to_dict()}

class FrameSliceRecordingSegment(BaseRecordingSegment):
    """
    Class to return a sliced segment traces.
    """

    def __init__(self, parent_recording_segment, start_frame, end_frame):
        BaseRecordingSegment.__init__(self)
        self._parent_recording_segment = parent_recording_segment

    def get_num_samples(self) -> SampleIndex:
        # TODO set self.total_length
        #return self._parent_recording_segment.get_num_samples()

        
```

samuelgarcia, 2021-05-12:
In my mind for the API, "time slice" is a bit more complex, it can be:
  1. take one or more among N segment for the original
  2. take a sample slice from one segment (your example)
  3. both

So we could have a one class complicated that handle both.
Or we can have two class, very simple to write, for case 1.  and 2..

For case 2., we need to test if the parent is one segment only.
I can implemnt it very soon if we are agree on th behavior.
@alejoe91 @timsainb

alejoe91, 2021-05-12:
I think that the most important than my is slicing one segment, because different segments will be sliced differently. Maybe TimeSlice could only work on single segments and externally one could slice multiple segments

samuelgarcia, 2021-05-12:
OK So you want a class that time slice mono segment.
I will do it very soon.

samuelgarcia, 2021-05-12:
Hi Tim,
have a look to this : https://github.com/SpikeInterface/spikeinterface/pull/143
and tell me if it works for you.
I will implement also a segment_slice soon.

timsainb, 2021-05-12:
Looks good, thanks! Trying this out now.

BTW, when I grabbed that PR I found a small bug in
https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/tridesclous/tridesclous.py#L105

Where 
```python
num_seg = recording.get_num_segments()
...
files_path = [str(output_folder / f'raw_signals_{i}.raw') for i in num_seg]
```

num_seg is an integer, so it should be `range(num_seg)` here I believe.

samuelgarcia, 2021-05-17:
Hi Tim,
yes of course.
Thank you for your review.

closed 2021-05-17 by samuelgarcia