SpikeInterface/spikeinterface #145: AssertionError on adding probe to ChannelSliceRecording and concatenate_recordings

timsainb, 2021-05-13:
There seems to be an issue with adding a probe to a recording comprised of a ChannelSliceRecording that has been concatenated. I've played around with it a bit and seem to reproduce the error in a few different ways but can't figure out the source. 

Here is a minimal example on a colab notebook:
https://colab.research.google.com/drive/1iLvlZ_NX_OSowhnQmAkt-evHmr9QKlQI?usp=sharing

Here are some local examples trying different things: 
### ChannelSliceRecording + concatenate_recordings:
```python
recording_list = []
for ri, row, in enumerate([prev_row, row]):
    recording = si.core.BinaryRecordingExtractor(
        row.dat_file,
        sampling_frequency=row.sample_rate,
        num_chan=row.num_channels_total,
        dtype="int16",
    )
    recording = si.core.ChannelSliceRecording(
        parent_recording=recording,
        channel_ids=row.channels
    )
    recording_list.append(recording)
recording = si.concatenate_recordings(recording_list)

recording = recording.set_probes(probe_group)
```
```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-66-0bea75e4bb82> in <module>
     15     recording_list.append(recording)
     16 recording = si.concatenate_recordings(recording_list)
---> 17 recording = recording.set_probes(probe_group)

/mnt/cube/tsainbur/Projects/github_repos/spikeinterface/spikeinterface/core/baserecording.py in set_probes(self, probe_or_probegroup, group_mode, in_place)
    259         else:
    260             if np.array_equal(new_channel_ids, self.get_channel_ids()):
--> 261                 sub_recording = self.clone()
    262             else:
    263                 sub_recording = ChannelSliceRecording(self, new_channel_ids)

/mnt/cube/tsainbur/Projects/github_repos/spikeinterface/spikeinterface/core/base.py in clone(self)
    280         """
    281         d = self.to_dict(include_annotations=True, include_properties=True, include_features=True)
--> 282         clone = BaseExtractor.from_dict(d)
    283         return clone
    284 

/mnt/cube/tsainbur/Projects/github_repos/spikeinterface/spikeinterface/core/base.py in from_dict(d)
    272             The loaded extractor object
    273         '''
--> 274         extractor = _load_extractor_from_dict(d)
    275         return extractor
    276 

/mnt/cube/tsainbur/Projects/github_repos/spikeinterface/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    579         cls = _get_class_from_string(class_name)
    580 
--> 581     assert cls is not None and class_name is not None, "Could not load spikeinterface class"
    582     if not _check_same_version(class_name, dic['version']):
    583         print('Versions are not the same. This might lead to errors. Use ', class_name.split('.')[0],

AssertionError: Could not load spikeinterface class
```

### concatenate_recordings without ChannelSliceRecording:
```python
recording_list = []
for ri, row, in enumerate([prev_row, row]):
    recording = si.core.BinaryRecordingExtractor(
        row.dat_file,
        sampling_frequency=row.sample_rate,
        num_chan=row.num_channels_total,
        dtype="int16",
    )

    recording_list.append(recording)
recording = si.concatenate_recordings(recording_list)

recording = recording.set_probes(probe_group)
```
```
No error
```

### ChannelSliceRecording without concatenate_recordings:
```python
recording_list = []
for ri, row, in enumerate([prev_row, row]):
    recording = si.core.BinaryRecordingExtractor(
        row.dat_file,
        sampling_frequency=row.sample_rate,
        num_chan=row.num_channels_total,
        dtype="int16",
    )
    recording = si.core.ChannelSliceRecording(
        parent_recording=recording,
        channel_ids=row.channels
    )
    recording_list.append(recording)

recording = recording.set_probes(probe_group)
```
```
No error
```

I can't figure out what's going on.

samuelgarcia, 2021-05-17:
Hi Tim,
thanks for this feedback. I will have a look soon.

samuelgarcia, 2021-05-17:
Hi Tim.
Have a look to  #147 this should solve this problem.
Big thanks for caching this.

samuelgarcia, 2021-05-17:
Laso note that

```python
recording = si.core.ChannelSliceRecording(
        parent_recording=recording,
        channel_ids=[0,1,2,3]
    )
```
can be
```python
recording = recording.channel_slice(channel_ids=[0,1,2,3])
```

closed 2021-05-17 by samuelgarcia