SpikeInterface/spikeinterface #148: Recording extractor for Herdingspikeresults and Ironclust results

Shivooone, 2021-05-17:
Hello,

I have a trouble concerning the spikeextractors since I believe that the herdingspikes extractor is HS2SortingExtractor but I get the following error

[image]

[image]

Could you please give me a hint on that ?

Moreover It seems that the ironclust results extractor doesn't exist?

Thanks a lot for your help,
Shiva

alejoe91, 2021-05-17:
Hi Shiva,

The `HS2SortingExtractor` needs the `file_path` (not folder path) to the hdf5 result file.

As for IronClust, did you run it with SpikeInterface? In that case you can load the `firings.mda` with the `MdaSortingExtractor`.

Cheers
Alessio

samuelgarcia, 2021-05-17:
Hi Shiva,
yes MdaSortingExtractor is not implemented yet. We will do at the end of the weeks.
Sam

alejoe91, 2021-05-17:
@samuelgarcia it's the current API (the error comes from spikeextractors)

samuelgarcia, 2021-05-17:
Shiva, you should use the new API !!

Shivooone, 2021-05-17:
Perfect, thanks for your answers :)

closed 2021-05-17 by Shivooone