SpikeInterface/spikeinterface #150: Saving sorting object to NIX

vojtasmekal, 2021-05-17:
Hi,

I am trying to write a sorting object created as the agreement sorting between three sorters to Nix format using the NIXIOSortingExtractor, but when I run:

se.NIXIOSortingExtractor.write_sorting(sorting=sorting_agreement, save_path='spikes_agreement.nix')

I get the following:

~/opt/anaconda3/lib/python3.8/site-packages/nixio/section.py in create_property(self, name, values_or_dtype, oid, copy_from, keep_copy_id)
    149             if (isinstance(vals, (Sequence, Iterable)) and
    150                     not isinstance(vals, string_types)):
--> 151                 single_val = vals[0]
    152             else:
    153                 # Make sure the data will always be created with an array.

KeyError: 0

This may be a question for the NIXIO people instead, but any advice is appreciated. The .nix file is created and seems to contain all the data, so the only issue is that the process doesn't complete. 

Thanks in advance!

alejoe91, 2021-05-17:
Hi,

Can you paste the entire traceback? We should get the nix developer involved in this as I'm not very familiar with it :)

vojtasmekal, 2021-05-17:
Hi,

Of course, here it is! 

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-64-1aef9d31c026> in <module>
      1 # Save the sorting object into NixIO format
      2 
----> 3 se.NIXIOSortingExtractor.write_sorting(sorting=sorting_agreement, save_path='spikes_agreement.nix')

~/opt/anaconda3/lib/python3.8/site-packages/spikeextractors/extractors/nixioextractors/nixioextractors.py in write_sorting(sorting, save_path, overwrite)
    245                 else:
    246                     values = propvalue
--> 247                 unit_md.create_property(propname, values)
    248 
    249         nf.close()

~/opt/anaconda3/lib/python3.8/site-packages/nixio/section.py in create_property(self, name, values_or_dtype, oid, copy_from, keep_copy_id)
    149             if (isinstance(vals, (Sequence, Iterable)) and
    150                     not isinstance(vals, string_types)):
--> 151                 single_val = vals[0]
    152             else:
    153                 # Make sure the data will always be created with an array.

KeyError: 0

closed 2021-07-22 by samuelgarcia