SpikeInterface/spikeinterface #153: Computing spike template with only a subset of the spikes from waveform extractor

timsainb, 2021-05-20:
Using the new waveform extractor, I set the waveform to grab all of the spikes, which is useful for certain things (like projecting spikes into low-d spaces) but for things like computing the template of a unit, using every spike is probably unnecessary. 

Is there a way to use the same waveform extractor and compute certain features with only a subset of the waveforms? 

e.g. ` we.get_template(unit_id=unit_id, max_spikes = 500, mode='median')`, which would only grab 500 random indices from the npy files.

samuelgarcia, 2021-05-20:
Hi Tim,
yes we can do that.

I have another plan in mind which should implemented soon.

The workflow would be:
  * make a waveformExtractor with feww spike 
  * make WaveformPrincipalComponent on it just to learn the projection for PCA
  * then create an object that stream all spike, extract waveform on the fly, project on the fly and store the PCs

This would be more efficient, Ithink that a full waveform extraction. For high channel count this is almost impossible.

timsainb, 2021-05-21:
Sounds good. The advantage of the former approach, for me personally, is I'm using the entire dataset of spikes for projection into my own embeddding algorithm (https://github.com/lmcinnes/umap/blob/master/umap/parametric_umap.py)

For high-channel count electrodes (or really any electrodes), perhaps it would make sense to  save only the "best channels", and treat the numpy arrays as sparse arrays. E.g. for each unit grab the best channels with your best_channels function from some initial subset of spikes, and save the npy arrays with dims (n_best_channels, n_spikes) rather than (n_channels, n_spikes).

timsainb, 2021-06-08:
Hey @samuelgarcia do you have some sample code for streaming spikes (as in get_snippets in the previous api), as you mention in your comment above?

It looks like the way you'd do this is to write a variation on _waveform_extractor_chunk that interacts with ChunkRecordingExecutor, but rather than write to a memmap,. return them.

samuelgarcia, 2021-06-09:
Hi Tim,
I guess you are mentioning this https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/toolkit/postprocessing/principal_component.py#L153

Indeed, I did something to stream all chunk with:
  1. extractor waveform
  2. keep best channel 
  3. project on PCA space
  4. save to file

This is very very specific to phy export engine.

In we could do something similar for your need but keep in mind that potentilly all waveofrm and also all PCA could not fit in memmory, we need a engine to consume on the fly batch per batch.

If you have time, we could have a short call so you could explain your goal and how we could implement it in SI.

timsainb, 2021-06-09:
Thanks @samuelgarcia ! I'd be grateful to Zoom over this, I'll email you.

I suppose my use is pretty similar to the PCA postprocessor, so it should be pretty adaptable and maybe I should write a post-processor for it. Generally I think this could be a useful tool for a lot of people so I wouldn't be averse to writing it. 

I wrote [a submodule for the python package UMAP, called Parametric UMAP](https://umap-learn.readthedocs.io/en/latest/parametric_umap.html). UMAP is a dimensionality reduction algorithm like PCA or t-SNE. Parametric UMAP learns a parametric relationship between data and embeddings (like PCA) so you can stream the dataset in the exact same was as you do in the aveformPrincipalComponent object.

The differences that I need to bake in are in training the embedder:
- I'd like to use subsamples of the full dataset, rather than the one sub-sample that the waveformextractor grabs. So for each 'epoch' of training, subset say, 200k spikes, and run fit_partial on the embedder. 
- I would want to train over multiple recordings/sortings. My datasets are weeks long recordings/sortings of the same populations, part of the point of these embeddings is to provide visual feedback on whether I've glued together units well across sorts/recordings  using my pipeline. 

So in (very) psuedo code:

```python
from umap.parametric_umap import ParametricUMAP

# train the embedder
embedder = ParametricUMAP()
for epoch in range(n_epochs):
   spikes_list = []
   for sort, recording in zip(sorts, recordings):
        spikes = grab_a_few_thousand_spikes(sorts, recordings)
        spikes_list.append(spikes)
   embedder.partial_fit(spikes_list)

# project every spike
memmap_array = np.memmap(n_spikes_total, 2)
for sort, recording in zip(sorts, recordings):
   project_all_spikes_into_memmap_array(sort, recording, memmap_array)
```

closed 2021-07-22 by samuelgarcia