SpikeInterface/spikeinterface #164: YASS error "exit()" function

ZiadAli, 2021-05-23:
Hello - I encountered an error running the YASS sorter where it tried to run what seems like an undefined function called "exit()". How should I fix this? I'm using Google Colab and running the code from the plot_getting_started.py example. Thanks!

My code:
ss.run_yass(recording=recording_cmr)

Error Message:
```
NameError                                 Traceback (most recent call last)
<ipython-input-27-f149463c870a> in <module>()
      1 #Must manually upload config_default.yaml file to yass folder below first
      2 #'/usr/local/lib/python3.7/dist-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/yass/config_default.yaml'
----> 3 ss.run_yass(recording=recording_cmr)

3 frames
/usr/local/lib/python3.7/dist-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py in run_yass(*args, **kwargs)
    130 def run_yass(*args, **kwargs):
    131     __doc__ = _common_run_doc.format('yass')
--> 132     return run_sorter('yass', *args, **kwargs)
    133 
    134 def run_pykilosort(*args, **kwargs):

/usr/local/lib/python3.7/dist-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, **sorter_params)
     48     output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
     49     SorterClass.set_params_to_folder(recording, output_folder, sorter_params, verbose)
---> 50     SorterClass.setup_recording(recording, output_folder, verbose=verbose)
     51     SorterClass.run_from_folder(output_folder, raise_error, verbose)
     52     sorting = SorterClass.get_result_from_folder(output_folder)

/usr/local/lib/python3.7/dist-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/basesorter.py in setup_recording(cls, recording, output_folder, verbose)
    172             all_params =  json.load(f)
    173             sorter_params = all_params['sorter_params']
--> 174         cls._setup_recording(recording, output_folder, sorter_params, verbose)
    175 
    176     @classmethod

/usr/local/lib/python3.7/dist-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/yass/yass.py in _setup_recording(cls, recording, output_folder, params, verbose)
    207         from pprint import pprint
    208         pprint(merge_params)
--> 209         exit()
    210 
    211         # to yaml again (for NNs update)

NameError: name 'exit' is not defined
```

alejoe91, 2021-05-23:
Hi @ZiadAli 

This is definitely a leftover bug on our side! Will push a fix soon. Note that YASS hasn't been thoroughly tested, so we really appreciate your feedback!

alejoe91, 2021-05-25:
@samuelgarcia already opened a PR: https://github.com/SpikeInterface/spikeinterface/pull/166

alejoe91, 2021-05-26:
@ZiadAli the problem seems to be arising from YASS itseld, not from the SpikeInterface wrapper. @catubc can you take a look?

catubc, 2021-05-26:
@ZiadAli, yes it seems the phy generating code is crashing likely due to some weird output. I don't have immediate time to look into this.

In any case, yass should have finished sorting by that stage and the spike trains should be available. Please turn off the phy flag in the config (set   generate_phy: 0 in the config or using the SI interface).  You should be able to visualize

alejoe91, 2021-05-26:
@catubc @samuelgarcia I think the phy flag should be false by default

samuelgarcia, 2021-05-26:
yes

ZiadAli, 2021-05-26:
Turning off the generate_phy flag fixed that error but now it encounters this one:

`"ValueError: Found array with 0 sample(s) (shape=(0, 32)) while a minimum of 1 is required by check_pairwise_arrays."
`

I wonder if this may be a consequence of the file I'm using to test YASS (mearec_test_10s.h5) which is potentially not well suited for the algorithm? Do you have a standard dataset you've been using to test YASS that I could try out to see if that executes without errors? The full json log is attached. Thanks for the help!
[spikeinterface_log.txt](https://github.com/SpikeInterface/spikeinterface/files/6545804/spikeinterface_log.txt)

catubc, 2021-05-26:
Hi @ZiadAli 
Yass is returning zero clusters and is crashing. We did not anticipate that and have not fixed it.  It will be probably fixed in a future version.

Not sure what type of data you're applying it to, but it may not work as is.  Perhaps it is too short, or low firing rate.  I see yass finds 40 units in the first iteration, then down to 19 which post-cleaning returns only 1 unit.  This usually means that the data is too noisy, or too short, or there is significant drift which Yass cannot currently handle.

alejoe91, 2021-05-26:
Yeah this is only 10s. Can you try to run it with a longer recording @ZiadAli ? You can generate one using the `toy_example` function

catubc, 2021-05-26:
Oh yeah, that will not work in yass. There are too many statistical dependencies / heuristics that will break.
You should concatenate perhaps 30 or more of these chunks and run together if they are from the same recording depth/penetration etc.

alejoe91, 2021-05-26:
Also that's a possibility! @ZiadAli you can do it with the `concatenate_recordings` function:
```
recording_concat = si.concatenate_recordings([recording_cmr] * 10)
```

closed 2021-07-22 by samuelgarcia