SpikeInterface/spikeinterface #165: New version of spikeinterface

benefron, 2021-05-24:
Hello I am trying to work with cached sorting that I saved before on the older version of spikeinterface.
I ran into 2 problems:
1. when I try to extract the recording I get the following:
"AttributeError: module 'neo.rawio' has no attribute 'OpenEphysBinaryRawIO'" from what I see when trying to import from neo the attribute is spelled as follows: "from neo.rawio.openephysrawio" only lower case letters.
2. when I try to extract my saved NPZ sorting files I get the following:
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-12-43cb9491cf52> in <module>
----> 1 spikeinterface.NpzSortingExtractor('/home/labs/lampllab/befron/ION_control/Sorting/C1_CUT/mnt.npz')

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/npzsortingextractor.py in __init__(self, file_path)
     25 
     26         npz = np.load(file_path)
---> 27         num_segment = int(npz['num_segment'][0])
     28         unit_ids = npz['unit_ids']
     29         sampling_frequency = float(npz['sampling_frequency'][0])

~/.local/lib/python3.7/site-packages/numpy/lib/npyio.py in __getitem__(self, key)
    264                 return self.zip.read(key)
    265         else:
--> 266             raise KeyError("%s is not a file in the archive" % key)
    267 
    268 

KeyError: 'num_segment is not a file in the archive'

3. I saved the cached recording with a pickle file. is their a way to read this cached recording in the new version?

Thanks,
Ben

alejoe91, 2021-05-24:
Hi @benefron 

The new API breaks support for the old version. Could you try run the entire pipeline with the new API? The syntax is very similar: https://spikeinterface.readthedocs.io/en/latest/getting_started/plot_getting_started.html

benefron, 2021-05-24:
I tried.
I still run into the same problem with the recording extractor.
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-18-2965e3b479b3> in <module>
----> 1 recording = se.OpenEphysBinaryRecordingExtractor(recording_folder)

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/extractors/neoextractors/openephys.py in __init__(self, folder_path, stream_id)
     59     def __init__(self, folder_path, stream_id=None):
     60         neo_kwargs = {'dirname' : folder_path}
---> 61         NeoBaseRecordingExtractor.__init__(self, stream_id=stream_id, **neo_kwargs)
     62 
     63 

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/extractors/neoextractors/neobaseextractor.py in __init__(self, stream_id, **neo_kwargs)
     24     def __init__(self, stream_id=None, **neo_kwargs):
     25 
---> 26         _NeoBaseExtractor.__init__(self, **neo_kwargs)
     27 
     28         # check channel

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/extractors/neoextractors/neobaseextractor.py in __init__(self, **neo_kwargs)
     12 
     13     def __init__(self, **neo_kwargs):
---> 14         neoIOclass = eval('neo.rawio.' + self.NeoRawIOClass)
     15         self.neo_reader = neoIOclass(**neo_kwargs)
     16         self.neo_reader.parse_header()

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/extractors/neoextractors/neobaseextractor.py in <module>

AttributeError: module 'neo.rawio' has no attribute 'OpenEphysBinaryRawIO'

alejoe91, 2021-05-24:
Did you install neo from sources? https://spikeinterface.readthedocs.io/en/latest/installation.html

benefron, 2021-05-24:
Yes. I followed the directions exectly. I will check on a different system cause maybe it is something with the cluster.
What version neo is suppose to be? what I see is 0.9.0

benefron, 2021-05-24:
OK
I unistalled it using pip and reinstalled it from source and know it is working
thanks

samuelgarcia, 2021-05-25:
Hi Ben.
Sorry for this ?
Yes neo from source is still 0.9.0 even it should 0.10.0.dev0

benefron, 2021-06-02:
So there is no more need to cache recordings?

samuelgarcia, 2021-06-02:
we have a `recording.save()` instead.
But depending what you do, sometimes it is not relevant anymore to "cache" with save().
For instance for `extract_waveform()` if the recording iput is not saved before but lazy it can be also super fast.
This will depend on your IOs bandwith and your CPU.
Make some benchmark.

benefron, 2021-06-02:
I tried to save
and I get the following
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-171-83fcd8af09e1> in <module>
----> 1 recording_cmr.save_to_folder(folder = ''.join([wrk_folder ,'recording_FVB1015.json']))

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in save_to_folder(self, name, folder, dump_ext, verbose, **save_kargs)
    559 
    560         # save data (done the subclass)
--> 561         cached = self._save(folder=folder, verbose=verbose, **save_kargs)
    562 
    563         # copy properties/

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/baserecording.py in _save(self, format, **save_kwargs)
    132 
    133             job_kwargs = {k: save_kwargs[k] for k in self._job_keys if k in save_kwargs}
--> 134             write_binary_recording(self, files_path=files_path, dtype=dtype, **job_kwargs)
    135 
    136             from .binaryrecordingextractor import BinaryRecordingExtractor

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/core_tools.py in write_binary_recording(recording, files_path, dtype, add_file_extension, verbose, byte_offset, **job_kwargs)
    184     executor = ChunkRecordingExecutor(recording, func, init_func, init_args, verbose=verbose,
    185                                       job_name='write_binary_recording', **job_kwargs)
--> 186     executor.run()
    187 
    188 

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/job_tools.py in run(self)
    164                 all_chunks = tqdm(all_chunks, ascii=True, desc=self.job_name)
    165 
--> 166             worker_ctx = self.init_func(*self.init_args)
    167             for segment_index, frame_start, frame_stop in all_chunks:
    168                 res = self.func(segment_index, frame_start, frame_stop, worker_ctx)

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/core_tools.py in _init_binary_worker(recording, rec_memmaps, dtype)
    104     if isinstance(recording, dict):
    105         from spikeinterface.core import load_extractor
--> 106         worker_ctx['recording'] = load_extractor(recording)
    107     else:
    108         worker_ctx['recording'] = recording

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in load_extractor(file_or_folder_or_dict, base_folder)
    720     """
    721     if isinstance(file_or_folder_or_dict, dict):
--> 722         return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
    723     else:
    724         return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in from_dict(d, base_folder)
    295             assert base_folder is not None, 'When  relative_paths=True, need to provide base_folder'
    296             d = _make_paths_absolute(d, base_folder)
--> 297         extractor = _load_extractor_from_dict(d)
    298         return extractor
    299 

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    651 
    652         if isinstance(v, dict) and is_dict_extractor(v):
--> 653             kwargs[k] = _load_extractor_from_dict(v)
    654 
    655     # handle list of extractors list

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    668 
    669     # instantiate extrator object
--> 670     extractor = cls(**kwargs)
    671 
    672     extractor._annotations.update(dic['annotations'])

TypeError: __init__() missing 2 required positional arguments: 'recording' and 'list_triggers'

samuelgarcia, 2021-06-02:
I guess you are using `common_reference()` no ?
I will fix it.

benefron, 2021-06-02:
yep.
I tried also on the recording without common_refernce(). 
and I get:
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-174-5617050636fc> in <module>
----> 1 recording_f.save(folder = ''.join([wrk_folder ,'recording_FVB10112.json']))

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in save(self, **kargs)
    478             return self.save_to_memory(**kargs)
    479         else:
--> 480             return self.save_to_folder(**kargs)
    481 
    482     def save_to_memory(self, **kargs):

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in save_to_folder(self, name, folder, dump_ext, verbose, **save_kargs)
    559 
    560         # save data (done the subclass)
--> 561         cached = self._save(folder=folder, verbose=verbose, **save_kargs)
    562 
    563         # copy properties/

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/baserecording.py in _save(self, format, **save_kwargs)
    132 
    133             job_kwargs = {k: save_kwargs[k] for k in self._job_keys if k in save_kwargs}
--> 134             write_binary_recording(self, files_path=files_path, dtype=dtype, **job_kwargs)
    135 
    136             from .binaryrecordingextractor import BinaryRecordingExtractor

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/core_tools.py in write_binary_recording(recording, files_path, dtype, add_file_extension, verbose, byte_offset, **job_kwargs)
    184     executor = ChunkRecordingExecutor(recording, func, init_func, init_args, verbose=verbose,
    185                                       job_name='write_binary_recording', **job_kwargs)
--> 186     executor.run()
    187 
    188 

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/job_tools.py in run(self)
    164                 all_chunks = tqdm(all_chunks, ascii=True, desc=self.job_name)
    165 
--> 166             worker_ctx = self.init_func(*self.init_args)
    167             for segment_index, frame_start, frame_stop in all_chunks:
    168                 res = self.func(segment_index, frame_start, frame_stop, worker_ctx)

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/core_tools.py in _init_binary_worker(recording, rec_memmaps, dtype)
    104     if isinstance(recording, dict):
    105         from spikeinterface.core import load_extractor
--> 106         worker_ctx['recording'] = load_extractor(recording)
    107     else:
    108         worker_ctx['recording'] = recording

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in load_extractor(file_or_folder_or_dict, base_folder)
    720     """
    721     if isinstance(file_or_folder_or_dict, dict):
--> 722         return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
    723     else:
    724         return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in from_dict(d, base_folder)
    295             assert base_folder is not None, 'When  relative_paths=True, need to provide base_folder'
    296             d = _make_paths_absolute(d, base_folder)
--> 297         extractor = _load_extractor_from_dict(d)
    298         return extractor
    299 

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    651 
    652         if isinstance(v, dict) and is_dict_extractor(v):
--> 653             kwargs[k] = _load_extractor_from_dict(v)
    654 
    655     # handle list of extractors list

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    651 
    652         if isinstance(v, dict) and is_dict_extractor(v):
--> 653             kwargs[k] = _load_extractor_from_dict(v)
    654 
    655     # handle list of extractors list

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    668 
    669     # instantiate extrator object
--> 670     extractor = cls(**kwargs)
    671 
    672     extractor._annotations.update(dic['annotations'])

TypeError: __init__() missing 1 required positional argument: 'folder_path'

samuelgarcia, 2021-06-02:
No I think this is maybe remove_artifacts().

Can you send me all your code for the preprocessing ?
I guess one of then, do something wring with internals kwargs.

samuelgarcia, 2021-06-02:
can you try with this PR #168  ?

benefron, 2021-06-02:
recording = se.OpenEphysBinaryRecordingExtractor(recording_folder)

probe = read_prb('/home/labs/lampllab/befron/Jan_March2021/A2x16-10mm-50-500-177-A32.prb')

recording_prb = recording.set_probegroup(probe,group_mode='by_shank')

recording_f = st.preprocessing.bandpass_filter(recording_prb, freq_min=300, freq_max=6000)

recording_clean = st.preprocessing.remove_artifacts(recording_f,list_triggers = artifacts,ms_before=3000,ms_after=artifact_window[1])

recording_cmr = st.preprocessing.common_reference(recording_clean,reference='global')

recording_cmr.save_to_folder(folder = ''.join([wrk_folder ,'recording_FVB10113.json']),name='none')

benefron, 2021-06-02:
I run it on a university cluster and I don't have gitHub CLI. not sure how to install it.

benefron, 2021-06-02:
Tried on my personal computer without any preprocessing I get the following:
TypeError                                 Traceback (most recent call last)
<ipython-input-5-5835934781e7> in <module>
----> 1 recording.save(folder = './test111')

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in save(self, **kargs)
    439             return self.save_to_memory(**kargs)
    440         else:
--> 441             return self.save_to_folder(**kargs)
    442 
    443     def save_to_memory(self, **kargs):

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in save_to_folder(self, name, folder, dump_ext, verbose, **save_kargs)
    521 
    522         # save data (done the subclass)
--> 523         cached = self._save(folder=folder, verbose=verbose, **save_kargs)
    524 
    525         # copy properties/

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/baserecording.py in _save(self, format, **save_kwargs)
    131 
    132             job_kwargs = {k: save_kwargs[k] for k in self._job_keys if k in save_kwargs}
--> 133             write_binary_recording(self, files_path=files_path, dtype=dtype, **job_kwargs)
    134 
    135             from .binaryrecordingextractor import BinaryRecordingExtractor

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/core_tools.py in write_binary_recording(recording, files_path, dtype, verbose, **job_kwargs)
    179     executor = ChunkRecordingExecutor(recording, func, init_func, init_args, verbose=verbose,
    180                                       job_name='write_binary_recording', **job_kwargs)
--> 181     executor.run()
    182 
    183 

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/job_tools.py in run(self)
    164                 all_chunks = tqdm(all_chunks, ascii=True, desc=self.job_name)
    165 
--> 166             worker_ctx = self.init_func(*self.init_args)
    167             for segment_index, frame_start, frame_stop in all_chunks:
    168                 res = self.func(segment_index, frame_start, frame_stop, worker_ctx)

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/core_tools.py in _init_binary_worker(recording, rec_memmaps, dtype)
    104     if isinstance(recording, dict):
    105         from spikeinterface.core import load_extractor
--> 106         worker_ctx['recording'] = load_extractor(recording)
    107     else:
    108         worker_ctx['recording'] = recording

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in load_extractor(file_or_folder_or_dict)
    635     """
    636     if isinstance(file_or_folder_or_dict, dict):
--> 637         return BaseExtractor.from_dict(file_or_folder_or_dict)
    638     else:
    639         return BaseExtractor.load(file_or_folder_or_dict)

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in from_dict(d)
    272             The loaded extractor object
    273         '''
--> 274         extractor = _load_extractor_from_dict(d)
    275         return extractor
    276 

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    583 
    584     # instantiate extrator object
--> 585     extractor = cls(**kwargs)
    586 
    587     extractor._annotations.update(dic['annotations'])

TypeError: __init__() missing 1 required positional argument: 'folder_path'

benefron, 2021-06-02:
I get the same error while trying to run sorters:
  File "/home/labs/lampllab/befron/FVB/FVB101.py", line 91, in <module>
    sort_klusta = ss.run_klusta(recording_cmr,delete_output_folder=True,verbose=True,output_folder=''.join([wrk_folder,'/klusta']))
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 73, in run_klusta
    return run_sorter('klusta', *args, **kwargs)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 50, in run_sorter
    SorterClass.setup_recording(recording, output_folder, verbose=verbose)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/basesorter.py", line 174, in setup_recording
    cls._setup_recording(recording, output_folder, sorter_params, verbose)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/klusta/klusta.py", line 107, in _setup_recording
    n_jobs=p["n_jobs_bin"], verbose=False, progress_bar=verbose)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/binaryrecordingextractor.py", line 68, in write_recording
    write_binary_recording(recording, files_path=files_path,  dtype=dtype, **job_kwargs)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/core_tools.py", line 186, in write_binary_recording
    executor.run()
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/job_tools.py", line 166, in run
    worker_ctx = self.init_func(*self.init_args)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/core_tools.py", line 106, in _init_binary_worker
    worker_ctx['recording'] = load_extractor(recording)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 722, in load_extractor
    return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 297, in from_dict
    extractor = _load_extractor_from_dict(d)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 653, in _load_extractor_from_dict
    kwargs[k] = _load_extractor_from_dict(v) 
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 670, in _load_extractor_from_dict
    extractor = cls(**kwargs)
TypeError: __init__() missing 2 required positional arguments: 'recording' and 'list_triggers'

samuelgarcia, 2021-06-02:
I think I fixed it.
kwargs where wrong for openephys reader

you can do to test with gh-cli, until my PR will be merge:
```
# activate your pythion env
# then
git clone https://github.com/samuelgarcia/spikeinterface.git
pip uninstall spikeinterface
cd spikeinterface
git check big_refactoring
python setup.py install # or python setup.py develop
```

benefron, 2021-06-02:
Thanks
I will check it tomorrow and let you know

benefron, 2021-06-03:
I get a new error:
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-17-1a6e542b347f> in <module>
----> 1 recording_cmr.save(folder = ''.join([wrk_folder ,'recording_FVB10118.json']))

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in save(self, **kargs)
    478             return self.save_to_memory(**kargs)
    479         else:
--> 480             return self.save_to_folder(**kargs)
    481 
    482     def save_to_memory(self, **kargs):

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in save_to_folder(self, name, folder, dump_ext, verbose, **save_kargs)
    544         provenance_file = folder / f'provenance.{dump_ext}'
    545         if self.check_if_dumpable():
--> 546             self.dump(provenance_file)
    547         else:
    548             provenance_file.write_text(

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in dump(self, file_path, relative_to)
    357         """
    358         if str(file_path).endswith('.json'):
--> 359             self.dump_to_json(file_path, relative_to)
    360         elif str(file_path).endswith('.pkl') or str(file_path).endswith('.pickle'):
    361             self.dump_to_pickle(file_path, relative_to)

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py in dump_to_json(self, file_path, relative_to)
    379         file_path = self._get_file_path(file_path, ['.json'])
    380         file_path.write_text(
--> 381             json.dumps(check_json(dump_dict), indent=4),
    382             encoding='utf8'
    383         )

/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/__init__.py in dumps(obj, skipkeys, ensure_ascii, check_circular, allow_nan, cls, indent, separators, default, sort_keys, **kw)
    236         check_circular=check_circular, allow_nan=allow_nan, indent=indent,
    237         separators=separators, default=default, sort_keys=sort_keys,
--> 238         **kw).encode(obj)
    239 
    240 

/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py in encode(self, o)
    199         chunks = self.iterencode(o, _one_shot=True)
    200         if not isinstance(chunks, (list, tuple)):
--> 201             chunks = list(chunks)
    202         return ''.join(chunks)
    203 

/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py in _iterencode(o, _current_indent_level)
    429             yield from _iterencode_list(o, _current_indent_level)
    430         elif isinstance(o, dict):
--> 431             yield from _iterencode_dict(o, _current_indent_level)
    432         else:
    433             if markers is not None:

/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py in _iterencode_dict(dct, _current_indent_level)
    403                 else:
    404                     chunks = _iterencode(value, _current_indent_level)
--> 405                 yield from chunks
    406         if newline_indent is not None:
    407             _current_indent_level -= 1

/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py in _iterencode_dict(dct, _current_indent_level)
    403                 else:
    404                     chunks = _iterencode(value, _current_indent_level)
--> 405                 yield from chunks
    406         if newline_indent is not None:
    407             _current_indent_level -= 1

/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py in _iterencode_dict(dct, _current_indent_level)
    403                 else:
    404                     chunks = _iterencode(value, _current_indent_level)
--> 405                 yield from chunks
    406         if newline_indent is not None:
    407             _current_indent_level -= 1

/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py in _iterencode_dict(dct, _current_indent_level)
    403                 else:
    404                     chunks = _iterencode(value, _current_indent_level)
--> 405                 yield from chunks
    406         if newline_indent is not None:
    407             _current_indent_level -= 1

/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py in _iterencode(o, _current_indent_level)
    436                     raise ValueError("Circular reference detected")
    437                 markers[markerid] = o
--> 438             o = _default(o)
    439             yield from _iterencode(o, _current_indent_level)
    440             if markers is not None:

/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py in default(self, o)
    177 
    178         """
--> 179         raise TypeError(f'Object of type {o.__class__.__name__} '
    180                         f'is not JSON serializable')
    181 

TypeError: Object of type uint16 is not JSON serializable

benefron, 2021-06-03:
Another error I get when trying to run the sorters:
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-39-b46dab75a482> in <module>
----> 1 recording_cmr.get_channel_locations()

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/baserecording.py in get_channel_locations(self, channel_ids, locations_2d)
    343         channel_indices = self.ids_to_indices(channel_ids)
    344         if self.get_property('contact_vector') is not None:
--> 345             probe = self.get_probe()
    346             return probe.contact_positions[channel_indices]
    347         else:

~/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/baserecording.py in get_probe(self)
    297     def get_probe(self):
    298         probes = self.get_probes()
--> 299         assert len(probes) == 1, 'there are several probe use .get_probes() or get_probegroup()'
    300         return probes[0]
    301 

AssertionError: there are several probe use .get_probes() or get_probegroup()

I am using a .prb file

samuelgarcia, 2021-06-03:
For the first error my gurss is that, it comes from `remove_artifacts()` could you try to `save()` just after to confirm ?
Or even better save() at each steps for debugging.

Note, you can use `save(n_jobs=10, chunk_memory='100M')` to be faster.

For the second error.
PRB is totally ambiguous for groups!!
Wer can't known if it is 2 shanks or 2 probes.
`probeinterface.read_prb()` give 2 probes when 2 groups.
You can merge sereval probes into one multi shank probe using `probeinterface.combine_probes()`
It should solve your problem.

If it is really 2 probes, you need to split the recording into subrecording for sorting because the channel position is by probe.
Check this : https://spikeinterface.readthedocs.io/en/latest/modules/sorters/plot_3_sorting_by_group.html
and also this
https://spikeinterface.readthedocs.io/en/latest/modules/core/plot_1_recording_extractor.html#sphx-glr-modules-core-plot-1-recording-extractor-py

benefron, 2021-06-03:
The probe I moved into a single probe. The problem is that whenever I read .prb or .json probe files it reads them to probegroup.
The first problem appears even if I just try to save straight after loading the openephys recording

benefron, 2021-06-03:
Sorry it does work now after loading the recording.
I am checking now after assigning probe and than after filtering.
I get the error after removing artifacts

samuelgarcia, 2021-06-03:
Yes, when you have a probegroup, you can recombinate then into a multi shank if it is the case see the end of the titorial:
https://probeinterface.readthedocs.io/en/main/examples/ex_04_multi_shank_probe.html

Did you pull from my working branch *?

samuelgarcia, 2021-06-03:
OK thanks. I will fix it very soon.

samuelgarcia, 2021-06-03:
I have just push a commit in my branch to fix this, can you try to git pull again please ?

benefron, 2021-06-03:
I get a new error:
Traceback (most recent call last):
  File "/home/labs/lampllab/befron/FVB/FVB101.py", line 92, in <module>
    recording_clean = st.preprocessing.remove_artifacts(recording_f,list_triggers = artifacts,ms_before=3000,ms_after=artifact_window[1])
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 208, in remove_artifacts
    return RemoveArtifactsRecording(*args, **kwargs)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in __init__
    self._kwargs = dict(recording=recording.to_dict(), list_triggers=[int(trig) for trig in list_triggers], 
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in <listcomp>
    self._kwargs = dict(recording=recording.to_dict(), list_triggers=[int(trig) for trig in list_triggers], 
TypeError: int() argument must be a string, a bytes-like object or a number, not 'list'
Traceback (most recent call last):
  File "/home/labs/lampllab/befron/FVB/FVB101.py", line 92, in <module>
    recording_clean = st.preprocessing.remove_artifacts(recording_f,list_triggers = artifacts,ms_before=3000,ms_after=artifact_window[1])
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 208, in remove_artifacts
    return RemoveArtifactsRecording(*args, **kwargs)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in __init__
    self._kwargs = dict(recording=recording.to_dict(), list_triggers=[int(trig) for trig in list_triggers], 
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in <listcomp>
    self._kwargs = dict(recording=recording.to_dict(), list_triggers=[int(trig) for trig in list_triggers], 
TypeError: int() argument must be a string, a bytes-like object or a number, not 'list'
Traceback (most recent call last):
  File "/home/labs/lampllab/befron/FVB/FVB101.py", line 92, in <module>
    recording_clean = st.preprocessing.remove_artifacts(recording_f,list_triggers = artifacts,ms_before=3000,ms_after=artifact_window[1])
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 208, in remove_artifacts
    return RemoveArtifactsRecording(*args, **kwargs)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in __init__
    self._kwargs = dict(recording=recording.to_dict(), list_triggers=[int(trig) for trig in list_triggers], 
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in <listcomp>
    self._kwargs = dict(recording=recording.to_dict(), list_triggers=[int(trig) for trig in list_triggers], 
TypeError: int() argument must be a string, a bytes-like object or a number, not 'list'
Traceback (most recent call last):
  File "/home/labs/lampllab/befron/FVB/FVB101.py", line 92, in <module>
    recording_clean = st.preprocessing.remove_artifacts(recording_f,list_triggers = artifacts,ms_before=3000,ms_after=artifact_window[1])
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 208, in remove_artifacts
    return RemoveArtifactsRecording(*args, **kwargs)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in __init__
    self._kwargs = dict(recording=recording.to_dict(), list_triggers=[int(trig) for trig in list_triggers], 
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in <listcomp>
    self._kwargs = dict(recording=recording.to_dict(), list_triggers=[int(trig) for trig in list_triggers], 
TypeError: int() argument must be a string, a bytes-like object or a number, not 'list'
Traceback (most recent call last):
  File "/home/labs/lampllab/befron/FVB/FVB101.py", line 92, in <module>
    recording_clean = st.preprocessing.remove_artifacts(recording_f,list_triggers = artifacts,ms_before=3000,ms_after=artifact_window[1])
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 208, in remove_artifacts
    return RemoveArtifactsRecording(*args, **kwargs)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in __init__
[... cut ...]
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in __init__
    self._kwargs = dict(recording=recording.to_dict(), list_triggers=[int(trig) for trig in list_triggers], 
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/preprocessing/remove_artifacts.py", line 85, in <listcomp>
    self._kwargs = dict(recording=recording.to_dict(), list_triggers=[int(trig) for trig in list_triggers], 
TypeError: int() argument must be a string, a bytes-like object or a number, not 'list'

samuelgarcia, 2021-06-03:
Oups. Sorry I fixed it too quickly.
I made a new fix. Can you try again.
Really sorry for this durty debug.

benefron, 2021-06-03:
I still get
Traceback (most recent call last):
  File "/home/labs/lampllab/befron/FVB/FVB101.py", line 120, in <module>
    recording_saved = recording_cmr.save_to_folder(folder = ''.join([wrk_folder ,'recording_FVB101']),dump_ext='json',n_jobs=10,chunk_memory='100M')
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 546, in save_to_folder
    self.dump(provenance_file)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 359, in dump
    self.dump_to_json(file_path, relative_to)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 381, in dump_to_json
    json.dumps(check_json(dump_dict), indent=4),
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py", line 201, in encode
    chunks = list(chunks)
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  [Previous line repeated 1 more time]
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py", line 438, in _iterencode
    o = _default(o)
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type uint16 is not JSON serializable

benefron, 2021-06-03:
I am know saving as pkl and it works.
I got another error while running sorters:
Traceback (most recent call last):
  File "/home/labs/lampllab/befron/FVB/FVB101.py", line 111, in <module>
    sort_kilo = ss.run_kilosort3(recording_saved,delete_output_folder=True,verbose=True,output_folder=''.join([wrk_folder,'/kilo']))
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 226, in run_kilosort3
    return run_sorter('kilosort3', *args, **kwargs)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 54, in run_sorter
    verbose=verbose, raise_error=raise_error, **sorter_params)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 74, in run_sorter_local
    SorterClass.setup_recording(recording, output_folder, verbose=verbose)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/basesorter.py", line 174, in setup_recording
    cls._setup_recording(recording, output_folder, sorter_params, verbose)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/kilosort3/kilosort3.py", line 158, in _setup_recording
    with (source_dir / 'kilosort3_master.m').open('r') as f:
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/pathlib.py", line 1203, in open
    opener=self._opener)
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/pathlib.py", line 1058, in _opener
    return self._accessor.open(self, flags, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/kilosort3/kilosort3_master.m'
Traceback (most recent call last):
  File "/home/labs/lampllab/befron/FVB/FVB101.py", line 121, in <module>
    sort_klusta = ss.run_klusta(recording_saved,delete_output_folder=True,verbose=True,output_folder=''.join([wrk_folder,'/klusta']))
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 191, in run_klusta
    return run_sorter('klusta', *args, **kwargs)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 54, in run_sorter
    verbose=verbose, raise_error=raise_error, **sorter_params)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 74, in run_sorter_local
    SorterClass.setup_recording(recording, output_folder, verbose=verbose)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/basesorter.py", line 174, in setup_recording
    cls._setup_recording(recording, output_folder, sorter_params, verbose)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/klusta/klusta.py", line 118, in _setup_recording
    with (source_dir / 'config_default.prm').open('r') as f:
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/pathlib.py", line 1203, in open
    opener=self._opener)
  File "/apps/RH7U2/gnu/anaconda/2020.02/python/3.7/lib/python3.7/pathlib.py", line 1058, in _opener
    return self._accessor.open(self, flags, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/klusta/config_default.prm'

and another error while running waveclus:
  File "/home/labs/lampllab/befron/FVB/FVB101.py", line 107, in <module>
    sort_wave = ss.run_waveclus(recording_saved,delete_output_folder=True,verbose=True,output_folder=''.join([wrk_folder,'/wave']))
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 241, in run_waveclus
    return run_sorter('waveclus', *args, **kwargs)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 54, in run_sorter
    verbose=verbose, raise_error=raise_error, **sorter_params)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py", line 74, in run_sorter_local
    SorterClass.setup_recording(recording, output_folder, verbose=verbose)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/basesorter.py", line 174, in setup_recording
    cls._setup_recording(recording, output_folder, sorter_params, verbose)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/waveclus/waveclus.py", line 162, in _setup_recording
    num_timepoints = recording.get_num_frames()
TypeError: get_num_samples() missing 1 required positional argument: 'segment_index'

samuelgarcia, 2021-06-03:
I fixed the second error (aveclust) in my branch.
Please git pull again.

For the first one, I will do it later, we need to force installation of all files.
At the moment you can fix it with
```
python setup.py develop
```
do a pip uninstall before.

@alejoe91 : how we copy all *.m and *.prm in spikesorter for installation ?

alejoe91, 2021-06-03:
@samuelgarcia you need to add the files to the manifest (https://github.com/SpikeInterface/spikesorters/blob/master/MANIFEST.in) and `include_package_data` in the `seup.py`: https://github.com/SpikeInterface/spikesorters/blob/master/setup.py#L21

samuelgarcia, 2021-06-03:
OK thanks.

benefron, 2021-06-03:
I was also wondering.
I am doing some of the work on a cluster and some on my PC.
When I dump the recording it saves in the .pkl file the pathway based on the system I am in, and than I can't access it from the other system.
Is there an easy fix for the pathway?

alejoe91, 2021-06-03:
If you save to folder instead of pkl the paths will be relative to the folder itself, so it can be moved between different computers. Alternatively, when you save to pkl you can use the `relative_to` arg to make all file paths relative to a certain folder. In when you load the recording you will then need to pass the `base_folder` path

benefron, 2021-06-03:
I am trying to export to phy and I get the following:
Traceback (most recent call last):
  File "/home/labs/lampllab/befron/FVB/FVB101_300521/FVB101_phy.py", line 47, in <module>
    verbose=True, n_jobs = -1,chunk_memory='100M',copy_binary=False,waveform_extractor = we,compute_pc_features=False,remove_if_exists=True)
  File "/home/labs/lampllab/befron/.local/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/toolkit/postprocessing/export_to_phy.py", line 105, in export_to_phy
    elif isinstance(recording, se.CacheRecordingExtractor):
AttributeError: module 'spikeinterface.extractors' has no attribute 'CacheRecordingExtractor'

samuelgarcia, 2021-06-03:
Oups. Sorry.
I bever tryed this without copy.
I have just push the fix on my branch.

benefron, 2021-06-06:
I still get:
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-13-e7a065024e05> in <module>
----> 1 recording_cached = recording_cmr.save_to_folder(folder = '~/Recording_FVB101',n_jobs = -1,chunk_memory='100M')

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in save_to_folder(self, name, folder, dump_ext, verbose, **save_kargs)
    544         provenance_file = folder / f'provenance.{dump_ext}'
    545         if self.check_if_dumpable():
--> 546             self.dump(provenance_file)
    547         else:
    548             provenance_file.write_text(

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in dump(self, file_path, relative_to)
    357         """
    358         if str(file_path).endswith('.json'):
--> 359             self.dump_to_json(file_path, relative_to)
    360         elif str(file_path).endswith('.pkl') or str(file_path).endswith('.pickle'):
    361             self.dump_to_pickle(file_path, relative_to)

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in dump_to_json(self, file_path, relative_to)
    379         file_path = self._get_file_path(file_path, ['.json'])
    380         file_path.write_text(
--> 381             json.dumps(check_json(dump_dict), indent=4),
    382             encoding='utf8'
    383         )

~/anaconda3/lib/python3.8/json/__init__.py in dumps(obj, skipkeys, ensure_ascii, check_circular, allow_nan, cls, indent, separators, default, sort_keys, **kw)
    232     if cls is None:
    233         cls = JSONEncoder
--> 234     return cls(
    235         skipkeys=skipkeys, ensure_ascii=ensure_ascii,
    236         check_circular=check_circular, allow_nan=allow_nan, indent=indent,

~/anaconda3/lib/python3.8/json/encoder.py in encode(self, o)
    199         chunks = self.iterencode(o, _one_shot=True)
    200         if not isinstance(chunks, (list, tuple)):
--> 201             chunks = list(chunks)
    202         return ''.join(chunks)
    203 

~/anaconda3/lib/python3.8/json/encoder.py in _iterencode(o, _current_indent_level)
    429             yield from _iterencode_list(o, _current_indent_level)
    430         elif isinstance(o, dict):
--> 431             yield from _iterencode_dict(o, _current_indent_level)
    432         else:
    433             if markers is not None:

~/anaconda3/lib/python3.8/json/encoder.py in _iterencode_dict(dct, _current_indent_level)
    403                 else:
    404                     chunks = _iterencode(value, _current_indent_level)
--> 405                 yield from chunks
    406         if newline_indent is not None:
    407             _current_indent_level -= 1

~/anaconda3/lib/python3.8/json/encoder.py in _iterencode_dict(dct, _current_indent_level)
    403                 else:
    404                     chunks = _iterencode(value, _current_indent_level)
--> 405                 yield from chunks
    406         if newline_indent is not None:
    407             _current_indent_level -= 1

~/anaconda3/lib/python3.8/json/encoder.py in _iterencode_dict(dct, _current_indent_level)
    403                 else:
    404                     chunks = _iterencode(value, _current_indent_level)
--> 405                 yield from chunks
    406         if newline_indent is not None:
    407             _current_indent_level -= 1

~/anaconda3/lib/python3.8/json/encoder.py in _iterencode_dict(dct, _current_indent_level)
    403                 else:
    404                     chunks = _iterencode(value, _current_indent_level)
--> 405                 yield from chunks
    406         if newline_indent is not None:
    407             _current_indent_level -= 1

~/anaconda3/lib/python3.8/json/encoder.py in _iterencode(o, _current_indent_level)
    436                     raise ValueError("Circular reference detected")
    437                 markers[markerid] = o
--> 438             o = _default(o)
    439             yield from _iterencode(o, _current_indent_level)
    440             if markers is not None:

~/anaconda3/lib/python3.8/json/encoder.py in default(self, o)
    177 
    178         """
--> 179         raise TypeError(f'Object of type {o.__class__.__name__} '
    180                         f'is not JSON serializable')
    181 

TypeError: Object of type uint16 is not JSON serializable

benefron, 2021-06-06:
I still have the same problem with the e.json and uint16 file.
Also when I try to run export to phy and other processes in parallel I get the following:
---------------------------------------------------------------------------
BrokenProcessPool                         Traceback (most recent call last)
<ipython-input-20-31035dd140c8> in <module>
----> 1 st.export_to_phy(recording_cached, agreement_sorting, '/home/ben/Z/FVB101_300521/Sorting/phy', we,
      2             compute_pc_features=True, compute_amplitudes=True,n_jobs = -1,chunk_memory='100M' )

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/toolkit/postprocessing/export_to_phy.py in export_to_phy(recording, sorting, output_folder, waveform_extractor, compute_pc_features, compute_amplitudes, max_channels_per_template, copy_binary, remove_if_exists, peak_sign, dtype, verbose, **job_kwargs)
    160 
    161     if compute_amplitudes:
--> 162         amplitudes = get_unit_amplitudes(waveform_extractor,  peak_sign=peak_sign, outputs='concatenated', **job_kwargs)
    163         # one segment only
    164         amplitudes = amplitudes[0][:, np.newaxis]

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/toolkit/postprocessing/unit_amplitudes.py in get_unit_amplitudes(waveform_extractor, peak_sign, outputs, **job_kwargs)
     36     processor = ChunkRecordingExecutor(recording, func, init_func, init_args,
     37                         handle_returns=True, job_name='extract amplitudes',  **job_kwargs)
---> 38     out = processor.run()
     39     amps, segments = zip(*out)
     40     amps = np.concatenate(amps)

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/job_tools.py in run(self)
    194 
    195             if self.handle_returns:
--> 196                 for res in results:
    197                     returns.append(res)
    198             else:

~/anaconda3/lib/python3.8/concurrent/futures/process.py in _chain_from_iterable_of_lists(iterable)
    482     careful not to keep references to yielded objects.
    483     """
--> 484     for element in iterable:
    485         element.reverse()
    486         while element:

~/anaconda3/lib/python3.8/concurrent/futures/_base.py in result_iterator()
    609                     # Careful not to keep a reference to the popped future
    610                     if timeout is None:
--> 611                         yield fs.pop().result()
    612                     else:
    613                         yield fs.pop().result(end_time - time.monotonic())

~/anaconda3/lib/python3.8/concurrent/futures/_base.py in result(self, timeout)
    437                 raise CancelledError()
    438             elif self._state == FINISHED:
--> 439                 return self.__get_result()
    440             else:
    441                 raise TimeoutError()

~/anaconda3/lib/python3.8/concurrent/futures/_base.py in __get_result(self)
    386     def __get_result(self):
    387         if self._exception:
--> 388             raise self._exception
    389         else:
    390             return self._result

BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending.

benefron, 2021-06-06:
I also get the following while exporting to phy:
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-21-6eb60eddd2d6> in <module>
----> 1 st.export_to_phy(recording_cached, agreement_sorting, '/home/ben/Z/FVB101_300521/Sorting/phy', we,
      2             compute_pc_features=True, compute_amplitudes=True)

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/toolkit/postprocessing/export_to_phy.py in export_to_phy(recording, sorting, output_folder, waveform_extractor, compute_pc_features, compute_amplitudes, max_channels_per_template, copy_binary, remove_if_exists, peak_sign, dtype, verbose, **job_kwargs)
    160 
    161     if compute_amplitudes:
--> 162         amplitudes = get_unit_amplitudes(waveform_extractor,  peak_sign=peak_sign, outputs='concatenated', **job_kwargs)
    163         # one segment only
    164         amplitudes = amplitudes[0][:, np.newaxis]

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/toolkit/postprocessing/unit_amplitudes.py in get_unit_amplitudes(waveform_extractor, peak_sign, outputs, **job_kwargs)
     36     processor = ChunkRecordingExecutor(recording, func, init_func, init_args,
     37                         handle_returns=True, job_name='extract amplitudes',  **job_kwargs)
---> 38     out = processor.run()
     39     amps, segments = zip(*out)
     40     amps = np.concatenate(amps)

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/job_tools.py in run(self)
    164                 all_chunks = tqdm(all_chunks, ascii=True, desc=self.job_name)
    165 
--> 166             worker_ctx = self.init_func(*self.init_args)
    167             for segment_index, frame_start, frame_stop in all_chunks:
    168                 res = self.func(segment_index, frame_start, frame_stop, worker_ctx)

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/toolkit/postprocessing/unit_amplitudes.py in _init_worker_unit_amplitudes(recording, sorting, extremum_channels_index, peak_shifts)
     68     if isinstance(sorting, dict):
     69         from spikeinterface.core import load_extractor
---> 70         sorting = load_extractor(sorting)
     71     worker_ctx['recording'] = recording
     72     worker_ctx['sorting'] = sorting

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in load_extractor(file_or_folder_or_dict, base_folder)
    720     """
    721     if isinstance(file_or_folder_or_dict, dict):
--> 722         return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
    723     else:
    724         return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in from_dict(d, base_folder)
    295             assert base_folder is not None, 'When  relative_paths=True, need to provide base_folder'
    296             d = _make_paths_absolute(d, base_folder)
--> 297         extractor = _load_extractor_from_dict(d)
    298         return extractor
    299 

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    668 
    669     # instantiate extrator object
--> 670     extractor = cls(**kwargs)
    671 
    672     extractor._annotations.update(dic['annotations'])

TypeError: __init__() missing 2 required positional arguments: 'sampling_frequency' and 'multisortingcomparison'

benefron, 2021-06-06:
And when I already have the waveform folder I get the following:
folder = '/home/ben/Z/FVB101_300521/Sorting/waveforms'

we = si.extract_waveforms(recording_cached, agreement_sorting, folder,load_if_exists=True)

print(we)

​

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-22-036d997f96dd> in <module>
      1 folder = '/home/ben/Z/FVB101_300521/Sorting/waveforms'
----> 2 we = si.extract_waveforms(recording_cached, agreement_sorting, folder,load_if_exists=True)
      3 print(we)

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/waveform_extractor.py in extract_waveforms(recording, sorting, folder, load_if_exists, ms_before, ms_after, max_spikes_per_unit, dtype, **job_kwargs)
    448     folder = Path(folder)
    449     if load_if_exists and folder.is_dir():
--> 450         we = WaveformExtractor.load_from_folder(folder)
    451     else:
    452         we = WaveformExtractor.create(recording, sorting, folder)

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/waveform_extractor.py in load_from_folder(cls, folder)
     84         folder = Path(folder)
     85         recording = load_extractor(folder / 'recording.json')
---> 86         sorting = load_extractor(folder / 'sorting.json')
     87         we = cls(recording, sorting, folder)
     88         return we

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in load_extractor(file_or_folder_or_dict, base_folder)
    722         return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
    723     else:
--> 724         return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
    725 
    726 

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in load(file_path, base_folder)
    428             else:
    429                 raise ValueError(f'Impossible to load {file_path}')
--> 430             extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
    431             return extractor
    432 

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in from_dict(d, base_folder)
    295             assert base_folder is not None, 'When  relative_paths=True, need to provide base_folder'
    296             d = _make_paths_absolute(d, base_folder)
--> 297         extractor = _load_extractor_from_dict(d)
    298         return extractor
    299 

~/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    668 
    669     # instantiate extrator object
--> 670     extractor = cls(**kwargs)
    671 
    672     extractor._annotations.update(dic['annotations'])

TypeError: __init__() missing 2 required positional arguments: 'sampling_frequency' and 'multisortingcomparison'

alejoe91, 2021-06-06:
This is an agreement sorting, right?

benefron, 2021-06-06:
Yes

alejoe91, 2021-06-06:
Ok we'll try to fix it over the week!

samuelgarcia, 2021-06-07:
Hi Ben.
That is a lot of bugs!!
I wil try to fix then tomorow or maybe leter today.
Sorry for you.

Your input file appears to be dtype utin16, maybe this cause some border effect.
If you have a small fils and a short script, it would help me to debug more efficiently.

benefron, 2021-06-07:
This the format that open ephys saves binary files

samuelgarcia, 2021-06-07:
Doesn't openephys use int16 instead of uint16 ?

alejoe91, 2021-06-07:
@samuelgarcia the `AgreementSorting` is not dumpable. In that case I think that the save funciton should skip the provenance and just cache it.

alejoe91, 2021-06-07:
@benefron can you please paste here the code you are running? It's hard to understand where the uint16 error comes from otherwise

samuelgarcia, 2021-06-07:
@alejoe91 : is is already the case when `save()` if the provenance is not dumpavble then it is skiped.

alejoe91, 2021-06-07:
Then the agreement sorting is probably missing a is_dumpable=False...

samuelgarcia, 2021-06-07:
@benefron : you should save the agreement sorting to a NPZ before extract_waveform it should solve the situation partially.

benefron, 2021-06-08:
I thought it might be the data I loaded from matlab of the times of artifacts but I still get the error.
I am attaching the code. 
It happens only after the remove_artifacts. I checked and common_reference doesn't cause the uint16 error.
 """
import spikeinterface as si
import spikeinterface.extractors as se 
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw
import matplotlib.pyplot as plt
import numpy as np
import loky
import sys
import scipy.io
#%matplotlib inline
import pandas as pd
from probeinterface.plotting import plot_probe, plot_probe_group
from probeinterface import write_probeinterface, read_probeinterface
from probeinterface import write_prb, read_prb
import numpy as np
import matplotlib.pyplot as plt

from probeinterface import Probe, ProbeGroup
from probeinterface import generate_linear_probe, generate_multi_shank
from probeinterface import combine_probes
from probeinterface.plotting import plot_probe
# for running on WEXAC
home_fldr = '/home/labs/lampllab/befron/'
# for running local
#home_fldr = '/WEXAC/'
case = int(sys.argv[1])

recording_folder = ''.join([home_fldr,'FVB/FVB101_300521/']) 
prb_file = ''.join([home_fldr,'Jan_March2021/A2x16-10mm-50-500-177-A32.prb'])
wrk_folder = ''.join([home_fldr,'FVB/FVB101_300521/Sorting'])
mat_load = '/home/labs/lampllab/befron/FVB/FVB101_300521/condition_extract.mat'

ss.IronClustSorter.set_ironclust_path('/home/labs/lampllab/befron/Sorters/ironclust')
ss.WaveClusSorter.set_waveclus_path('/home/labs/lampllab/befron/Sorters/wave_clus/')
ss.Kilosort3Sorter.set_kilosort3_path('/home/labs/lampllab/befron/Sorters/Kilosort/')

recording = se.OpenEphysBinaryRecordingExtractor(recording_folder)

channel_ids = recording.get_channel_ids()
fs = recording.get_sampling_frequency()
num_chan = recording.get_num_channels()

print(f'Channel ids: {channel_ids}')
print(f'Sampling frequency: {fs}')
print(f'Number of channels: {num_chan}')

mat = scipy.io.loadmat(mat_load)

artifacts = mat['condition_extract']['all_changes'][0][0]
artifact_window = mat['condition_extract']['artifact_times'][0][0][0]

artifacts = artifacts.flatten()
artifacts = artifacts.tolist()

n = 16
positions = [[  0.,   0.],
       [  0.,  100.],
       [  0.,  200.],
       [  0.,  300.],
       [  0.,  400.],
       [  0., 500.],
       [  0., 600.],
       [  0., 700.],
       [ 0., 750.],
       [ 0.,  650.],
       [ 0.,  550.],
       [ 0.,  450.],
       [ 0.,  350.],
       [ 0.,  250.],
       [ 0., 150.],
       [ 0., 50.]]

probe = Probe(ndim=2, si_units='um')
probe.set_contacts(positions=positions, shapes='circle', shape_params={'radius': 5})
probe1 = probe.copy()
probe1.move([100, 0])
A2X16_50 = combine_probes([probe, probe1])
A2X16_50.set_device_channel_indices(np.arange(32))
recording_prb = recording.set_probe(A2X16_50)

recording_f = st.preprocessing.bandpass_filter(recording_prb, freq_min=300, freq_max=6000)

recording_clean = st.preprocessing.remove_artifacts(recording_f,list_triggers = artifacts,ms_before=3000,ms_after=artifact_window[1])

recording_cmr = st.preprocessing.common_reference(recording_clean,reference='global')
"""

benefron, 2021-06-08:
I cached and saved the sorting before running the extract_waveforms and export_to_phy.
It works now but when I open phy The feature view is completely flattened. I run the following code for export:
folder = '/home/ben/Z/FVB101_300521/Sorting/waveforms'
we = si.extract_waveforms(recording_cached, agreement, folder)
print(we)
st.export_to_phy(recording_cached, agreement_sorting, '/home/ben/Z/FVB101_300521/Sorting/phy', we,
            compute_pc_features=True, compute_amplitudes=True)

I am attaching a screenshot of the phy:
[image]

It is the same for all units

samuelgarcia, 2021-06-08:
@benefron : if you re run something you need to manually remove the cached waveform folder each times because you are maybe viewing the old cache.
Maybe it is safer to do `extract_waveforms(recording, sorting, folder, load_if_exists=False)``False is the dfault normally.

Are you using my latest branch "big_refactoring" or the latest master branch from the main repo ?

samuelgarcia, 2021-06-08:
Also : are you sure you filter **before** `extract_waveforms()` other wise waveform or done on raw signal and best channel are totally random. that could alspo explain.

samuelgarcia, 2021-06-08:
The bug you have for dumping after `remove_artifact()` is normally fix here:
https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/toolkit/preprocessing/remove_artifacts.py#L85
can you check ?

benefron, 2021-06-08:
I manually remove the waveform folder.
I am using your branch.
I filter before, but when I extract to phy it says that the recording is not filtered.
I didn't understand what should I check in the link you attached

samuelgarcia, 2021-06-08:
OK.
It was just for checking that you have locally this line.
Because it is supposed to solve the dump of `remove_artifact()`.
I will inverstigate more.

A small file and a full script would help a lot for this debuging.

[12 more replies]

closed 2021-07-22 by samuelgarcia