SpikeInterface/spikeinterface #169: Finding Proper Channels

farazmoradi, 2021-05-31:
Hi,

I want to know if the spike interface has any analysis for each channel? For instance, I want to know for each unit which channels are most important? Basically, I mean I want to know which channels were better and the information in them was more prominent than the other one.

Thanks

samuelgarcia, 2021-05-31:
Hi,
have a look to 
spikeinterface.toolkit.postprocessing.template_tools
You have:
  * `get_template_extremum_channel()` > best channel (on channel)
  * `get_template_best_channels()` > n best channels

samuelgarcia, 2021-05-31:
See also this
https://spikeinterface.readthedocs.io/en/latest/modules/toolkit/plot_2_postprocessing.html#sphx-glr-modules-toolkit-plot-2-postprocessing-py

closed 2021-07-22 by samuelgarcia