SpikeInterface/spikeinterface #172: Adding a nice & ionice argument to run_sorter [feature request]

timsainb, 2021-06-04:
I'm wondering what your thoughts are for adding an argument to control niceness/ionice with spikesorting.

I think most folks are spikesorting on shared compute servers and want to leave those sorts in the background while they do other things. Since spikesorting & the preprocessing spikeinterface does is io/cpu heavy it might be useful to set a flag to leave these sorts on in the background. 

Maybe this would be a bit difficult for multiprocessing and running different sorters though. 

```
import psutil, os
p = psutil.Process(os.getpid())

# set nice 
p.nice(psutil.BELOW_NORMAL_PRIORITY_CLASS)

# set ionice
p.ionice(psutil.IOPRIO_CLASS_IDLE)
```

samuelgarcia, 2021-06-04:
We can think about this.
note:
  *  than `run_sorter()` do the run in the main thread so the use acn already do that before calling it.
   But we don't dontrol how sorters spawn.
  * `run_sorters()` (in launcher.py) have several engine 'loop'/'joblib'/'dask'.
    For 'loop' same answer the `run_sorter()`
    For 'joblib', I don't known how to do it, there is certainly option. engine_kwargs can be used for this I think.
    For 'dask', it is another story.

Feel frree to hack and make PR for your needs.

timsainb, 2021-06-04:
Awesome, I'll play around with that, thanks!

closed 2021-06-04 by timsainb