SpikeInterface/spikeinterface #175: Loading OpenephysData - CAR

gaiabianchini, 2021-06-11:
Hi all!
I am performing Neuropixels recordings with openephys and would like to take advantage of the 'common_reference' function of SpikeInterface.
This is the current code I am using to load the data:

```
import numpy as np
import matplotlib.pyplot as plt
import pandas as pd

import spikeinterface as si
import spikeinterface.extractors as se
import spikeinterface.toolkit as st
import spikeextractors as sp
import pyopenephys as po

dir='Z:\\data\\instruments\\invivoephys\\behav\\non_learner\\2021_05_28\\2021-05-28_10-56-49'
recording = sp.OpenEphysNPIXRecordingExtractor(dir)
recording_car = st.common_reference(recording, reference='local')

```

And this is the output error that I have:

```
Traceback (most recent call last):

  File "<ipython-input-41-43f8a49866a4>", line 1, in <module>
    recording_car = st.common_reference(recording, reference='local')

  File "C:\Users\ephyssetup\.conda\envs\spikeinterface\lib\site-packages\spikeinterface-0.90.0.dev0-py3.7.egg\spikeinterface\toolkit\preprocessing\common_reference.py", line 157, in common_reference
    return CommonReferenceRecording(*args, **kwargs)

  File "C:\Users\ephyssetup\.conda\envs\spikeinterface\lib\site-packages\spikeinterface-0.90.0.dev0-py3.7.egg\spikeinterface\toolkit\preprocessing\common_reference.py", line 83, in __init__
    BasePreprocessor.__init__(self, recording)

  File "C:\Users\ephyssetup\.conda\envs\spikeinterface\lib\site-packages\spikeinterface-0.90.0.dev0-py3.7.egg\spikeinterface\toolkit\preprocessing\basepreprocessor.py", line 10, in __init__
    assert isinstance(recording, BaseRecording), "'recording' must be a RecordingExtractor"

AssertionError: 'recording' must be a RecordingExtractor
```

I am able to perform global demux CAR if I load my data like this: 

```
recording=se.OpenEphysBinaryRecordingExtractor(dir,stream_id='1')
recording_car = st.common_reference(recording, reference='global', operator='average')
```

But unfortunately I am not able to perform local CAR since the channel locations do not seem to be loaded. Any ideas on how I could solve this problem?

Thanks in advance for your help!

alejoe91, 2021-06-11:
Hi @gaiabianchini thanks for the issue. 

In the new API (>0.90, which is the one you are using), there is no `OpenEphysNPIXExtractor` anymore (you should use insted the `OpenEphysBinaryExtractor`).
Moreover, in the new API, the `spikeextractors` package DOES NOT exist anymore (now all the code-base is in the spikeinterface package). So the problem here arises from the fact that you are trying to use the new API (`spikeinterface>0.90`) with the old API (`spikeextractors`).

I recommend using the new API and loading the locations with a `prb` file or with `set_channel_locations`.

@samuelgarcia we should make an open ephys NPIX extractors which automatically loads the channel locations as well.

samuelgarcia, 2021-06-11:
In the new API you can do
`rec = read_openephys('path/tooe/folder')`
this auto select the open ephys format and recording class.

@alejoe91 : we should implement first the probe in probeinterace and then optional load it. There is no need for a special class for this.

XIN-TU, 2021-06-15:
Hi all,
I used the new API mentioned above and open the open ephys file 
the could is :
`recording = se.read_openephys(recording_folder)`

and the error output is :
```
Traceback (most recent call last)
<ipython-input-8-708a33ff4a15> in <module>
      1 recording_folder = 'open-ephys-dataset/'
----> 2 recording = se.read_openephys(recording_folder)

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/extractors/neoextractors/openephys.py in read_openephys(folder_path, **kwargs)
     99     else:
    100         # format = 'binary'
--> 101         recording = OpenEphysBinaryRecordingExtractor(folder_path,  **kwargs)
    102     return recording
    103 

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/extractors/neoextractors/openephys.py in __init__(self, folder_path, stream_id)
     63     def __init__(self, folder_path, stream_id=None):
     64         neo_kwargs = {'dirname' : folder_path}
---> 65         NeoBaseRecordingExtractor.__init__(self, stream_id=stream_id, **neo_kwargs)
     66 
     67         self._kwargs = dict(folder_path=folder_path, stream_id=stream_id)

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/extractors/neoextractors/neobaseextractor.py in __init__(self, stream_id, **neo_kwargs)
     24     def __init__(self, stream_id=None, **neo_kwargs):
     25 
---> 26         _NeoBaseExtractor.__init__(self, **neo_kwargs)
     27 
     28         # check channel

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/extractors/neoextractors/neobaseextractor.py in __init__(self, **neo_kwargs)
     12 
     13     def __init__(self, **neo_kwargs):
---> 14         neoIOclass = eval('neo.rawio.' + self.NeoRawIOClass)
     15         self.neo_reader = neoIOclass(**neo_kwargs)
     16         self.neo_reader.parse_header()

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/extractors/neoextractors/neobaseextractor.py in <module>

AttributeError: module 'neo.rawio' has no attribute 'OpenEphysBinaryRawIO'
```

And I already install the neo package and the version is 0.9.0.

and any ideas on how I could solve this problem?

Thanks very much!

closed 2021-07-22 by samuelgarcia