SpikeInterface/spikeinterface #176: Speed issues

lfrank, 2021-06-11:
Hi all. We now have a working nwb - datajoint - spikeinterface pipeline, but speed remains an issue, and thoughts on whether newer versions of the spiketoolkit and spikeextractors code could speed things up significantly would be welcome. 

For reference, our workflow is as follows:
1. All raw data and experimental data are stored in an NWB file.
2. We define all sort parameters in tables in datajoint.
3. The sorting table references the selected parameters, and uses spikeinterface tools to sort
4. We then curate the data using Jeremy Magland's labbox-ephys jupyter curation interface, and store the curated units in NWB with references to them in a datajoint table for easy searching.

The spikesorting workflow is as follows, and I'm including times for a dataset with 30 tetrodes where we are sorting one 90 minute chunk of data from one tetrode.

1. Create NwbSortingExtractor from NWB file. (time ~12 seconds)
2. Add restrictions to four channels of tetrode, create a cached recording extractor, and set up sub recording extractors for referencing, filtering, etc.: (time ~237 seconds). 
3. Re-cache recording extractor to ensure that filtering, etc. is done and that we have an efficient object to work with (time ~600 seconds)
4. Write filtered recording extractor to disk in Nwb format. This is currently necessary so that we can read it in to Jeremy's labbox ephys suite for curation, but even if we don't use NWB, we need to save a filtered recording extractor of some form for curation / visualization. (time ~1260 seconds)
5. Add whitening and cache again (time not measured)
6. sort whitended data (time ~470 seconds)
7. compute noise overlap, nn_hit_rate and other metrics on whitened data (time 478 seconds)

We tried without each of the intermediate caching steps, and in each case it took significantly longer without the caching, but it does seem a bit silly to have to cache so many times.

Note that we need to save the filtered recording extractor and then sort / compute metrics on the whitened extractor.

samuelgarcia, 2021-06-11:
Hi.
For speed on the old API of spiketoolkit I let Alessio anwser how to speed up things.

I would strongly encourage you to go the new API.
Many steps should faster a lot.

Have a look to this for an overview of change.
https://spikeinterface.readthedocs.io/en/latest/releases/0.90.0.html

You should also have a deep look on tutorial.
https://spikeinterface.readthedocs.io/en/latest/modules/index.html

Note that near almost everything is now implemented in the version 0.90.0.
The only missing piece is writting to nwb.
The main feeling of spikeinterface is pretty much the same as before for the end user.
But many small details can be suprising, explore as much as possible the new doc.

lfrank, 2021-06-11:
Thanks very much for that; we weren’t sure if it was ready. We do still need the ability to write to an external recording extractor, but we can solve that a different way, so we’ll give this a try.

samuelgarcia, 2021-09-17:
@lfrank : have you try the new API and test new speed ?

lfrank, 2021-09-17:
Thanks for checking. Kyu has been working on this, but I think there is some functionality (masking and metrics I believe) that isn’t yet implemented. Kyu - can you comment?

khl02007, 2021-09-21:
@samuelgarcia @lfrank Yes, it seems the new spikeinterface is faster. Our main concern in the previous version was that passing an NWB recording extractor to spike sorting was too slow, and as a result we had to first save recordings to binary and then load it before spike sorting. Now it seems they are roughly on par (sorting a 4 channel, 1.5hr recording with mountainsort from nwb file takes ~ 380s, whereas saving to binary and then sorting takes 150+240 = ~390 s). The waveform extraction on the other hand is still slower for the NWB file (23.5 vs 7.5 s) but quite reasonable.  What we now need is the [ability to save to NWB file](https://github.com/SpikeInterface/spikeinterface/issues/244) and [handle explicit timestamps of recordings](https://github.com/SpikeInterface/spikeinterface/issues/245). There was also a masking function in the previous version to set large transients to zero, but it seems this has been replaced with `remove_artifact` that asks for the timing of stimulations (since this removes the stimulation artifacts). But we could probably just determine the times of artifact with threshold crossing and then pass it in.

alejoe91, 2021-09-21:
Thank you for the patience @khl02007 

- The NWB writer is almost ready! We are going to wrap in SI [this function](https://github.com/catalystneuro/nwb-conversion-tools/blob/ephys_writer/nwb_conversion_tools/utils/ephys_writer.py#L35) from [nwb-conversion-tools](https://github.com/catalystneuro/nwb-conversion-tools). 

- Timestamps and start times are also our priority now and we'll try to have it done by the end of the week.

- The masking function indeed can be replaced with the remove artifact. Note that the masking function was very inefficient, as the mask (same size of the traces) needed to be saved in the object dictionary for parallel processing.. the `remove_artifact` instead only saves a bunch of indexes. Maybe we can think about something in between...

Cheers
Alessio

samuelgarcia, 2021-09-21:
@khl02007 : thanks for feedback on this.
I propose to close this issue that was reading NWB speed in spiekinterface.

wirting to NWB and time vector are sperated ticket.

closed 2021-09-21 by samuelgarcia