SpikeInterface/spikeinterface #177: Issue with Running Sorting

adityas1997, 2021-06-19:
Hi,
I'm trying to run a sorting on test data generated through spikeinterface.

Code:
recording, sorting_true = se.toy_example(duration=10, seed=0, num_segments=1)
print(recording)
print(ss.installed_sorters())
sortingHS = ss.run_sorter(sorter_name='herdingspikes', recording=recording,output_folder='./tmp2/HS_output/')

I get a traceback error of:
NumpyRecording: 4 channels - 1 segments - 30.0kHz - 10.000s
['herdingspikes', 'klusta', 'mountainsort4', 'tridesclous', 'yass']
Traceback (most recent call last):
  File "/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/basesorter.py", line 203, in run_from_folder
    SorterClass._run_from_folder(output_folder, sorter_params, verbose)
  File "/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/herdingspikes/herdingspikes.py", line 152, in _run_from_folder
    recording = load_extractor(output_folder / 'spikeinterface_recording.json')
  File "/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 724, in load_extractor
    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
  File "/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 430, in load
    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
  File "/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 294, in from_dict
    if d['relative_paths']:
KeyError: 'relative_paths'

Help would be much appreciated.

adityas1997, 2021-06-19:
This is running the latest spikeinterface and attempting to the follow the tutorials there.

samuelgarcia, 2021-06-21:
Hi @adityas1997 
What is happening certaoinly is that you run some sortign with the new work-in-progress spikeinterface one month ago.
And you update with a more recent work-in-progress spikeinterface and the dump mechanisim is now  abit more enhance and make reading old folder imcompatible.
Sorry for that.
You need to delete folder and re run everything.

adityas1997, 2021-06-21:
Hi, thanks for the response.
Which folder are you referring to that needs to be deleted, and do you have documentation for the latest version on this git that would allow for this method to be run?
Thanks.

samuelgarcia, 2021-06-21:
This one `'./tmp2/HS_output/'` no ?

adityas1997, 2021-06-21:
Deleted the folder you mentioned, and reran the code. I receive the same traceback error.
Traceback (most recent call last):
  File "/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/basesorter.py", line 203, in run_from_folder
    SorterClass._run_from_folder(output_folder, sorter_params, verbose)
  File "/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/herdingspikes/herdingspikes.py", line 152, in _run_from_folder
    recording = load_extractor(output_folder / 'spikeinterface_recording.json')
  File "/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 724, in load_extractor
    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
  File "/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 430, in load
    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
  File "/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/core/base.py", line 294, in from_dict
    if d['relative_paths']:
KeyError: 'relative_paths'

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
<ipython-input-1-20583f882dae> in <module>
     13 print(recording)
     14 print(ss.installed_sorters())
---> 15 sortingHS = ss.run_sorter(sorter_name='herdingspikes', recording=recording,output_folder='/home/aditya/tmp2/HS_output/')

/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, **sorter_params)
     53         sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     54             remove_existing_folder=remove_existing_folder, delete_output_folder=delete_output_folder,
---> 55             verbose=verbose, raise_error=raise_error, **sorter_params)
     56     else:
     57         sorting = run_sorter_docker(sorter_name, recording, docker_image, output_folder=output_folder,

/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, **sorter_params)
     74     SorterClass.set_params_to_folder(recording, output_folder, sorter_params, verbose)
     75     SorterClass.setup_recording(recording, output_folder, verbose=verbose)
---> 76     SorterClass.run_from_folder(output_folder, raise_error, verbose)
     77     sorting = SorterClass.get_result_from_folder(output_folder)
     78 

/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface-0.90.0.dev0-py3.7.egg/spikeinterface/sorters/basesorter.py in run_from_folder(cls, output_folder, raise_error, verbose)
    237         if has_error and raise_error:
    238             print(log['error_trace'])
--> 239             raise SpikeSortingError("Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json")
    240 
    241         return run_time

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json

Is there a package missing like datalad?

samuelgarcia, 2021-06-21:
OK sorry. I will fix it soon.

If you want you can do `recording_cached = recording.save(folder='/tmp/reco/path')`
 and run the sorter with `recording_cached` it make the recording dumpable and should avoid you issue.

samuelgarcia, 2021-07-22:
Fixed.

closed 2021-07-22 by samuelgarcia