SpikeInterface/spikeinterface #178: Issue with read file

XIN-TU, 2021-06-22:
Hi all,
I used the new API mentioned above and open the open ephys file
the could is :
`recording = se.read_openephys(recording_folder)`

and the error output is :
```
Traceback (most recent call last)
<ipython-input-8-708a33ff4a15> in <module>
      1 recording_folder = 'open-ephys-dataset/'
----> 2 recording = se.read_openephys(recording_folder)

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/extractors/neoextractors/openephys.py in read_openephys(folder_path, **kwargs)
     99     else:
    100         # format = 'binary'
--> 101         recording = OpenEphysBinaryRecordingExtractor(folder_path,  **kwargs)
    102     return recording
    103 

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/extractors/neoextractors/openephys.py in __init__(self, folder_path, stream_id)
     63     def __init__(self, folder_path, stream_id=None):
     64         neo_kwargs = {'dirname' : folder_path}
---> 65         NeoBaseRecordingExtractor.__init__(self, stream_id=stream_id, **neo_kwargs)
     66 
     67         self._kwargs = dict(folder_path=folder_path, stream_id=stream_id)

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/extractors/neoextractors/neobaseextractor.py in __init__(self, stream_id, **neo_kwargs)
     24     def __init__(self, stream_id=None, **neo_kwargs):
     25 
---> 26         _NeoBaseExtractor.__init__(self, **neo_kwargs)
     27 
     28         # check channel

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/extractors/neoextractors/neobaseextractor.py in __init__(self, **neo_kwargs)
     12 
     13     def __init__(self, **neo_kwargs):
---> 14         neoIOclass = eval('neo.rawio.' + self.NeoRawIOClass)
     15         self.neo_reader = neoIOclass(**neo_kwargs)
     16         self.neo_reader.parse_header()

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/extractors/neoextractors/neobaseextractor.py in <module>

AttributeError: module 'neo.rawio' has no attribute 'OpenEphysBinaryRawIO'
```

And I already install the neo package and the version is 0.9.0.

and any ideas on how I could solve this problem?

Thanks very much!

alejoe91, 2021-06-22:
Hi @XIN-TU 

Can you install also Neo from master? The new version (0.10.0) will be released soon, but it's not up yet :)

That should fix it!

XIN-TU, 2021-06-22:
hi @alejoe91
I followed the instruction to install Neo which git from the master, but the version still shows 0.9.0, and did you know why?

```
Using /Users/xin/opt/anaconda3/envs/spike2/lib/python3.8/site-packages
Finished processing dependencies for neo==0.9.0
```

alejoe91, 2021-06-22:
I think you have to pip uninstall it first

XIN-TU, 2021-06-22:
@alejoe91 
I already uninstall it and install it again, but it still shows the 0.9.0 version. :(

XIN-TU, 2021-06-22:
@alejoe91 
And I git from https://github.com/NeuralEnsemble/python-neo, and the license shows the version is 0.9.0, and I git from the wrong repo?

alejoe91, 2021-06-22:
@XIN-TU 

Still with the master version you should have the `OpenEphysBinaryRawIO` (https://github.com/NeuralEnsemble/python-neo/blob/master/neo/rawio/openephysbinaryrawio.py)

@samuelgarcia any clue?

alejoe91, 2021-06-22:
> @alejoe91
> And I git from https://github.com/NeuralEnsemble/python-neo, and the license shows the version is 0.9.0, and I git from the wrong repo?

No that's actually correct (https://github.com/NeuralEnsemble/python-neo/blob/master/neo/version.py), I was wrong :)

samuelgarcia, 2021-06-22:
Hi @XIN-TU.
neo/spikeinetrface/probeinterface have to installed from source see this:
https://spikeinterface.readthedocs.io/en/latest/installation.html#installation

If you did not do it please uninstall and re install.
And there is a mistake in neo from source. The version is 0.9.0 but it should be 0.10.0.dev0.

XIN-TU, 2021-06-22:
@alejoe91  @samuelgarcia I attempt to uninstall and reinstall Neo to solve this problem. Thanks very much!

and I got a new problem when I run the spike sorter. I try to run `sorting_TDC = ss.run_tridesclous(recording=recording_preprocessed)` and I also install the tridesclous-1.6.3 via pip, but the program shows I did not install this sorter, and the error show :

```
Exception                                 Traceback (most recent call last)
<ipython-input-18-b0ba14f87364> in <module>
----> 1 sorting_TDC = ss.run_tridesclous(recording=recording_preprocessed)

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/sorters/runsorter.py in run_tridesclous(*args, **kwargs)
    245 
    246 def run_tridesclous(*args, **kwargs):
--> 247     return run_sorter('tridesclous', *args, **kwargs)
    248 run_tridesclous.__doc__ = _common_run_doc.format('tridesclous')
    249 

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, **sorter_params)
     51 
     52     if docker_image is None:
---> 53         sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     54             remove_existing_folder=remove_existing_folder, delete_output_folder=delete_output_folder,
     55             verbose=verbose, raise_error=raise_error, **sorter_params)

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/sorters/runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, **sorter_params)
     71 
     72     # only classmethod call not instance (stateless at instance level but state is in folder)
---> 73     output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
     74     SorterClass.set_params_to_folder(recording, output_folder, sorter_params, verbose)
     75     SorterClass.setup_recording(recording, output_folder, verbose=verbose)

~/opt/anaconda3/envs/spike2/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/sorters/basesorter.py in initialize_folder(cls, recording, output_folder, verbose, remove_existing_folder)
     80         # installed ?
     81         if not cls.is_installed():
---> 82             raise Exception(f"The sorter {cls.sorter_name} is not installed."
     83                         f"Please install it with:  \n{cls.installation_mesg} ")
     84 

Exception: The sorter tridesclous is not installed.Please install it with:  

To use Tridesclous run:

       >>> pip install tridesclous

    More information on tridesclous at:
      * https://github.com/tridesclous/tridesclous
      * https://tridesclous.readthedocs.io
```

did you know why? thanks very much!

samuelgarcia, 2021-06-22:
You need to install `hdbscan` package. (pip or conda)

XIN-TU, 2021-06-22:
> You need to install `hdbscan` package. (pip or conda)

I already install `hdbscan` which version is 0.8.27, it still can not run tridesclous

samuelgarcia, 2021-06-22:
Are you able to do `tdc` in a console and see an opening window ?

XIN-TU, 2021-06-22:
> Are you able to do `tdc` in a console and see an opening window ?

Sorry, I don't know what "able to do tdc in a console" means? And I use jupyter notebook to run code and the error is do not have tridesclous.

samuelgarcia, 2021-06-22:
`tdc` is a command to open tridesclous GUI when it is installed.
It is a good test to check if there no missing dependency.

The other way is do `import tridesclous` in jupyter or a script check get the error message.
There must be something wrong with the installation.
And so spikeinterface think it is not installed.

Maybe this is somethign else.
I will check this.

closed 2021-07-22 by samuelgarcia