SpikeInterface/spikeinterface #181: ValueError: There is not Probe attached to recording. use set_probe(...)

ChrisCurrin, 2021-06-29:
I'm transitioning to v0.90 and ran into the following issue for 

```python
recording = se.MaxwellRecordingExtractor("data.raw.h5", stream_id="well000", rec_name='rec0000')
ss.run_spykingcircus(recording=recording)
```

```python
ValueError                                Traceback (most recent call last)
<ipython-input-12-b5ac23c3d039> in <module>
      1 sc_params = ss.get_default_params('spykingcircus')
      2 
----> 3 sorting_SC = ss.run_spykingcircus(recording=recording, **sc_params)

~/miniconda3/envs/mea-spike-pipe/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py in run_spykingcircus(*args, **kwargs)
    280 
    281 def run_spykingcircus(*args, **kwargs):
--> 282     return run_sorter('spykingcircus', *args, **kwargs)
    283 run_spykingcircus.__doc__ = _common_run_doc.format('spykingcircus')
    284 

~/miniconda3/envs/mea-spike-pipe/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, **sorter_params)
     51 
     52     if docker_image is None:
---> 53         sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     54             remove_existing_folder=remove_existing_folder, delete_output_folder=delete_output_folder,
     55             verbose=verbose, raise_error=raise_error, **sorter_params)

~/miniconda3/envs/mea-spike-pipe/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, **sorter_params)
     73     output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
     74     SorterClass.set_params_to_folder(recording, output_folder, sorter_params, verbose)
---> 75     SorterClass.setup_recording(recording, output_folder, verbose=verbose)
     76     SorterClass.run_from_folder(output_folder, raise_error, verbose)
     77     sorting = SorterClass.get_result_from_folder(output_folder)

~/miniconda3/envs/mea-spike-pipe/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py in setup_recording(cls, recording, output_folder, verbose)
    172             all_params =  json.load(f)
    173             sorter_params = all_params['sorter_params']
--> 174         cls._setup_recording(recording, output_folder, sorter_params, verbose)
    175 
    176     @classmethod

~/miniconda3/envs/mea-spike-pipe/lib/python3.8/site-packages/spikeinterface/sorters/spyking_circus/spyking_circus.py in _setup_recording(cls, recording, output_folder, params, verbose)
    109         # note: only one group here, the split is done in basesorter
    110         prb_file = output_folder / 'probe.prb'
--> 111         probegroup = recording.get_probegroup()
    112         write_prb(prb_file, probegroup,
    113                         total_nb_channels=recording.get_num_channels(),

~/miniconda3/envs/mea-spike-pipe/lib/python3.8/site-packages/spikeinterface/core/baserecording.py in get_probegroup(self)
    309             positions = self.get_property('location')
    310             if positions is None:
--> 311                 raise ValueError('There is not Probe attached to recording. use set_probe(...)')
    312             else:
    313                 warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')

ValueError: There is not Probe attached to recording. use set_probe(...)
```

Previously, the probe information was automatically extracted from the recording.

`> https://github.com/SpikeInterface/spikesorters/blob/master/spikesorters/spyking_circus/spyking_circus.py`
```python
recording.save_to_probe_file(probe_file, grouping_property=None,
                                     radius=p['adjacency_radius'])
```

samuelgarcia, 2021-06-29:
Hi CHristopher,
this is right.
I will fix this and add automatically the probe to the recording.
Thanks for the feedback.

ChrisCurrin, 2021-06-29:
Thanks @samuelgarcia 

I've done this in the mean time:
```python
recording = se.MaxwellRecordingExtractor("data.raw.h5", stream_id="well000", rec_name='rec0000')

probe = probeinterface.io.read_maxwell(
    data_path, well_name=well, rec_name="rec0000"
)
recording = recording.set_probe(probe)

ss.run_spykingcircus(recording=recording)
```

samuelgarcia, 2021-06-29:
This is the plan but it is better io have autotic.

samuelgarcia, 2021-06-29:
#183 done here
@ChrisCurrin can you check this with latest probeinterface also ?

ChrisCurrin, 2021-06-30:
looks good and tested successfully

closed 2021-06-30 by alejoe91