SpikeInterface/spikeinterface #182: Issue with duration of open-ehyps file

tpfschaffhauser, 2021-06-29:
Hi, 

I have some problems when reading my open-ephys dataset using spike-interface (v0.12.0): 

`rec = se.OpenEphysRecordingExtractor(fileName,recording_id=0)`
`fs = rec.get_sampling_frequency() `
`dur = np.floor((rec.get_num_frames()/fs)/60)`

This dataset contains 5 separate recordings (recording1/ ... recording5/) folders but the duration I obtain with spike-interface (dur) does not correspond to the duration of the different recordings, it looks like the recording ids are shuffled or do not correspond to the folder order I have on my drive.

alejoe91, 2021-07-01:
Hi @tpfschaffhauser 

What version of spikeinterface are you using? Would you mind sharing a sample dataset with me so that I can take a look?

tpfschaffhauser, 2021-07-14:
Hi @alejoe91, 

sorry for the late reply. I'm using spikeinterface 0.12.0.
I'll be very happy to share this dataset but the compressed dataset is 23GB.

alejoe91, 2021-07-14:
@tpfschaffhauser no worries. Maybe you could share it via google drive? Or maybe your institution has a file transfer service. we can arrange via email :)

alejoe91, 2021-07-15:
@tpfschaffhauser can you try with this patch in pyopenephys? https://github.com/CINPLA/pyopenephys/pull/30

Note that for the master version of SI (>0.90), the open ephys wrapper uses NEO directly, so the problem should be solved already.

closed 2021-07-15 by alejoe91