SpikeInterface/spikeinterface #184: maping spikes to channels

farazmoradi, 2021-07-01:
Hi, is there any way that we can figure out each spike in each unit(neuron) comes from which channel?
Thanks.

samuelgarcia, 2021-07-01:
yes.
in
toolkit.postprocessing.template_tools you have:

`get_template_extremum_channel()`

`get_template_channel_sparsity()`

alejoe91, 2021-07-01:
@samuelgarcia I think @farazmoradi means for each waveform, not for the template. Right @farazmoradi?

farazmoradi, 2021-07-01:
Actually, for instance, I have some spike trains for a single neuron. I want to know each spike train comes from which channel.
Also, If there are some bursts in my spike train, what should I do about them? Are they going to affect my analysis? I am doing classical feature extraction methods like Victor Purpura and ISI.

farazmoradi, 2021-07-01:
After Spike sorting, I compute spike waveforms and I have reached this picture. is the plot reasonable?
[image]

alejoe91, 2021-07-01:
@farazmoradi what are you plotting exactly? What are the different colors? 

Something is definitely off.. can you paste your entire code here?

farazmoradi, 2021-07-01:
For sure, at first, I should mention that I have edf file from a mouse running on a linear track. the data has 17 electrodes which 2 of them is for EMG and one is for wireless signal. So I have created a 2D matrix called **chan_corr** which is 14 channels by time point. Also, I have fixed the locations of channels by putting them in order previously. 

**the first part is like this: **

num_channels = 14
sampling_frequency = 10000  # in Hz
duration           = int(np.floor(chan_corr.shape[1] / (10000 * 60)) * 60)
chan_corr    = chan_corr[:,0:duration*sampling_frequency]
num_timepoints     = int(sampling_frequency * duration)
geom       = np.zeros((num_channels, 2))
geom[:, 0] = range(num_channels)

**Second part I created Numpy Recording: **

recording = se.NumpyRecordingExtractor(timeseries=chan_corr, geom=geom, sampling_frequency=sampling_frequency)
recording_f = st.preprocessing.bandpass_filter(recording, freq_min=300, freq_max=4900)

**spike sorting by Mountain Sorter**

default_ms4_params = ss.Mountainsort4Sorter.default_params()
default_ms4_params['detect_threshold'] = 4
default_ms4_params['curation'] = False
# parameters set by params dictionary
sorting_MS4 = ss.run_mountainsort4(recording=recording_f, **default_ms4_params,
                                   output_folder='tmp_MS4')

** Waveforms **
wf = st.postprocessing.get_unit_waveforms(recording_f, sorting_MS4, ms_before=1, ms_after=2,
                                          save_as_features=True, verbose=True)
fig, ax = plt.subplots()
ax.plot(wf[0][:, 0, :].T, color='k', lw=0.3)
ax.plot(wf[1][:, 0, :].T, color='r', lw=0.3)
ax.plot(wf[2][:, 0, :].T, color='b', lw=0.3)

alejoe91, 2021-07-01:
How do the signals look? Do you see spikes in the highpass filtered data?

Note that you are also plotting the waveforms on a single channel, which might not be the best channel. Can I have more info on the probe? Are all channels close to each other or quite separated?

farazmoradi, 2021-07-01:
This is the only information that I have from the prob.
The hipass filtered data seems to have spikes. Also, I have plotted the raster plot. there are many spikes in it.
The red lines indicates the time when the mouse was running on the track.
[image]

[image]

alejoe91, 2021-07-01:
Ok then probably you are not plotting the waveforms on the 'right' channel. Can you try to plot the template instead?

farazmoradi, 2021-07-01:
[image]

alejoe91, 2021-07-01:
that looks better :) you see that the channels with a large signals are 1/2 per unit, so you need to identify those

farazmoradi, 2021-07-01:
I get that thanks a lot for your help. So I have done the spike sorting somehow correct am I right? :)

farazmoradi, 2021-07-01:
I had another question regarding burst, I will appreciate it if you answer this too. 
Also, If there are some bursts in my spike train, what should I do about them? Are they going to affect my analysis? I am doing classical feature extraction methods like Victor Purpura and ISI.

alejoe91, 2021-07-01:
Yes Spike sorting looks reasonable. i don't have experience in analysing burst specific features unfortunately...

farazmoradi, 2021-07-01:
Thanks a lot.

farazmoradi, 2021-07-03:
Hi  again,
Is there anyway that we can see detected spikes on each channel before spike sorting?
Thanks

alejoe91, 2021-07-03:
Hi,

You can detect spikes with the `st.sortingcomponents.detect_spikes()` function. However this is a separate spike detection and it might be different than the one used by the single sorters. Since the spike sorters are run separately, there is no other way to access the spikes detected by the sorters.

farazmoradi, 2021-07-05:
Hi, I get this error while using from any function in the spike interface. is it a serious thing?

/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/spikeextractors/extraction_tools.py:838: DeprecationWarning: `np.float` is a deprecated alias for the builtin `float`. To silence this warning, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  if isinstance(start_frame, (float, np.float)):

alejoe91, 2021-07-05:
Don't worry, it's just a warning. We'll.make an update to remove it. Closing this issue for now. Open a new one if you find other problems!

closed 2021-07-05 by alejoe91

samuelgarcia, 2021-07-05:
It come from numpy, you can:

  * downgrade numpy
  * update neo with the current master, it have been fixed.