SpikeInterface/spikeinterface #186: Spike Detection on spike sorters

farazmoradi, 2021-07-05:
Hi again, 

I was checking my spikes sorted by mountain sorter, but I was a little hesitant about the spikes that the toolbox has detected, so I decided to use this code as you mentioned earlier:

**sorted = st.sortingcomponents.detect_spikes(recording_f,detect_threshold=4)**

after plotting the spikes on the LFP signal to see how true are them I get a little confused:
take the pink line as the threshold, how is it possible that spikes below the threshold are not detected?
is the same phenomenon happing in the sorters like mountain sorter too?
[image]

alejoe91, 2021-07-05:
Are you plotting the filtered data? The large deflections seem to be low frequency, so they are removed with the filtering steps. Try to plot the filtered data.

mhhennig, 2021-07-05:
This looks like the detection is done on transposed data! Is this the old or new API?

samuelgarcia, 2021-07-05:
This looks strange.

Please note that by default the output is "numpy_compact".
In short you get a numpy vector with several fields
`peak_dtype = [('sample_ind', 'int64'), ('channel_ind', 'int64'), ('amplitude', 'float64'), ('segment_ind', 'int64')]`

Are you sure you are plotting on the same recording used for detection ?

Are you sure you get plot peaks according the channel they are detected ?

Can you send the codes ?

Note that there is 2 methods implemented 'by_channel' and 'locally_exclusive'

```python
peaks = detect_peaks(recording, method='by_channel', 
        peak_sign='neg', detect_threshold=5, n_shifts=2, 
        local_radius_um=100,
        noise_levels=None,
        random_chunk_kwargs={},
        outputs='numpy_compact',
        **job_kwargs)
```

For plotting peak on trace channel please have a look to "test_peak_detection.py" in "spikeinterface/sortingcomponents/tests" folder.

farazmoradi, 2021-07-06:
Thanks, everyone, this was the problem as @alejoe91  mentioned. I was plotting on the raw data instead of filtered Data.

closed 2021-07-13 by alejoe91