SpikeInterface/spikeinterface #187: Problems Running Spike Sorters

farazmoradi, 2021-07-08:
Hi, I had some errors while running spike sorting on the new spike interface:

Traceback (most recent call last):
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/IPython/core/interactiveshell.py", line 3437, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-9-6b2963e2f014>", line 5, in <module>
    sorting_MS4 = ss.run_mountainsort4(recording=recording_f)
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/spikeinterface-0.90.0.dev0-py3.9.egg/spikeinterface/sorters/runsorter.py", line 252, in run_mountainsort4
    return run_sorter('mountainsort4', *args, **kwargs)
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/spikeinterface-0.90.0.dev0-py3.9.egg/spikeinterface/sorters/runsorter.py", line 53, in run_sorter
    sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/spikeinterface-0.90.0.dev0-py3.9.egg/spikeinterface/sorters/runsorter.py", line 73, in run_sorter_local
    output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/spikeinterface-0.90.0.dev0-py3.9.egg/spikeinterface/sorters/basesorter.py", line 120, in initialize_folder
    recording.dump_to_json(rec_file)
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/spikeinterface-0.90.0.dev0-py3.9.egg/spikeinterface/core/base.py", line 377, in dump_to_json
    assert self.check_if_dumpable()
AssertionError

samuelgarcia, 2021-07-08:
Some recording are not dumpable to json because they describe for instance a array in memory.
In that case you can do `rec_on_disk = rec.save(folder='path_save')` and make the sorting with `rec_on_disk` which is dumpable.

farazmoradi, 2021-07-09:
Could you please specify that is **rec** exactly?

samuelgarcia, 2021-07-09:
Oups. Sorry rec is a `RecordingExtactor` object.
Have a look to [Core tutorial](https://spikeinterface.readthedocs.io/en/latest/modules/index.html#core-tutorials) for detail on objects.

farazmoradi, 2021-07-09:
I fixed that but still getting this error.

This is my code:
"" recording_f is my data " "

rec_on_disk = recording_f.save(folder='/Volumes/Project/test')
default_ms4_params = ss.Mountainsort4Sorter.default_params()
default_ms4_params['detect_threshold'] = 4
default_ms4_params['curation'] = False
# parameters set by params dictionary
sorting_MS4 = ss.run_mountainsort4(recording=rec_on_disk, **default_ms4_params,
                                   output_folder='tmp_MS4')

Error:

Traceback (most recent call last):
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/IPython/core/interactiveshell.py", line 3437, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-9-1ae3e740085b>", line 6, in <module>
    sorting_MS4 = ss.run_mountainsort4(recording=rec_on_disk, **default_ms4_params,
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/spikeinterface-0.90.0.dev0-py3.9.egg/spikeinterface/sorters/runsorter.py", line 252, in run_mountainsort4
    return run_sorter('mountainsort4', *args, **kwargs)
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/spikeinterface-0.90.0.dev0-py3.9.egg/spikeinterface/sorters/runsorter.py", line 53, in run_sorter
    sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/spikeinterface-0.90.0.dev0-py3.9.egg/spikeinterface/sorters/runsorter.py", line 76, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "/Users/farazmoradi/.conda/envs/Mouse/lib/python3.9/site-packages/spikeinterface-0.90.0.dev0-py3.9.egg/spikeinterface/sorters/basesorter.py", line 239, in run_from_folder
    raise SpikeSortingError("Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json")
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json

samuelgarcia, 2021-07-09:
The problem now is inside MS4.
You should open the log file and check what happen.
Maybe ask on issue in MS4 github page.

farazmoradi, 2021-07-09:
Thanks a lot.

farazmoradi, 2021-07-09:
But the problem did not happen with the old spike interface. I think maybe I should downgrade the version. Can I do that?

samuelgarcia, 2021-07-09:
You can try in a diffrents env to check.

Long time ago MS4 use to be built on top of msalg_ms4 paquage and now it is moutainsort4 paquage.
This is indepedant from SI. Now SI use only moutainsort4 internally.

closed 2021-07-22 by samuelgarcia