SpikeInterface/spikeinterface #191: post_processing_tools export_to_phy ms_before and ms_after params

klaragerlei, 2021-07-13:
Hi, 

I'm using spikeinterface 0.12.0. I can see that the code on the repo is different, so apologies if this isn't the right place for this issue.

I was trying to set a non-default value for ms_before and ms_after when using the phy extractor:

`    st.postprocessing.export_to_phy(recording, sorting, output_folder=output_folder,
                                    copy_binary=True, ms_before=0.5, ms_after=0.5)`

but I got this error:

>   File "/home/ubuntu/miniconda3/envs/env/lib/python3.6/site-packages/spiketoolkit/postprocessing/postprocessing_tools.py", line 1712, in _get_quality_metric_data
>     joblib_backend=joblib_backend, return_idxs=True)
>   File "/home/ubuntu/miniconda3/envs/env/lib/python3.6/site-packages/spiketoolkit/postprocessing/postprocessing_tools.py", line 738, in get_unit_amplitudes
>     amps = np.max(np.abs(wf_cut), axis=-1)
>   File "<__array_function__ internals>", line 6, in amax
>   File "/home/ubuntu/miniconda3/envs/env/lib/python3.6/site-packages/numpy/core/fromnumeric.py", line 2706, in amax
>     keepdims=keepdims, initial=initial, where=where)
>   File "/home/ubuntu/miniconda3/envs/env/lib/python3.6/site-packages/numpy/core/fromnumeric.py", line 87, in _wrapreduction
>     return ufunc.reduce(obj, axis, dtype, out, **passkwargs)
> ValueError: zero-size array to reduction operation maximum which has no identity

I think this was because the `get_unit_waveforms` function gets the ms_before and ms_after parameters and extracts the waveforms accordingly, but `get_unit_amplitudes` doesn't, so the ms_before and ms_after parameters go back to the default value of 3 in params_dict.
This causes a problem when slicing the array here and w_cut will be empty:
`wf_cut = wf[:, max_channels[i], center_frame - frames_before:center_frame + frames_after]`

I edited the code to add the missing parameters to get_unit_amplitudes and it seems to work well now.

alejoe91, 2021-07-13:
Thanks @klaragerlei 

Could you make a PR to spiketoolkit with your fix?

Right we are in the process of updating the api, but we would still like to keep fixing bugs on the 0.12 version series. Let me know if you need help with the PR! :)

klaragerlei, 2021-07-13:
I sent a PR https://github.com/SpikeInterface/spiketoolkit/pull/491

alejoe91, 2021-07-13:
> I sent a PR [SpikeInterface/spiketoolkit#491](https://github.com/SpikeInterface/spiketoolkit/pull/491)

Merged :) closing the issue!

closed 2021-07-13 by alejoe91