SpikeInterface/spikeinterface #194: Trouble loading Intan Data

congcong96, 2021-07-16:
Hi,

I've been using Intan system for my recordings and have trouble importing data to SpikeInterface. 

When I use the IntanExtractor, it expects a data format with .rhd extension. However, my data from the Intan acquisition system is saved in a folder, including time series with .dat extension and only information about the recording, e.g. acquisition frequency, is saved in a info.rhd file. The data is saved as one file per channel, which is supported by the Intan software.

How should I import Intan files saved in this format to SpikeInterface?

alejoe91, 2021-07-19:
Hi @congcong96 

Is the `.dat` file saved directly from Intan? I though it only saved `.rhd` or `.rhs` files. Which acquisition system are you using? 
And which version of the Intan software?
Also, what version of SpikeInterface are you using?

In your case, you could parse the information needed for a `BinaryRecordingExtractor` (sampling frequency, number of channels, type of the data) form the `info.rhd` file and import the `dat` file directly using the `BinaryRecordingExtractor`.

samuelgarcia, 2021-07-19:
Hi,
I have heard that intan recently included also a binary format also.
Maybe there is a small header and we could implement to read it. I am not sure that the header is in rhd.

alejoe91, 2021-07-19:
@congcong96 would you be able to gathe a short 1-second data file and share it with us so we can test it?

congcong96, 2021-07-19:
The .dat file is saved directly from Intan. I used both Intan RHD2000 and RHX software (RHX is the newer version). The SpikeInterface version is 0.10.0

The time series from each channel are saved in .dat files, one file per channel (int16, to convert to electrode voltage in microvolts, first subtract 32768 then multiply by 0.195). 
Information about the recording is saved in the info.rhd file. It can be read by pyintan, just without the time series as they are saved in .dat files.

The test data is a short blank recording from a 64-channel probe. Thanks!
[SpikeInterface_Intan_test_210719_110554.zip](https://github.com/SpikeInterface/spikeinterface/files/6843204/SpikeInterface_Intan_test_210719_110554.zip)

h-mayorquin, 2023-03-06:
Did you ever solve your problem @congcong96 ?

zm711, 2024-04-01:
We have fixed this at the neo level. So this can be done with an install of neo from source.

closed 2024-04-01 by zm711