SpikeInterface/spikeinterface #200: Allow retrieval of waveforms from WaveformExtractor by property

alejoe91, 2021-07-21:
The WaveformExtractor `get_waveforms` and `get_templates` function could handle grouping. In turn, `export_to_phy` could use this info to sparsify the templates

farazmoradi, 2021-07-21:
I was using si.extract_waveforms and I received this error! 

Traceback (most recent call last):
  File "/Users/faraz/.conda/envs/SpikeSorting/lib/python3.9/site-packages/IPython/core/interactiveshell.py", line 3441, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-9-b1c7f5a5284f>", line 2, in <module>
    we = si.extract_waveforms(recording_cmr, sorting, folder)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/waveform_extractor.py", line 461, in extract_waveforms
    we = WaveformExtractor.create(recording, sorting, folder)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/waveform_extractor.py", line 105, in create
    recording.dump(folder / 'recording.json', relative_to=None)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/base.py", line 359, in dump
    self.dump_to_json(file_path, relative_to)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/base.py", line 377, in dump_to_json
    assert self.check_if_dumpable()
AssertionError

alejoe91, 2021-07-21:
Can you open a separate issue for your problem @farazmoradi ?

farazmoradi, 2021-07-21:
Sure