SpikeInterface/spikeinterface #201: Problems running extract_waveforms

farazmoradi, 2021-07-21:
Hi,

Traceback (most recent call last):
  File "/Users/faraz/.conda/envs/SpikeSorting/lib/python3.9/site-packages/IPython/core/interactiveshell.py", line 3441, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-9-b1c7f5a5284f>", line 2, in <module>
    we = si.extract_waveforms(recording_cmr, sorting, folder)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/waveform_extractor.py", line 461, in extract_waveforms
    we = WaveformExtractor.create(recording, sorting, folder)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/waveform_extractor.py", line 105, in create
    recording.dump(folder / 'recording.json', relative_to=None)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/base.py", line 359, in dump
    self.dump_to_json(file_path, relative_to)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/base.py", line 377, in dump_to_json
    assert self.check_if_dumpable()
AssertionError

alejoe91, 2021-07-21:
Hi @farazmoradi 

what type is your recording object? Is it mearec? Can you paste here the output of:
`recording_cmr.to_dict()`?

farazmoradi, 2021-07-21:
I solved it by : 

rec_on_disk = recording_cmr.save(folder='path_save')
folder = 'waveforms_mearecc'
we = si.extract_waveforms(rec_on_disk, sorting, folder)

farazmoradi, 2021-07-21:
> Hi @farazmoradi
> 
> what type is your recording object? Is it mearec? Can you paste here the output of:
> `recording_cmr.to_dict()`?

{'class': 'spikeinterface.toolkit.preprocessing.common_reference.CommonReferenceRecording',
 'module': 'spikeinterface',
 'kwargs': {'recording': {'class': 'spikeinterface.toolkit.preprocessing.filter.BandpassFilterRecording',
   'module': 'spikeinterface',
   'kwargs': {'recording': {'class': 'spikeinterface.core.numpyextractors.NumpyRecording',
     'module': 'spikeinterface',
     'kwargs': {'traces_list': [array([[-7.44247407e-03,  4.19120887e-03, -5.58814962e-03, ...,
               -1.07837676e-02, -8.51402259e-03, -1.09031676e-02],
              [-7.44361848e-03,  4.18968299e-03, -5.59272726e-03, ...,
               -1.07597350e-02, -8.51554847e-03, -1.09073638e-02],
              [-7.43675202e-03,  4.18777564e-03, -5.59616048e-03, ...,
               -1.07677459e-02, -8.50677467e-03, -1.09008788e-02],
              ...,
              [ 7.34874495e-10,  7.34874495e-10,  7.34874495e-10, ...,
                7.34874495e-10,  7.34874495e-10,  7.34874495e-10],
              [ 7.34874495e-10,  7.34874495e-10,  7.34874495e-10, ...,
                7.34874495e-10,  7.34874495e-10,  7.34874495e-10],
              [ 7.34874495e-10,  7.34874495e-10,  7.34874495e-10, ...,
                7.34874495e-10,  7.34874495e-10,  7.34874495e-10]])],
      'sampling_frequency': 10000},
     'dumpable': False,
     'version': '0.90.0.dev0',
     'relative_paths': False,
     'annotations': {'is_filtered': False},
     'properties': {'group': array([0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]),
      'location': array([[  0.,   0.],
             [  0.,  20.],
             [  0.,  40.],
             [  0.,  60.],
             [  0.,  80.],
             [  0., 100.],
             [  0., 120.],
             [  0., 140.],
             [  0., 160.],
             [  0., 180.],
             [  0., 200.],
             [  0., 220.],
             [  0., 240.],
             [  0., 260.]])}},
    'freq_min': 300,
    'freq_max': 4900,
    'margin_ms': 5.0},
   'dumpable': True,
   'version': '0.90.0.dev0',
   'relative_paths': False,
   'annotations': {'is_filtered': True},
   'properties': {'group': array([0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]),
    'location': array([[  0.,   0.],
           [  0.,  20.],
           [  0.,  40.],
           [  0.,  60.],
           [  0.,  80.],
           [  0., 100.],
           [  0., 120.],
           [  0., 140.],
           [  0., 160.],
           [  0., 180.],
           [  0., 200.],
           [  0., 220.],
           [  0., 240.],
           [  0., 260.]])}},
  'reference': 'global',
  'groups': None,
  'ref_channels': None,
  'local_radius': (30, 55)},
 'dumpable': True,
 'version': '0.90.0.dev0',
 'relative_paths': False,
 'annotations': {'is_filtered': True},
 'properties': {'group': array([0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]),
  'location': array([[  0.,   0.],
         [  0.,  20.],
         [  0.,  40.],
         [  0.,  60.],
         [  0.,  80.],
         [  0., 100.],
         [  0., 120.],
         [  0., 140.],
         [  0., 160.],
         [  0., 180.],
         [  0., 200.],
         [  0., 220.],
         [  0., 240.],
         [  0., 260.]])}}

alejoe91, 2021-07-21:
yes the problem was that the `NumpyRecording`, which is the base of your processing, is in memory, and therefore not *dumpable*. 

Saving it to file solves the issue!

closed 2021-07-21 by alejoe91

farazmoradi, 2021-07-21:
But @alejoe91 something strange happened today. I was using the Tridesclous on my dataset and after sorting it founds 15 units. But while using the spike interface with same parameters and filters did not found anything. what is your guess?

alejoe91, 2021-07-21:
mmm not sure! Try to relaunch restart the notebook and relaunch it! BTW tridesclous was written by @samuelgarcia