SpikeInterface/spikeinterface #203: Error File Using Export to Phy

farazmoradi, 2021-07-21:
Hi, 

Code : 

output_folder = 'mearec_exported_to_phy'
so.export_to_phy(rec_on_disk, sorting, output_folder,we,sampling_frequency=10000,
                 compute_pc_features=False, compute_amplitudes=True,
                 remove_if_exists=True)

Error : 

Traceback (most recent call last):
  File "/Users/faraz/.conda/envs/SpikeSorting/lib/python3.9/site-packages/IPython/core/interactiveshell.py", line 3441, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-15-9a4f6dc9fb3b>", line 2, in <module>
    so.export_to_phy(rec_on_disk, sorting, output_folder,we,sampling_frequency=10000,
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/exporters/to_phy.py", line 98, in export_to_phy
    write_binary_recording(recording, files_path=rec_path, verbose=verbose, dtype=dtype, **job_kwargs)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/core_tools.py", line 184, in write_binary_recording
    executor = ChunkRecordingExecutor(recording, func, init_func, init_args, verbose=verbose,
TypeError: __init__() got an unexpected keyword argument 'sampling_frequency'

alejoe91, 2021-07-21:
Why are you passing sampling_frequency? If you read the error it tells you that it's not needed...

farazmoradi, 2021-07-21:
Actually, the problem continues after I removed sampling_ frequency

Traceback (most recent call last):
  File "/Users/faraz/.conda/envs/SpikeSorting/lib/python3.9/site-packages/IPython/core/interactiveshell.py", line 3441, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-17-c665d9d3404c>", line 2, in <module>
    so.export_to_phy(rec_on_disk, sorting, output_folder,we,
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/exporters/to_phy.py", line 160, in export_to_phy
    amplitudes = get_unit_amplitudes(waveform_extractor,  peak_sign=peak_sign, outputs='concatenated', **job_kwargs)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/toolkit/postprocessing/unit_amplitudes.py", line 37, in get_unit_amplitudes
    out = processor.run()
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/job_tools.py", line 166, in run
    worker_ctx = self.init_func(*self.init_args)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/toolkit/postprocessing/unit_amplitudes.py", line 69, in _init_worker_unit_amplitudes
    sorting = load_extractor(sorting)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/base.py", line 722, in load_extractor
    return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/base.py", line 297, in from_dict
    extractor = _load_extractor_from_dict(d)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/base.py", line 670, in _load_extractor_from_dict
    extractor = cls(**kwargs)
TypeError: __init__() missing 1 required positional argument: 'sampling_frequency'

alejoe91, 2021-07-21:
Ok, what's your sorting object? @samuelgarcia we might be missing some kwargs

farazmoradi, 2021-07-21:
I have done spike sorting with Tridesclous GUI, after that I export spikes to .mat. Then load them as Neo.SpikeTrain and input them to the spike interface. total consist of SpikeTrain for each unit. 

sorting = se.NumpySorting.from_neo_spiketrain_list(total,sampling_frequency=10000,unit_ids=list(range(16)))

samuelgarcia, 2021-07-21:
I tghink you should use the `sorting = read_tridesclous(tdc_folder)` instead of theses export.

When you get a sorting from Neo.SpikeTrain you need the sampling rate indeed. The other method is straight forward

farazmoradi, 2021-07-21:
Thanks Sam,  still receiving that error!

Code:

sorting = se.read_tridesclous('/Volumes/Projects/Antwerp Mice 1/EDF/d1-b1/tdc_d1_b1_TG1')
output_folder = 'mearec_exported_to_phy'
so.export_to_phy(rec_on_disk, sorting, output_folder,we)

Error:

Traceback (most recent call last):
  File "/Users/faraz/.conda/envs/SpikeSorting/lib/python3.9/site-packages/IPython/core/interactiveshell.py", line 3441, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-19-cc58e63a0468>", line 3, in <module>
    so.export_to_phy(rec_on_disk, sorting, output_folder,we)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/exporters/to_phy.py", line 160, in export_to_phy
    amplitudes = get_unit_amplitudes(waveform_extractor,  peak_sign=peak_sign, outputs='concatenated', **job_kwargs)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/toolkit/postprocessing/unit_amplitudes.py", line 37, in get_unit_amplitudes
    out = processor.run()
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/job_tools.py", line 166, in run
    worker_ctx = self.init_func(*self.init_args)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/toolkit/postprocessing/unit_amplitudes.py", line 69, in _init_worker_unit_amplitudes
    sorting = load_extractor(sorting)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/base.py", line 722, in load_extractor
    return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/base.py", line 297, in from_dict
    extractor = _load_extractor_from_dict(d)
  File "/Volumes/Projects/Antwerp Mice 1/SpikeSorting/python-neo/probeinterface/spikeinterface/spikeinterface/core/base.py", line 670, in _load_extractor_from_dict
    extractor = cls(**kwargs)
TypeError: __init__() missing 1 required positional argument: 'sampling_frequency'

farazmoradi, 2021-07-21:
I will try but it was surprising that **read_tridesclous** did not work!

farazmoradi, 2021-07-21:
And the error still exists unfortunately :(

samuelgarcia, 2021-09-17:
Is the error still there or is fix with the current master ?
Can we close this ?

farazmoradi, 2021-09-17:
For sure, we can close this. It has been fixed.

closed 2021-09-17 by alejoe91